max-baz/wluma #1: Failed to find backlight device

piozylka, 2020-09-25:
running wluma:
`ERROR: Failed to find backlight device file in base dir: /sys/class/backlight`

ls /sys/class/backlight
`lrwxrwxrwx root root 0 B Thu Sep 24 08:15:56 2020 intel_backlight ⇒ ../../devices/pci0000:00/0000:00:02.0/drm/card0/card0-eDP-1/intel_backlight`

any chances to wluma automatically detects type of graphic card ?
or some command line switches to point to correct folder ?

maximbaz, 2020-09-25:
Interesting because I have the same path as you.

Could you check if these files exist for you?

```
$ ls /sys/class/backlight/intel_backlight/max_brightness
$ ls /sys/class/backlight/intel_backlight/brightness
```

piozylka, 2020-09-25:
yes these files exists and i can print values inside these files

`.r--r--r-- root root 4 KB Thu Sep 24 21:56:09 2020 actual_brightness                               
.rw-r--r-- root root 4 KB Thu Sep 24 08:15:56 2020 bl_power                                        
.rw-r--r-- root root 4 KB Thu Sep 24 08:15:56 2020 brightness                                      
lrwxrwxrwx root root 0 B  Thu Sep 24 21:56:09 2020 device ⇒ ../../card0-eDP-1                      
.r--r--r-- root root 4 KB Thu Sep 24 08:15:56 2020 max_brightness                                  
drwxr-xr-x root root 0 B  Thu Sep 24 21:56:09 2020 power                                           
.r--r--r-- root root 4 KB Thu Sep 24 21:56:09 2020 scale                                           
lrwxrwxrwx root root 0 B  Thu Sep 24 08:15:56 2020 subsystem ⇒ ../../../../../../../class/backlight
.r--r--r-- root root 4 KB Thu Sep 24 08:15:56 2020 type                                            
.rw-r--r-- root root 4 KB Thu Sep 24 08:15:56 2020 uevent `

piozylka, 2020-09-25:
tried to run with sudo 
`ERROR: Failed to open light sensor base dir: /sys/bus/iio/devices`
maybe this will help

maximbaz, 2020-09-25:
Ah I think I know, `wluma` process must be able to write values into the `brightness` file, but in your case it is only writeable by `root`, while in my case it is also writeable by `video` group, and my user is part of this group:

```
$ la /sys/class/backlight/intel_backlight/brightness
.rw-rw-r-- 4.1k root video 25 Sep 11:58 /sys/class/backlight/intel_backlight/brightness
```

I think this `video` group is something standard, I definitely have since the time I was using `light` tool.

The fact that you get further with `sudo` confirms this 🙂

maximbaz, 2020-09-25:
As for the light sensor error, it seems your laptop doesn't have light sensor (so wluma cannot tell how bright it is around you right now), have a look here on how to simulate this device by various strategies:

https://github.com/maximbaz/wluma#strategies-other-than-ambient-light-sensor

I'm actually re-considering the algorithm, so if you have any ideas or feedback as you will use the tool, they are very very appreciated.

I'm not super happy with using light sensor directly, because (1) it is not present in all laptops, and (2) on my laptop it is quite imprecise, it cannot tell dim light from complete darkness. 

So I'm considering to change the algorithm, maybe make wluma use time by default, and only if sensor is present, then additionally slightly correct the brightness...

maximbaz, 2021-08-17:
Fixed in `main`, new major breaking release will be soon, but for now you can apply https://github.com/maximbaz/wluma/commit/827629d618ac38ddd8fcbb2baab535c93859d11c manually or use `AUR/wluma-git`

closed 2021-08-17 by maximbaz