max-baz/wluma #2: Support for IIO RGBC "intensity" based sensors

archseer, 2020-12-12:
My laptop uses an apds9960 based sensor that doesn't report a single processed illuminance value, but uses RGBC. An illuminance value has to be manually calculated from the values.

> What:		/sys/bus/iio/devices/device[n]/in_intensity_clear[_input|_raw]
> What:		/sys/bus/iio/devices/device[n]/in_intensity_red[_input|_raw]
> What:		/sys/bus/iio/devices/device[n]/in_intensity_green[_input|_raw](
> What:		/sys/bus/iio/devices/device[n]/in_intensity_blue[_input|_raw]
> KernelVersion:	3.6.0
> Contact:	linux-iio@vger.kernel.org
> Description:
>  This property is supported by sensors that have a RGBC and if expressed in SI units, should include _input. If this value is not in SI units (irradiance, uW/mm^2), then it should include _raw.

There's an easy formula that produces wrong results but is a good start:

```c
(-0.32466F * color_data.r) + (1.57837F * color_data.g) + (-0.73191F * color_data.b)
```

There's a more complex formula but we don't know the constants for apds9960 sadly:

https://github.com/adafruit/Adafruit_CircuitPython_TCS34725/blob/master/adafruit_tcs34725.py#L282-L332

maximbaz, 2020-12-13:
Very interesting, thanks for letting me know! I will get back to you once I implement something, I'm still considering what is the best approach moving forward, light sensors are too imprecise (at least on my laptop) to provide a completely pleasant auto-adjusting brightness experience, so I'm considering alternative algorithms, maybe base the decision on time of the day and only take illuminance as an additional factor... 🤔 

If you have time, could you try to test if your sensor is able to distinguish complete darkness from dim light? Especially if the light is not shining directly at your laptop sensor or to your face (thus reflecting directly to your laptop sensor), but e.g. the light is on your right or left hand side.

archseer, 2020-12-13:
I think it should be quite reliable! It measures ambient light levels as opposed to reflected light. The sensor in my device (Surface Laptop 3) is not only used for brightness, but also for adaptive color profiles on Windows. (I think that's why they chose a RGBC sensor. It also does proximity and gesture detection, I think they use that for the "Windows Hello" login) The `iio` sysfs entries have a few options for setting the sensitivity and scale of the sensor.

I think you might need to apply a kalman filter over the sensor data to smooth out the values, and then map the values to a custom curve. This whitepaper [intended for hardware manufacturers](https://docs.microsoft.com/en-us/windows-hardware/design/whitepapers/integrating-ambient-light-sensors-with-computers-running-windows-10-creators-update) might be interesting because it provides a few example curves and a reference table of what the lux values are at various settings (indoors, etc.).

I'm primarily interested in ambient light adjustments for potential battery savings, not the screen content sensing (wlsunset takes care of day vs night changes by adjusting the color profile). Was going to write my own daemon but then I found wluma :)

maximbaz, 2020-12-13:
Thank you, this is very interesting info!

I honestly think that when done right, screen content sensing enhances the experience so much that it is then impossible to live without once it you try it out, especially with the fast reaction time and low CPU usage that I managed to achieve so far. 

Unfortunately I don't think my current algorithm is good enough yet, so I want to try to win you over, and with your additional feedback and ideas to perfect the algorithm 😜 

Implementing the approximation formula for your sensor should be easy enough to get you going. I'm only not sure if the formula applies to both `_input` and `_raw` or only to one of them? 🤔 And which of these values does your specific sensor provide?

By the way if you feel like sending a PR, by all means, please do 🙂

maximbaz, 2021-08-17:
Implemented in `main`, untested but should work 😅 Thanks for all the hints.

closed 2021-08-17 by maximbaz

AeliusSaionji, 2025-03-13:
Hello! I have the same sensor device, and I'm not seeing it working.

I was able to follow your advice in another thread to operate on the values from red alone:
```
/home/aelius/.config/wluma
 ├──config.toml 
 └──sensor 
    └──values 
       ├──in_illuminance_raw -> /sys/devices/pci0000:00/0000:00:19.0/i2c_designware.4/i2c-4/i2c-MSHW0184:00/iio:device0/in_intensity_red_raw 
       └──name
```

But I was wondering if your implementation needed some configuration to turn on?

Thanks!

Here's the pdf spec sheet for the apds9960, if it's worth anything to you :)

https://docs.broadcom.com/doc/AV02-4191EN

AeliusSaionji, 2025-03-13:
I figured out the problem- "name" is not `als`, it's `apds9960`. Tricking it allows wluma to find the device.

Unfortunately it's always reporting 0.

```
/home/aelius/.config/wluma
 ├──config.toml 
 └──sensor 
    └──values 
       ├──in_intensity_blue_raw -> /sys/devices/pci0000:00/0000:00:19.0/i2c_designware.4/i2c-4/i2c-MSHW0184:00/iio:device0/in_intensity_blue_raw 
       ├──in_intensity_green_raw -> /sys/devices/pci0000:00/0000:00:19.0/i2c_designware.4/i2c-4/i2c-MSHW0184:00/iio:device0/in_intensity_green_raw 
       ├──in_intensity_red_raw -> /sys/devices/pci0000:00/0000:00:19.0/i2c_designware.4/i2c-4/i2c-MSHW0184:00/iio:device0/in_intensity_red_raw 
       └──name 
```

```
[2025-03-13T04:49:43Z DEBUG wluma] == wluma v4.8.0 ==
[2025-03-13T04:49:43Z DEBUG wluma] Using Config {
        als: Iio {
            path: "/home/aelius/.config/wluma/sensor",
            thresholds: {
                500: "bright",
                800: "outdoors",
                20: "dark",
                0: "night",
                80: "dim",
                250: "normal",
            },
        },
        output: [
            Backlight(
                BacklightOutput {
                    name: "eDP-1",
                    path: "/sys/class/backlight/intel_backlight",
                    capturer: Wayland(
                        Any,
                    ),
                    min_brightness: 1,
                    predictor: Adaptive,
                },
            ),
        ],
    }
[2025-03-13T04:49:43Z DEBUG wluma::brightness::backlight] Using DBUS for /sys/class/backlight/intel_backlight to change brightness value
[2025-03-13T04:49:43Z INFO  wluma] Continue adjusting brightness and wluma will learn your preference over time.
[2025-03-13T04:49:43Z DEBUG wluma::frame::capturer::wayland] Detected support for linux-dmabuf-v1 protocol
[2025-03-13T04:49:43Z DEBUG wluma::frame::capturer::wayland] Detected support for wlr-export-dmabuf-unstable-v1 protocol
[2025-03-13T04:49:43Z DEBUG wluma::frame::capturer::wayland] Detected support for wlr-screencopy-unstable-v1 protocol
[2025-03-13T04:49:43Z DEBUG wluma::frame::capturer::wayland] Using output 'Panasonic Industry Company TDM13O56     0x05A11100 (eDP-1)' for config 'eDP-1'
[2025-03-13T04:49:43Z DEBUG wluma::frame::capturer::wayland] Using wlr-screencopy-unstable-v1 protocol to request frames
[2025-03-13T04:49:43Z TRACE wluma::als::iio] ALS (iio): night (0)
[2025-03-13T04:49:43Z TRACE wluma::predictor::controller::adaptive] [eDP-1] Prediction: 14400 (lux: night, luma: 29)
[2025-03-13T04:49:43Z TRACE wluma::als::iio] ALS (iio): night (0)
[2025-03-13T04:49:43Z TRACE wluma::predictor::controller::adaptive] [eDP-1] Prediction: 14400 (lux: night, luma: 29)
[2025-03-13T04:49:44Z TRACE wluma::als::iio] ALS (iio): night (0)
[2025-03-13T04:49:44Z TRACE wluma::predictor::controller::adaptive] [eDP-1] Prediction: 14400 (lux: night, luma: 29)
[2025-03-13T04:49:44Z TRACE wluma::als::iio] ALS (iio): night (0)
[2025-03-13T04:49:44Z TRACE wluma::predictor::controller::adaptive] [eDP-1] Prediction: 14400 (lux: night, luma: 29)
[2025-03-13T04:49:44Z TRACE wluma::als::iio] ALS (iio): night (0)
[2025-03-13T04:49:44Z TRACE wluma::predictor::controller::adaptive] [eDP-1] Prediction: 14400 (lux: night, luma: 29)
[2025-03-13T04:49:44Z TRACE wluma::als::iio] ALS (iio): night (0)
[2025-03-13T04:49:44Z TRACE wluma::predictor::controller::adaptive] [eDP-1] Prediction: 14400 (lux: night, luma: 29)
[2025-03-13T04:49:44Z TRACE wluma::als::iio] ALS (iio): night (0)
[2025-03-13T04:49:44Z TRACE wluma::predictor::controller::adaptive] [eDP-1] Prediction: 14400 (lux: night, luma: 29)
[2025-03-13T04:49:44Z TRACE wluma::als::iio] ALS (iio): night (0)
[2025-03-13T04:49:44Z TRACE wluma::als::iio] ALS (iio): night (0)
[2025-03-13T04:49:44Z TRACE wluma::predictor::controller::adaptive] [eDP-1] Prediction: 14400 (lux: night, luma: 29)
```

maximbaz, 2025-03-13:
Interesting! Could you please show me which files you have inside the folder, and their values?
Something like this, for example:

```sh
for file in /sys/devices/pci0000:00/0000:00:19.0/i2c_designware.4/i2c-4/i2c-MSHW0184:00/iio:device0/*; do
  echo "== $file"
  cat "$file"
done
```

AeliusSaionji, 2025-03-13:
```log
cat /sys/bus/iio/devices/iio:device0/buffer/data_available
	0

cat /sys/bus/iio/devices/iio:device0/buffer/direction
	in

cat /sys/bus/iio/devices/iio:device0/buffer/enable
	0

cat /sys/bus/iio/devices/iio:device0/buffer/length
	2

cat /sys/bus/iio/devices/iio:device0/buffer/watermark
	1

cat /sys/bus/iio/devices/iio:device0/buffer0/data_available
	0

cat /sys/bus/iio/devices/iio:device0/buffer0/direction
	in

cat /sys/bus/iio/devices/iio:device0/buffer0/enable
	0

cat /sys/bus/iio/devices/iio:device0/buffer0/in_proximity1_en
	0

cat /sys/bus/iio/devices/iio:device0/buffer0/in_proximity1_index
	0

cat /sys/bus/iio/devices/iio:device0/buffer0/in_proximity1_type
	le:u8/8>>0

cat /sys/bus/iio/devices/iio:device0/buffer0/in_proximity2_en
	0

cat /sys/bus/iio/devices/iio:device0/buffer0/in_proximity2_index
	1

cat /sys/bus/iio/devices/iio:device0/buffer0/in_proximity2_type
	le:u8/8>>0

cat /sys/bus/iio/devices/iio:device0/buffer0/in_proximity3_en
	0

cat /sys/bus/iio/devices/iio:device0/buffer0/in_proximity3_index
	2

cat /sys/bus/iio/devices/iio:device0/buffer0/in_proximity3_type
	le:u8/8>>0

cat /sys/bus/iio/devices/iio:device0/buffer0/in_proximity4_en
	0

cat /sys/bus/iio/devices/iio:device0/buffer0/in_proximity4_index
	3

cat /sys/bus/iio/devices/iio:device0/buffer0/in_proximity4_type
	le:u8/8>>0

cat /sys/bus/iio/devices/iio:device0/buffer0/length
	2

cat /sys/bus/iio/devices/iio:device0/buffer0/watermark
	1

cat /sys/bus/iio/devices/iio:device0/dev
	508:0

cat /sys/bus/iio/devices/iio:device0/events/in_intensity_clear_thresh_either_en
	0

cat /sys/bus/iio/devices/iio:device0/events/in_intensity_clear_thresh_falling_value
	0

cat /sys/bus/iio/devices/iio:device0/events/in_intensity_clear_thresh_rising_value
	0

cat /sys/bus/iio/devices/iio:device0/events/in_proximity_thresh_either_en
	0

cat /sys/bus/iio/devices/iio:device0/events/in_proximity_thresh_falling_value
	0

cat /sys/bus/iio/devices/iio:device0/events/in_proximity_thresh_rising_value
	0

cat /sys/bus/iio/devices/iio:device0/in_intensity_blue_raw
	5

cat /sys/bus/iio/devices/iio:device0/in_intensity_clear_raw
	1

cat /sys/bus/iio/devices/iio:device0/in_intensity_green_raw
	17

cat /sys/bus/iio/devices/iio:device0/in_intensity_integration_time
	0.028000

cat /sys/bus/iio/devices/iio:device0/in_intensity_red_raw
	5

cat /sys/bus/iio/devices/iio:device0/in_intensity_scale
	1

cat /sys/bus/iio/devices/iio:device0/in_proximity1_calibbias
	0

cat /sys/bus/iio/devices/iio:device0/in_proximity2_calibbias
	0

cat /sys/bus/iio/devices/iio:device0/in_proximity3_calibbias
	0

cat /sys/bus/iio/devices/iio:device0/in_proximity4_calibbias
	0

cat /sys/bus/iio/devices/iio:device0/in_proximity_calibbias
	0

cat /sys/bus/iio/devices/iio:device0/in_proximity_raw
	0

cat /sys/bus/iio/devices/iio:device0/in_proximity_scale
	1

cat /sys/bus/iio/devices/iio:device0/integration_time_available
	0.028 0.1 0.2 0.7

cat /sys/bus/iio/devices/iio:device0/intensity_scale_available
	1 4 16 64

cat /sys/bus/iio/devices/iio:device0/name
	apds9960

cat /sys/bus/iio/devices/iio:device0/power/autosuspend_delay_ms
	cat: '/sys/bus/iio/devices/iio:device0/power/autosuspend_delay_ms': Input/output error

cat /sys/bus/iio/devices/iio:device0/power/control
	auto

cat /sys/bus/iio/devices/iio:device0/power/runtime_active_time
	0

cat /sys/bus/iio/devices/iio:device0/power/runtime_status
	unsupported

cat /sys/bus/iio/devices/iio:device0/power/runtime_suspended_time
	0

cat /sys/bus/iio/devices/iio:device0/proximity_scale_available
	1 2 4 8

cat /sys/bus/iio/devices/iio:device0/scan_elements/in_proximity1_en
	0

cat /sys/bus/iio/devices/iio:device0/scan_elements/in_proximity1_index
	0

cat /sys/bus/iio/devices/iio:device0/scan_elements/in_proximity1_type
	le:u8/8>>0

cat /sys/bus/iio/devices/iio:device0/scan_elements/in_proximity2_en
	0

cat /sys/bus/iio/devices/iio:device0/scan_elements/in_proximity2_index
	1

cat /sys/bus/iio/devices/iio:device0/scan_elements/in_proximity2_type
	le:u8/8>>0

cat /sys/bus/iio/devices/iio:device0/scan_elements/in_proximity3_en
	0

cat /sys/bus/iio/devices/iio:device0/scan_elements/in_proximity3_index
	2

cat /sys/bus/iio/devices/iio:device0/scan_elements/in_proximity3_type
	le:u8/8>>0

cat /sys/bus/iio/devices/iio:device0/scan_elements/in_proximity4_en
	0

cat /sys/bus/iio/devices/iio:device0/scan_elements/in_proximity4_index
	3

cat /sys/bus/iio/devices/iio:device0/scan_elements/in_proximity4_type
	le:u8/8>>0

cat /sys/bus/iio/devices/iio:device0/uevent
	MAJOR=508
	MINOR=0
	DEVNAME=iio:device0
	DEVTYPE=iio_device

cat /sys/bus/iio/devices/iio:device0/waiting_for_supplier
	0
```

maximbaz, 2025-03-13:
Thanks! Whitelisting the sensor name `apds9960` definitely makes sense, this is now pushed to `main`. Your sensor is the same one as for the previous person, so I don't feel like we need to adjust any code. Are you sure it doesn't work? Could you shine phone flashlight into your sensor? 😅 

Plugging the raw values from your message above, into the formula from the first message here, I'm getting:

`(−0.32466 × 5) + (1.57837 × 17) + (−0.73191 × 5) = 21.55 = dark` in the default config...

Could you try the latest `main`, just to be sure?

AeliusSaionji, 2025-03-13:
latest commit is working! Guess there was just something funky about my symlinks. Thanks!

Aside- looks like the wluma-git package in the AUR is from before you switched to `just`.

maximbaz, 2025-03-13:
Great to hear! 🥳 

I'm not in control of AUR packages, but I know that the non-git one did update the build steps - I just released 4.9.0 for you, so you can enjoy the app using the non-git AUR package 😁