max-baz/wluma #3: Fake device not working

jedugsem, 2021-02-09:
ERROR: Failed to find ambient light sensor device file in base dir: /tmp/fake-devices/light-sensor/      although i started wluma-als-emulator --strategy=time

jedugsem, 2021-02-09:
cat /tmp/fake-device/light/sensor/in_illumiance_raw returns 48 and i did : export WLUMA_AMBIENT_LIGHT_SENSOR_BASE_PATH=/tmp/fake-devices/light-sensor/

maximbaz, 2021-02-09:
> cat /tmp/fake-device/light/sensor/in_illumiance_raw

seems like you have `light/sensor` instead of `light-sensor`?

jedugsem, 2021-02-09:
sry wrote that by hand i have the correct one, thanks for the fast reply

maximbaz, 2021-02-09:
Ahh I know, `WLUMA_AMBIENT_LIGHT_SENSOR_BASE_PATH` should be set to `/tmp/fake-devices/`, please try 🙂

jedugsem, 2021-02-10:
thanks works now but if i'am on light apps it doesnt change the brightness

maximbaz, 2021-02-10:
awesome! it is not expected, you need to train the algorithm first, see https://github.com/maximbaz/wluma#usage