max-baz/wluma #6: use of unstable library feature 'option_insert'

532910, 2021-08-24:
```
error[E0658]: use of unstable library feature 'option_insert': newly added
   --> src/frame/processor/vulkan.rs:215:33
    |
215 |         self.image.borrow_mut().insert(image);
    |                                 ^^^^^^
    |
    = note: see issue #78271 <https://github.com/rust-lang/rust/issues/78271> for more information

error[E0658]: use of unstable library feature 'option_insert': newly added
   --> src/frame/processor/vulkan.rs:216:40
    |
216 |         self.image_memory.borrow_mut().insert(image_memory);
    |                                        ^^^^^^
    |
    = note: see issue #78271 <https://github.com/rust-lang/rust/issues/78271> for more information

error: aborting due to 2 previous errors

For more information about this error, try `rustc --explain E0658`.
error: could not compile `wluma`.

To learn more, run the command again with --verbose.
make: *** [Makefile:11: build-dev] Error 101
```

```
% apt policy rustc 
rustc:
  Installed: 1.52.1+dfsg1-1~exp3
  Candidate: 1.52.1+dfsg1-1~exp3
  Version table:
 *** 1.52.1+dfsg1-1~exp3 200
        200 https://deb.debian.org/debian experimental/main amd64 Packages
        100 /var/lib/dpkg/status
     1.48.0+dfsg1-2 800
        800 https://deb.debian.org/debian sid/main amd64 Packages
```

maximbaz, 2021-08-24:
Hi there 🙂 You need to use the latest stable rust, I haven't used `rustup` myself but I think it's what everyone recommends to do.

I added a little note in README, feel free to expand 🙂 https://github.com/maximbaz/wluma/commit/73d45e06389c315fd107451cf601faf8cb8d42fb

Finally I also attached pre-compiled release archive, on linux you can download that and it contains everything necessary to run `make install`.

532910, 2021-08-25:
Indeed, it builds with rust 1.54. Should version number be checked at build time?

maximbaz, 2021-08-25:
I don't think I want to chase rapidly-changing Rust versions in this project 😅 

I expanded README a bit, also added CI badge showing whether Github itself was able to compile the app and execute tests: https://github.com/maximbaz/wluma/commit/3a7ab6281738c0f002f485e11018c7ccd2c3330e

The logic being, if Github was able to compile, the issue is probably on end-user side. How does this sound?

532910, 2021-08-25:
sounds ok

closed 2021-08-25 by 532910