max-baz/wluma #7: binary release glibc dependency

532910, 2021-08-25:
debian bullseye:
```
% ./wluma 
./wluma: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required by ./wluma)
./wluma: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.32' not found (required by ./wluma)

% apt policy libc6
libc6:
  Installed: 2.31-13
  Candidate: 2.31-13
  Version table:
 *** 2.31-13 500
        500 https://deb.debian.org/debian bullseye/main amd64 Packages
        100 /var/lib/dpkg/status
```

debian sid:
```
% ./target/release/wluma             
./target/release/wluma: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required by ./target/release/wluma)
./target/release/wluma: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.32' not found (required by ./target/release/wluma)

% apt policy libc6                                                                                  
libc6:
  Installed: 2.31-17
  Candidate: 2.31-17
  Version table:
     2.32-0experimental1 200
        200 https://deb.debian.org/debian experimental/main amd64 Packages
 *** 2.31-17 800
        800 https://deb.debian.org/debian sid/main amd64 Packages
        100 /var/lib/dpkg/status
```

maximbaz, 2021-08-25:
Looks like nothing I can do, I tried to compile a statically-linked binary (which uses musl instead of glibc) but it just doesn't support all the necessary dependencies for `wluma` to operate:

```
$ cargo build --locked --release --target x86_64-unknown-linux-musl
...
$ ./target/x86_64-unknown-linux-musl/release/wluma
thread 'main' panicked at 'Unable to initialize Vulkan: LibraryLoadFailure(DlOpen { desc: "Dynamic loading not supported" })', src/main.rs:27:56
```

I guess I'll have to document that release binary is compiled against latest stable `glibc`.

maximbaz, 2021-08-25:
https://github.com/maximbaz/wluma/commit/600ce707456dc398ba6c0e025f33c69a366bf569

As usual, any other suggestions are welcome 🙂 

It's not new problem with rust + debian btw... https://github.com/sayanarijit/xplr/issues/269#issuecomment-867894057

closed 2021-08-25 by maximbaz

532910, 2021-08-25:
> statically-linked binary (which uses musl instead of glibc

is it possible to link statically with glibc?

maximbaz, 2021-08-25:
With my limited knowledge, I think no... Happy to be proven wrong though 😁 I think musl was made particularly to allow making statically-linked binaries for those who until then used glibc... But obviously this doesnt work every time, I did have some success with musl in the past, but not with wluma, not with xplr...