max-baz/wluma #8: Frames with multiple objects are not supported yet

532910, 2021-08-25:
```
% RUST_BACKTRACE=full sudo ./target/debug/wluma
Using config: Config { frame: Frame { capturer: Wlroots, processor: Vulkan }, als: None, output_by_type: OutputByType { backlight: {}, ddcutil: {} }, keyboard: None, output: {"eDP-1": Backlight(BacklightOutput { path: "/sys/class/backlight/intel_backlight", use_contents: false })} }
Continue adjusting brightness and wluma will learn your preference over time.
thread 'main' panicked at 'assertion failed: `(left == right)`
  left: `1`,
 right: `2`: Frames with multiple objects are not supported yet', src/frame/processor/vulkan.rs:33:9
stack backtrace:
   0:     0x55999594f530 - std::backtrace_rs::backtrace::libunwind::trace::h34055254b57d8e79
                               at /rustc/a178d0322ce20e33eac124758e837cbd80a6f633/library/std/src/../../backtrace/src/backtrace/libunwind.rs:90:5
   1:     0x55999594f530 - std::backtrace_rs::backtrace::trace_unsynchronized::h8f1e3fbd9afff6ec
                               at /rustc/a178d0322ce20e33eac124758e837cbd80a6f633/library/std/src/../../backtrace/src/backtrace/mod.rs:66:5
   2:     0x55999594f530 - std::sys_common::backtrace::_print_fmt::h3a99a796b770c360
                               at /rustc/a178d0322ce20e33eac124758e837cbd80a6f633/library/std/src/sys_common/backtrace.rs:67:5
   3:     0x55999594f530 - <std::sys_common::backtrace::_print::DisplayBacktrace as core::fmt::Display>::fmt::h32d1f94a80615d18
                               at /rustc/a178d0322ce20e33eac124758e837cbd80a6f633/library/std/src/sys_common/backtrace.rs:46:22
   4:     0x55999597059c - core::fmt::write::h306731c068f7162c
                               at /rustc/a178d0322ce20e33eac124758e837cbd80a6f633/library/core/src/fmt/mod.rs:1110:17
   5:     0x55999594d265 - std::io::Write::write_fmt::hd2fa90334eee2a21
                               at /rustc/a178d0322ce20e33eac124758e837cbd80a6f633/library/std/src/io/mod.rs:1588:15
   6:     0x55999595148b - std::sys_common::backtrace::_print::h5abaa2601a852287
                               at /rustc/a178d0322ce20e33eac124758e837cbd80a6f633/library/std/src/sys_common/backtrace.rs:49:5
   7:     0x55999595148b - std::sys_common::backtrace::print::h8d81445442bb638f
                               at /rustc/a178d0322ce20e33eac124758e837cbd80a6f633/library/std/src/sys_common/backtrace.rs:36:9
   8:     0x55999595148b - std::panicking::default_hook::{{closure}}::hcfe804496a9fa747
                               at /rustc/a178d0322ce20e33eac124758e837cbd80a6f633/library/std/src/panicking.rs:208:50
   9:     0x559995950f61 - std::panicking::default_hook::hbea8e3ccf2ba8901
                               at /rustc/a178d0322ce20e33eac124758e837cbd80a6f633/library/std/src/panicking.rs:225:9
  10:     0x559995951b54 - std::panicking::rust_panic_with_hook::h7ee9e1a2d0f8975a
                               at /rustc/a178d0322ce20e33eac124758e837cbd80a6f633/library/std/src/panicking.rs:622:17
  11:     0x559995951637 - std::panicking::begin_panic_handler::{{closure}}::h8ab3b4491718b2c7
                               at /rustc/a178d0322ce20e33eac124758e837cbd80a6f633/library/std/src/panicking.rs:519:13
  12:     0x55999594fa2c - std::sys_common::backtrace::__rust_end_short_backtrace::hd489062ffa586a9f
                               at /rustc/a178d0322ce20e33eac124758e837cbd80a6f633/library/std/src/sys_common/backtrace.rs:141:18
  13:     0x559995951599 - rust_begin_unwind
                               at /rustc/a178d0322ce20e33eac124758e837cbd80a6f633/library/std/src/panicking.rs:515:5
  14:     0x5599956cd1d1 - core::panicking::panic_fmt::hca6330e3e14086b4
                               at /rustc/a178d0322ce20e33eac124758e837cbd80a6f633/library/core/src/panicking.rs:92:14
  15:     0x55999596ecee - core::panicking::assert_failed_inner::h33b45280069a80ed
  16:     0x55999592e51a - core::panicking::assert_failed::hd5d39009fc2e6e38
                               at /rustc/a178d0322ce20e33eac124758e837cbd80a6f633/library/core/src/panicking.rs:117:5
  17:     0x5599957eca86 - <wluma::frame::processor::vulkan::Processor as wluma::frame::processor::Processor>::luma_percent::hf23bf165a7e7c977
                               at /home/sergio/wluma/wluma/src/frame/processor/vulkan.rs:33:9
  18:     0x559995723a7e - wluma::frame::capturer::wlroots::Capturer::capture_frame::{{closure}}::hde5f027331db226b
                               at /home/sergio/wluma/wluma/src/frame/capturer/wlroots.rs:84:32
  19:     0x55999571e6f9 - wayland_client::proxy::Main<I>::quick_assign::{{closure}}::hc1e7f3c4bfd2ecaf
                               at /home/sergio/.cargo/registry/src/github.com-1ecc6299db9ec823/wayland-client-0.29.0/src/proxy.rs:273:64
  20:     0x559995783182 - wayland_commons::filter::Filter<E>::send::h43a3d7e7c9d79941
                               at /home/sergio/.cargo/registry/src/github.com-1ecc6299db9ec823/wayland-commons-0.29.0/src/filter.rs:100:13
  21:     0x5599956dffef - wayland_client::imp::make_dispatcher::{{closure}}::hd5e454f34ef927ba
                               at /home/sergio/.cargo/registry/src/github.com-1ecc6299db9ec823/wayland-client-0.29.0/src/rust_imp/mod.rs:194:49
  22:     0x5599956dfb4d - <wayland_client::imp::ImplDispatcher<I,F> as wayland_client::imp::Dispatcher>::dispatch::hb15f2f9aef44e051
                               at /home/sergio/.cargo/registry/src/github.com-1ecc6299db9ec823/wayland-client-0.29.0/src/rust_imp/mod.rs:179:9
  23:     0x559995725c39 - wayland_client::imp::queues::EventQueueInner::dispatch_buffer::h1c059b0df934cc63
                               at /home/sergio/.cargo/registry/src/github.com-1ecc6299db9ec823/wayland-client-0.29.0/src/rust_imp/queues.rs:163:23
  24:     0x55999572ab5f - wayland_client::imp::queues::EventQueueInner::dispatch_pending::h4f77d1714405bd5d
                               at /home/sergio/.cargo/registry/src/github.com-1ecc6299db9ec823/wayland-client-0.29.0/src/rust_imp/queues.rs:198:31
  25:     0x55999572b3f7 - wayland_client::imp::queues::EventQueueInner::dispatch::h6836c9a3dc9983ca
                               at /home/sergio/.cargo/registry/src/github.com-1ecc6299db9ec823/wayland-client-0.29.0/src/rust_imp/queues.rs:102:28
  26:     0x559995730ee7 - wayland_client::event_queue::EventQueue::dispatch::h94e6e99670791d0c
                               at /home/sergio/.cargo/registry/src/github.com-1ecc6299db9ec823/wayland-client-0.29.0/src/event_queue.rs:152:9
  27:     0x55999573251c - <wluma::frame::capturer::wlroots::Capturer as wluma::frame::capturer::Capturer>::run::h9649790b732a7765
                               at /home/sergio/wluma/wluma/src/frame/capturer/wlroots.rs:29:13
  28:     0x559995749d43 - wluma::main::h82011a5073039c33
                               at /home/sergio/wluma/wluma/src/main.rs:61:5
  29:     0x5599957f3b4b - core::ops::function::FnOnce::call_once::hcde0aa4c2b506cf5
                               at /rustc/a178d0322ce20e33eac124758e837cbd80a6f633/library/core/src/ops/function.rs:227:5
  30:     0x55999576806e - std::sys_common::backtrace::__rust_begin_short_backtrace::hbd999ebb6582003e
                               at /rustc/a178d0322ce20e33eac124758e837cbd80a6f633/library/std/src/sys_common/backtrace.rs:125:18
  31:     0x5599957986b1 - std::rt::lang_start::{{closure}}::hbe445be29ab97048
                               at /rustc/a178d0322ce20e33eac124758e837cbd80a6f633/library/std/src/rt.rs:49:18
  32:     0x559995952059 - core::ops::function::impls::<impl core::ops::function::FnOnce<A> for &F>::call_once::h2aabc384aab89b7b
                               at /rustc/a178d0322ce20e33eac124758e837cbd80a6f633/library/core/src/ops/function.rs:259:13
  33:     0x559995952059 - std::panicking::try::do_call::hc5fcacb7a85fc7b1
                               at /rustc/a178d0322ce20e33eac124758e837cbd80a6f633/library/std/src/panicking.rs:401:40
  34:     0x559995952059 - std::panicking::try::hb5d9603af3abbe3a
                               at /rustc/a178d0322ce20e33eac124758e837cbd80a6f633/library/std/src/panicking.rs:365:19
  35:     0x559995952059 - std::panic::catch_unwind::h98fe6ac3925e64b4
                               at /rustc/a178d0322ce20e33eac124758e837cbd80a6f633/library/std/src/panic.rs:434:14
  36:     0x559995952059 - std::rt::lang_start_internal::h22ac7383c516f93e
                               at /rustc/a178d0322ce20e33eac124758e837cbd80a6f633/library/std/src/rt.rs:34:21
  37:     0x559995798690 - std::rt::lang_start::h3ba78dae42f73555
                               at /rustc/a178d0322ce20e33eac124758e837cbd80a6f633/library/std/src/rt.rs:48:5
  38:     0x559995749fdc - main
  39:     0x7f1326609d0a - __libc_start_main
  40:     0x5599956cd9ca - _start
  41:                0x0 - <unknown>
```

maximbaz, 2021-08-25:
Welcome to the club, he-he 😄 I've updated the README about this issue, hopefully this will bring some interest to get help on this, as my head is already overwhelmed by working with GPU. 

https://github.com/maximbaz/wluma/commit/b733d9be99f49748400e135cf4af10d305c84bbf

Workaround is to `export WLR_DRM_NO_MODIFIERS=1` before you start your compositor (e.g. sway). This might increase CPU usage a bit, so adding a proper support is definitely wanted, but I've been using this workaround for many months and it's not that bad...

maximbaz, 2022-05-29:
As this becomes a popular request, leaving a few links in case a brave soul decides to help out:

- https://www.khronos.org/registry/vulkan/specs/1.3-extensions/man/html/VK_EXT_image_drm_format_modifier.html#_usage_patterns (Import part)
- https://github.com/FFmpeg/FFmpeg/blob/a7e099732405c5f048fb306d8e3c1dffe0d60e41/libavutil/hwcontext_vulkan.c#L2642 (most promising code example I think -> make sure to switch to latest `master`)
- https://github.com/nowrep/obs-vkcapture/blob/master/src/vklayer.c#L548
- https://github.com/nyorain/kms-vulkan/blob/master/vulkan.c#L975

If anyone does start working on it, please leave a ping, I'll do my best to help out 🙂