max-baz/wluma #10: webcam as an alternative light sensor

manorit2001, 2022-01-03:
It would be nice if this can also work with laptops that don't have a light sensor, Similar to what https://github.com/FedeDP/Clight does ( using webcam to predict the lighting )

maximbaz, 2022-01-03:
Sounds good, we even had @cyrinux implement it in the past, when it was a C project with python extensions 😜 

https://github.com/cyrinux/wluma-als-emulator/blob/master/wluma_als_emulator/strategies/webcam_strategy.py

We both really ended up using time based ALS simulation on laptops without ALS, but there is definitely room for webcam based simulation if you or someone else wants.

cyrinux, 2022-01-03:
I will *try* to port my code to rust and open a open ;-)

cyrinux, 2022-01-07:
Hi @manorit2001 ,

We implement the new `als.webcam` in https://github.com/maximbaz/wluma/pull/13 .

You can check the README.md if you need more details about the configuration of this one, but basically you need to enable the new ALS, and so disable the others:

```
[als.webcam]
video = 0
thresholds = [ 15, 30, 45, 60, 75 ]
```

Can you please try it and give us a feedback ?

manorit2001, 2022-01-09:
Hi, I am getting this error while running. Can you help me out
```
$ RUST_BACKTRACE=full sudo -E wluma                                                                                                                                                [14:25:30]
Using config: Config { frame: Frame { capturer: Wlroots, processor: Vulkan }, als: Webcam { video: 0, thresholds: [15, 30, 45, 60, 75] }, output_by_type: OutputByType { backlight: {}, ddcutil: {} }, keyboard: None, output: {"eDP-1": Backlight(BacklightOutput { path: "/sys/class/backlight/intel_backlight", use_contents: false })} }
thread 'main' panicked at 'Unable to initialize Vulkan: ERROR_EXTENSION_NOT_PRESENT', src/main.rs:38:56
stack backtrace:
   0:     0x563775e60f7c - <unknown>
   1:     0x563775e82b6c - <unknown>
   2:     0x563775e5dce5 - <unknown>
   3:     0x563775e62b00 - <unknown>
   4:     0x563775e626b5 - <unknown>
   5:     0x563775e631b4 - <unknown>
   6:     0x563775e62c90 - <unknown>
   7:     0x563775e61424 - <unknown>
   8:     0x563775e62bf9 - <unknown>
   9:     0x563775daa0e1 - <unknown>
  10:     0x563775daa1d3 - <unknown>
  11:     0x563775dde418 - <unknown>
  12:     0x563775dd84a3 - <unknown>
  13:     0x563775dedfa9 - <unknown>
  14:     0x563775e60c5b - <unknown>
  15:     0x563775dde7b2 - <unknown>
  16:     0x7f085a66ab25 - __libc_start_main
  17:     0x563775daa89e - <unknown>
  18:                0x0 - <unknown>

```

manorit2001, 2022-01-09:
I installed my vulkan drivers for intel ( assuming that the AUR package didn't have updated dependencies ). Now I am getting this error
```
$ sudo -E wluma                                                                                                                                                [14:31:08]
Using config: Config { frame: Frame { capturer: Wlroots, processor: Vulkan }, als: Webcam { video: 0, thresholds: [15, 30, 45, 60, 75] }, output_by_type: OutputByType { backlight: {}, ddcutil: {} }, keyboard: None, output: {"eDP-1": Backlight(BacklightOutput { path: "/sys/class/backlight/intel_backlight", use_contents: false })} }
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: NoCompositorListening', src/frame/capturer/wlroots.rs:39:56

```

cyrinux, 2022-01-09:
Hi @manorit2001 ,

I was glade to train my rust with your request ;-)

Are you well using wayland and not X11 ?

manorit2001, 2022-01-09:
> I think @maximbaz will prefer we open another issue if you got the first error while using wayland.
> 
> According to the `Unable to initialize Vulkan: ERROR_EXTENSION_NOT_PRESENT` error, I can guess you are maybe not using a wayland session but X11 ? `wlroots` is a compositor library for wayland only. In this case we got this issue #5 about the same problem.

Yes, you guessed it right. I am using `X11`. I saw that `wlroots` was the default capturer in the configs so I tried installing and using it when I got that error. Does X11 have some different capturer?

> Also, you don't need to use sudo but better, add you in video group.

Oh, will do so

> Can you give use more informations about your distro and session used and maybe ?

I am using xfce under X11. My distro is Manjaro Linux with kernel `5.15.12-1-MANJARO #1`

cyrinux, 2022-01-09:
> 

Ok well, yes, for the moment, until issue #5 only wayland is supported. Not needed to open another issue.
The main stuff for the capturer `wlrtools` is related to wayland and so there is still some works to make this working with X11 but it's not a priority TBH.

With archlinux I can read you can easily make use of wayland with sway, gnome or KDE, XFCE don't support wayland also. If you are not married with XFCE, you can maybe give a try ? Good luck 😉

manorit2001, 2022-01-09:
Unfortunately, the commitment with XFCE has been way too long for me ;) . Thanks for suggesting my divorce though, will think of it when I start having fights with it.

closed 2022-01-09 by manorit2001