max-baz/wluma #38: [Bug]: Panick if needed Vulkan extensions are not available

Mynacol, 2022-01-27:
### Steps for reproducing the issue

I have an old Nvidia GPU running with nouveau, which does not provide Vulkan. Instead, a probably not very conformant software implementation, lavapipe (part of mesa), is available and used.

### What is the buggy behavior?

Shortly after launching wluma it panickes/crashes (see logs).

I think by moving the initialization of the `frame_processor` in `main.rs` line 70 to the case when the capturer config is Wlroots, no Vulkan is required/initialized if no capturer is used, fixing my problem.

### What is the expected behavior?

As I set `capturer = "none"` on all outputs in the config, no Vulkan is required and wluma doesn't choke on missing features or other errors regarding Vulkan.

### Logs

```shell
$ RUST_LOG=trace wluma
[2022-01-27T12:12:51Z DEBUG wluma] Using Config {
        als: Time {
            thresholds: {
                7: "dark",
                11: "normal",
                0: "night",
                9: "dim",
                18: "dark",
                13: "bright",
                16: "normal",
                20: "night",
            },
        },
        output: [
            DdcUtil(
                DdcUtilOutput {
                    name: "DELL 1908FP",
                    capturer: None,
                    min_brightness: 1,
                },
            ),
            DdcUtil(
                DdcUtilOutput {
                    name: "W2443",
                    capturer: None,
                    min_brightness: 1,
                },
            ),
        ],
    }
[2022-01-27T12:12:52Z TRACE ddc_hi] DisplayInfo::from_edid(I2cDevice, 22785)
[2022-01-27T12:12:52Z TRACE ddc_hi] DisplayInfo::from_edid(I2cDevice, 22789)
[2022-01-27T12:12:53Z TRACE ddc_hi] DisplayInfo::from_capabilities(I2cDevice, 22785)
[2022-01-27T12:12:54Z DEBUG wluma::brightness::ddcutil] Discovered displays: ["W2443 "]
[2022-01-27T12:12:54Z WARN  wluma] Skipping 'DELL 1908FP' as it might be disconnected: Unable to find display
[2022-01-27T12:12:54Z TRACE ddc_hi] DisplayInfo::from_edid(I2cDevice, 22785)
[2022-01-27T12:12:54Z TRACE ddc_hi] DisplayInfo::from_edid(I2cDevice, 22789)
[2022-01-27T12:12:55Z TRACE ddc_hi] DisplayInfo::from_capabilities(I2cDevice, 22785)
[2022-01-27T12:12:56Z DEBUG wluma::brightness::ddcutil] Discovered displays: ["W2443 "]
[2022-01-27T12:12:56Z DEBUG wluma::brightness::ddcutil] Using display 'W2443 ' for config 'W2443'
[2022-01-27T12:12:56Z INFO  wluma] Continue adjusting brightness and wluma will learn your preference over time.
[2022-01-27T12:12:56Z TRACE wluma::als::time] ALS (time): bright (13)
thread 'predictor-W2443' panicked at 'Unable to initialize Vulkan: ERROR_EXTENSION_NOT_PRESENT', src/main.rs:72:38
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace

$ vulkaninfo
ERROR: [Loader Message] Code 0 : loader_scanned_icd_add: Driver /usr/lib/libvulkan_intel.so supports Vulkan 1.2, but only supports loader interface version 4. Interface version 5 or newer required to support this version of Vulkan (Policy #LDP_DRIVER_7)
ERROR: [Loader Message] Code 0 : loader_scanned_icd_add: Driver /usr/lib/libvulkan_lvp.so supports Vulkan 1.1, but only supports loader interface version 4. Interface version 5 or newer required to support this version of Vulkan (Policy #LDP_DRIVER_7)
ERROR: [Loader Message] Code 0 : loader_scanned_icd_add: Driver /usr/lib/libvulkan_radeon.so supports Vulkan 1.2, but only supports loader interface version 4. Interface version 5 or newer required to support this version of Vulkan (Policy #LDP_DRIVER_7)
WARNING: [Loader Message] Code 0 : terminator_CreateInstance: Failed to CreateInstance in ICD 3.  Skipping ICD.
WARNING: lavapipe is not a conformant vulkan implementation, testing use only.
==========
[... 996 lines cut ...]
### Version

wluma 4.1.0-1 from AUR

### Environment

```shell
- Arch Linux
- development mesa-git version 22.0 with a MR containing fixes for nouveau
- sway

wluma config:
[als.time]
thresholds = { 0 = "night", 7 = "dark", 9 = "dim", 11 = "normal", 13 = "bright", 16 = "normal", 18 = "dark", 20 = "night" }

[[output.ddcutil]]
name = "DELL 1908FP"
min_brightness = 0
capturer = "none"

[[output.ddcutil]]
name = "W2443"
min_brightness = 0
capturer = "none"
```

maximbaz, 2022-01-27:
Thank you very much for reporting, we shall indeed do as you suggest and not initialize Vulkan if it's not needed.

If you feel like sending a PR, please feel free to, otherwise we'll get it fixed tonight.

Mynacol, 2022-01-27:
Wow, that was fast. No Rust expert, so you are probably better off doing the commit yourself.

Thanks for this wonderful project!

maximbaz, 2022-01-27:
Fixed in https://github.com/maximbaz/wluma/commit/3901acc75a25f9db8aae41512ca26a74fec2b5b2

The diff may look large but what I did was simply let wlroots itself initialize Vulkan, instead of it being passed from outside - it was done like that because initially I had plans to support multiple variations of GPU calculations (I even have non-vulkan implementation somewhere in the history of this repo, back when it was written in C), but it was very complicated to write, buggy and not pleasant, so I later scraped the idea 😅

maximbaz, 2022-02-07:
Sorry for the delay, 4.1.1 is released, let me know if something is still not working as expected!

closed 2022-02-07 by maximbaz

maximbaz, 2022-02-07:
And 4.1.2 will now not require you to install `vulkan-driver` at all as a dependency 🙂

Mynacol, 2022-02-13:
Thanks @maximbaz for the follow up, wluma does not crash on my desktop now. I have another problem and wrote about it in the new issue #46 :)