max-baz/wluma #43: [Feature request]: Replace unmaintained v4l crate with some maintained alternative

jirutka, 2022-02-06:
### Please describe your feature request

Crate [v4l](https://crates.io/crates/v4l) from [raymanfx/libv4l-rs](https://github.com/raymanfx/libv4l-rs) is buggy ([#43](https://github.com/raymanfx/libv4l-rs/pull/43), [#16](https://github.com/raymanfx/libv4l-rs/issues/16)) and seems to be unmaintained (see opened [pull requests](https://github.com/raymanfx/libv4l-rs/pulls)).

It fails to build on Alpine Linux (musl libc) due to wrong types, I had to patch it (which is a bit tricky for rust dependencies). I managed to build it on 64-bit architectures (and published Alpine package), but there are more problems on 32-bit architectures (armv7).

raymanfx, 2022-02-17:
Hi, `libv4l-rs` author here. I'd be happy to review patches to improve the `armv7` and `musl` support - would you mind opening PRs so we can go through the changes together? Other issues can be sorted out as well I'm sure.

MarijnS95, 2022-02-17:
On that note it does not seem like there are any alternatives to `libv4l-rs` at all, not to mention a "better" one: https://crates.io/search?q=v4l

Glad you're back by the way, @raymanfx!

jirutka, 2022-02-17:
> I'd be happy to review patches to improve the armv7

There are many problems with types on 32-bit arches (not just armv7). I’m sorry, but I don’t care enough about 32-bit architectures to invest my time in it. Maybe @PureTryOut from postmartketOS will be interested.

raymanfx, 2022-02-17:
Quick heads-up: I am actively working on armv7 support in my crate as we speak and expect to fix these issues myself before long. The `next` branch [[1](https://github.com/raymanfx/libv4l-rs/tree/next)] compiles fine for 32 bit triplets like `armv7-unknown-linux-musleabihf` right now and I expect to publish an update to crates.io soon.

maximbaz, 2022-02-20:
Sounds like the issue is resolved, the library is proven to be maintained, issues are being worked on, and we don't know of any better alternative 😉 Thanks to everyone involved, closing!

closed 2022-02-20 by maximbaz