max-baz/wluma #44: [Bug]: als.webcam panics on ThinkPad T14s AMD with integrated camera

jirutka, 2022-02-07:
### Steps for reproducing the issue

I’m afraid that this will be some device-specific issue. I’ll test it with external webcam later.

### What is the buggy behavior?

wluma crashes when `als.webcam` is enabled.

### What is the expected behavior?

wluma shouldn’t crash

### Logs

```shell
[2022-02-06T23:54:19Z DEBUG wluma] Using Config {
        als: Webcam {
            video: 0,
            thresholds: {
                60: "bright",
                45: "normal",
                75: "outdoors",
                15: "dark",
                30: "dim",
                0: "night",
            },
        },
        output: [
            Backlight(
                BacklightOutput {
                    name: "eDP-1",
                    path: "/sys/class/backlight/amdgpu_bl0",
                    capturer: None,
                    min_brightness: 1,
                },
            ),
            Backlight(
                BacklightOutput {
                    name: "keyboard-thinkpad",
                    path: "/sys/bus/platform/devices/thinkpad_acpi/leds/tpacpi::kbd_backlight",
                    capturer: None,
                    min_brightness: 0,
                },
            ),
        ],
    }
[2022-02-06T23:54:19Z INFO  wluma] Continue adjusting brightness and wluma will learn your preference over time.
thread 'als-webcam' panicked at 'called `Option::unwrap()` on a `None` value', src/frame/mod.rs:16:35
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

### Version

4.1.0

### Environment

```shell
$ lsb_release -a; uname -a; pgrep -l sway; pacman -Q | egrep "(wlroots|vulkan|sway|clang|rust)"; dpkg -l | egrep "(wlroots|vulkan|sway|clang|rust)"
/bin/sh: lsb_release: not found
Linux tron 5.15.16-0-lts #1-Alpine SMP Wed, 26 Jan 2022 16:46:12 +0000 x86_64 Linux
3587 sway
3634 swayidle
3647 swayrd
4753 swaybg
/bin/sh: pacman: not found
/bin/sh: dpkg: not found

-----------------------------------------------------------
$ v4l2-ctl --all
[... 178 lines cut ...]
	test VIDIOC_G_SLICED_VBI_CAP: OK (Not Supported)
	test Cropping: OK (Not Supported)
	test Composing: OK (Not Supported)
	test Scaling: OK (Not Supported)

Codec ioctls (Input 0):
	test VIDIOC_(TRY_)ENCODER_CMD: OK (Not Supported)
	test VIDIOC_G_ENC_INDEX: OK (Not Supported)
	test VIDIOC_(TRY_)DECODER_CMD: OK (Not Supported)

Buffer ioctls (Input 0):
	test VIDIOC_REQBUFS/CREATE_BUFS/QUERYBUF: OK
	test VIDIOC_EXPBUF: OK
	test Requests: OK (Not Supported)

Total for uvcvideo device /dev/video0: 46, Succeeded: 45, Failed: 1, Warnings: 7

-----------------------------------------------------------
- Alpine Linux v3.15
- Sway 1.6.1
- wlroots 0.14.1
- Vulkan 1.3.204
- Rust 1.56.1
- ThinkPad T14s AMD with integrated AMD graphic
```

jirutka, 2022-02-07:
I just tested it with external USB webcam and it sorta works – doesn’t crash, responds to light, but a little weird. Does wluma take into account the exposure adjustment in the camera?

maximbaz, 2022-02-07:
The crash was likely fixed by https://github.com/maximbaz/wluma/commit/e47933fa4193259f8d91c1ffdb8906fa9bd6bc25, I'll make a patch release today, hope that will fix it.

> Does wluma take into account the exposure adjustment in the camera?

Not really, we couldn't find a way using `v4l` to disable auto-exposure or get any way into correcting for it - so just getting a raw image "as is" and processing it to detect how bright it is. As mentioned in https://github.com/maximbaz/wluma/pull/42#issuecomment-1030936528 please do share if you already have some suggestions what we can replace v4l crate with, but no worries in you don't know any, we'll have a look 😉

maximbaz, 2022-02-07:
> This notebook has even IR camera, I’m curious if it can be used as an ambient light sensor…

Oh that's a curious idea! Feel free to open a separate ticket and we can poke around together to see what can be done, happy to explain in more details what happens today with regular webcam, or why things are done the way they are. 

> v4l exposes this setting, so maybe it’s just not included in the rust bindings

Ok, good to know! FYI the plan is to try https://crates.io/crates/nokhwa, it seems interesting and maintained. Don't know yet how easy it is to compile with musl though.