max-baz/wluma #46: [Bug]: Monitors with missing Model / MCCS version fields are not detected

Mynacol, 2022-02-13:
### Steps for reproducing the issue

Having a bad monitor with unspecified Model and MCCS version :/

### What is the buggy behavior?

Wluma does not detect one of my two monitors. See logs.

### What is the expected behavior?

Even with broken/missing information, this monitor is able to change brightness and other parameters over DDC. Wluma should be able to support those partially broken monitors as well.

### Logs

```shell
$ RUST_LOG=debug wluma
[2022-02-13T19:48:56Z DEBUG wluma] Using Config {
        als: Time {
            thresholds: {
                11: "normal",
                0: "night",
                7: "dark",
                16: "normal",
                20: "night",
                9: "dim",
                18: "dark",
                13: "bright",
            },
        },
        output: [
            DdcUtil(
                DdcUtilOutput {
                    name: "1908FP",
                    capturer: None,
                    min_brightness: 1,
                },
            ),
            DdcUtil(
                DdcUtilOutput {
                    name: "W2443",
                    capturer: None,
                    min_brightness: 1,
                },
            ),
        ],
    }
[2022-02-13T19:48:58Z DEBUG wluma::brightness::ddcutil] Discovered displays: ["W2443 "]
[2022-02-13T19:48:58Z WARN  wluma] Skipping '1908FP' as it might be disconnected: Unable to find display
[2022-02-13T19:49:00Z DEBUG wluma::brightness::ddcutil] Discovered displays: ["W2443 "]
[2022-02-13T19:49:00Z DEBUG wluma::brightness::ddcutil] Using display 'W2443 ' for config 'W2443'
[2022-02-13T19:49:00Z INFO  wluma] Continue adjusting brightness and wluma will learn your preference over time.
```

### Version

wluma 4.1.2-1 from AUR

### Environment

```shell
Still sway on Arch Linux.

Monitor information from ddcutil:
$ ddcutil detect
Display 2
   I2C bus:  /dev/i2c-5
   EDID synopsis:
      Mfg id:               DEL
      Model:                DELL 1908FP
      Product code:         16422
[... 304 lines cut ...]
VCP code 0xfe (Manufacturer Specific         ): Unsupported feature code
VCP code 0xff (Manufacturer Specific         ): mh=0xff, ml=0xff, sh=0x00, sl=0x00

Comparing declared capabilities to observed features...

Readable features declared in capabilities string: 02 0b 0c 0e 10 12 14 16 18 1a 1e 20 30 3e 52 6c 6e 70 ac ae b2 b6 c0 c8 c9 ca d6 df

MCCS (VCP) version reported by capabilities: Unqueried
MCCS (VCP) version reported by feature 0xDf: 2.0
Versions do not match!!!

All readable features declared in capabilities were found by scanning.

Features found by scanning but not declared as capabilities:
   Feature x60 - Input Source
   Feature x68 - Unknown feature
   Feature x9b - 6 axis hue control: Red
   Feature x9d - 6 axis hue control: Green
   Feature x9f - 6 axis hue control: Blue
   Feature xff - Manufacturer Specific

Color temperature increment (x0b) = 184 degrees Kelvin
Color temperature request   (x0c) = 2
Requested color temperature = (3000 deg Kelvin) + 2 * (184 degrees Kelvin) = 3368 degrees Kelvin
```

maximbaz, 2022-02-13:
Could you please add the following log:

```rust

    ddc_hi::Display::enumerate()
        .into_iter()
        .for_each(|mut display| {
            log::debug!(
                "update_capabilities result: {:?}",
                display.update_capabilities()
            );
        });
```

before this line:

https://github.com/maximbaz/wluma/blob/0c49755fdcf19b22841844df6508f29c31975ce2/src/brightness/ddcutil.rs#L82

and show the output here?

Mynacol, 2022-02-13:
The full log is now:
```
[2022-02-13T21:24:12Z DEBUG wluma] Using Config {
        als: Time {
            thresholds: {
                7: "dark",
                20: "night",
                16: "normal",
                9: "dim",
                0: "night",
                13: "bright",
                18: "dark",
                11: "normal",
            },
        },
        output: [
            DdcUtil(
                DdcUtilOutput {
                    name: "1908FP",
                    capturer: None,
                    min_brightness: 1,
                },
            ),
            DdcUtil(
                DdcUtilOutput {
                    name: "W2443",
                    capturer: None,
                    min_brightness: 1,
                },
            ),
        ],
    }
[2022-02-13T21:24:15Z DEBUG wluma::brightness::ddcutil] update_capabilities result: Ok(())
[2022-02-13T21:24:16Z DEBUG wluma::brightness::ddcutil] update_capabilities result: Err(Failed to parse MCCS capabilities

    Caused by:
        [255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 97, 115, 115, 101, 116, 95, 101, 101, 112, 40, 52, 48, 41, 0, 255, 255, 109, 99, 99, 115, 95, 118, 101, 114, 40, 50, 46, 48, 41, 0, 255, 255, 9, 0, 0, 22, 161, 30, 0, 0, 23, 161, 30, 30, 0, 23, 161, 30, 60, 97, 115, 115, 101, 116, 95, 101, 101, 112, 40, 52, 48, 41, 109, 99, 99, 115, 95, 118, 101, 114, 40, 50, 46, 48, 41]:Eof)
[2022-02-13T21:24:16Z DEBUG wluma::brightness::ddcutil] Discovered displays: ["W2443 "]
[2022-02-13T21:24:16Z WARN  wluma] Skipping '1908FP' as it might be disconnected: Unable to find display
[2022-02-13T21:24:20Z DEBUG wluma::brightness::ddcutil] update_capabilities result: Ok(())
[2022-02-13T21:24:21Z DEBUG wluma::brightness::ddcutil] update_capabilities result: Err(Failed to parse MCCS capabilities

    Caused by:
        [255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 97, 115, 115, 101, 116, 95, 101, 101, 112, 40, 52, 48, 41, 0, 255, 255, 109, 99, 99, 115, 95, 118, 101, 114, 40, 50, 46, 48, 41, 0, 255, 255, 9, 0, 0, 22, 161, 30, 0, 0, 23, 161, 30, 30, 0, 23, 161, 30, 60, 97, 115, 115, 101, 116, 95, 101, 101, 112, 40, 52, 48, 41, 109, 99, 99, 115, 95, 118, 101, 114, 40, 50, 46, 48, 41]:Eof)
[2022-02-13T21:24:21Z DEBUG wluma::brightness::ddcutil] Discovered displays: ["W2443 "]
[2022-02-13T21:24:21Z DEBUG wluma::brightness::ddcutil] Using display 'W2443 ' for config 'W2443'
[2022-02-13T21:24:21Z INFO  wluma] Continue adjusting brightness and wluma will learn your preference over time.
```

maximbaz, 2022-02-13:
Cool, thanks! 

Could you please try `ddcutil-add-id` branch, and provide debug logs once more if it still doesn't work? It should hopefully allow you to also use DDC ID field (whatever that is, you will see what wluma detects in debug logs), if getting name and models fails, instead of skipping that monitor as before. 

> update_capabilities result: Err(Failed to parse MCCS capabilities

This is what I was afraid of... In order to populate the name and model, so that we can use it for matching, we call `update_capabilities` from the library, which fills this info, and it fails for one of your monitors.

The library also has `update_from_ddc` function, but that one only fills in `mccs` version and database, not any of the identifying information about the monitor which we could use to match against....

However it looks like ID field is always present. So in the branch I added it to the string that you can match in your config, and ignoring the error from `update_capabilities` function.

This could be problematic though, @cyrinux and I saw some ghost entries (the same monitor would be listed twice, and only one of them was actually usable, the one for which `update_capabilities` function succeeded), so I'm not sure yet if this would be a viable fix to merge in the current form, even if it solves your problem...

Mynacol, 2022-02-14:
Thanks! This solves the detection problem for me.
I modified your change and the ID is not even needed! I guess ddcutil throws an error, so the `ok()` function stops the handling of this monitor. If you ignore the result value, the model name and serial number are available.

The relevant section is now:
```rust
    let displays = ddc_hi::Display::enumerate()
        .into_iter()
        .filter_map(|mut display| {
            let _ignored = display.update_capabilities();
            let empty = "".to_string();
            let merged = format!(
                "{} {}",
                display.info.model_name.as_ref().unwrap_or(&empty),
                display.info.serial_number.as_ref().unwrap_or(&empty)
            );
            Some((merged, display))
        })
        .collect_vec();
```

And the debug log is now:
```
[2022-02-14T12:48:21Z DEBUG wluma] Using Config {
        als: Time {
            thresholds: {
                18: "dark",
                9: "dim",
                0: "night",
                16: "normal",
                7: "dark",
                11: "normal",
                13: "bright",
                20: "night",
            },
        },
        output: [
            DdcUtil(
                DdcUtilOutput {
                    name: "1908FP",
                    capturer: None,
                    min_brightness: 1,
                },
            ),
            DdcUtil(
                DdcUtilOutput {
                    name: "W2443",
                    capturer: None,
                    min_brightness: 1,
                },
            ),
        ],
    }
[2022-02-14T12:48:23Z DEBUG wluma::brightness::ddcutil] Discovered displays: ["W2443 ", "DELL 1908FP G314H86D7UFA"]
[2022-02-14T12:48:23Z DEBUG wluma::brightness::ddcutil] Using display 'DELL 1908FP G314H86D7UFA' for config '1908FP'
[2022-02-14T12:48:25Z DEBUG wluma::brightness::ddcutil] Discovered displays: ["W2443 ", "DELL 1908FP G314H86D7UFA"]
[2022-02-14T12:48:25Z DEBUG wluma::brightness::ddcutil] Using display 'W2443 ' for config 'W2443'
[2022-02-14T12:48:25Z INFO  wluma] Continue adjusting brightness and wluma will learn your preference over time.
```

Maybe you can write up some smart detection if `update_capabilities` fails, but the values are still available?
This might be a cleaner approach, which is better to merge and maintain.
I did not experience duplicate monitor entries.

Another idea: I believe `ddcutil detect` gets monitor info from EDID, so you could get manufacturer/model/serial nr. from there.

The next problem I experience: I2C errors. Wluma spam-reports `[ERROR wluma::brightness::controller] Unable to get brightness value: DDC/CI I2C error: No such device or address (os error 6)`, the kernel log contains some `kernel: i2c i2c-0: sendbytes: NAK bailout.` i2c-0 is the other monitor, so not related with this issue.

maximbaz, 2022-02-14:
Cool, thanks for confirming!

I think we'll have to keep IDs because this is what I'm getting when only *one* external monitor is connected, if we don't ignore `update_capabilities` errors (as in main branch):

```
Discovered displays: ["22794 DELL P2317H 9KFDW67326GB", "22793  ",  "22796 DELL P2317H 9KFDW67326GB"]
```

As you can see it's actually generating 3 entries, one of which has no info (and failed `update_capabilities` run), two others have exactly the same make and model, but different ID, and only for one of them the `update_capabilities` succeeded.

If wluma picks the wrong one in the list above, then it will not be able to control brightness...

However before thinking how to solve it, do I understand it correctly that wluma is still unable to get/set brightness for your screen? In other words, I2C errors are not just fake errors, but actually screen is not being managed by wluma? If that's the case, it sounds like the library we are using is simply unable to deal with weird monitors, for which `update_capabilities` fails...

willbou1, 2022-02-24:
diff --git a/src/brightness/ddcutil.rs b/src/brightness/ddcutil.rs
index 09a5108..eb8fd77 100644
--- a/src/brightness/ddcutil.rs
+++ b/src/brightness/ddcutil.rs
@@ -69,10 +69,12 @@ fn find_display_by_name(name: &str) -> Option<Display> {
         .filter_map(|mut display| {
             display.update_capabilities().ok().map(|_| {
                 let empty = "".to_string();
+                let empty_int = 0;
                 let merged = format!(
-                    "{} {}",
+                    "{} {} {}",
                     display.info.model_name.as_ref().unwrap_or(&empty),
-                    display.info.serial_number.as_ref().unwrap_or(&empty)
+                    display.info.serial_number.as_ref().unwrap_or(&empty),
+                    display.info.serial.as_ref().unwrap_or(&empty_int)
                 );
                 (merged, display)
             })

I'm not a rust programmer, but this works for me :)

maximbaz, 2022-02-24:
Could you explain some more, what was your issue and how does the patch help? I'm still at a loss whether it's possible for wluma to control brightness of these "weird" monitors or the patches are just to hide the errors...

CorvetteCole, 2023-03-07:
It would be great if we could have a manual override for the i2c address to get around this. For example, I have two identical monitors which actually have identical EDID data. Running ddcutil detect results in:
```
Display 1
   I2C bus:  /dev/i2c-6
   DRM connector:           card0-DP-1
   EDID synopsis:
      Mfg id:               ACR - Acer Technologies
      Model:                ED273UR
      Product code:         1682  (0x0692)
      Serial number:        
      Binary serial number: 0 (0x00000000)
      Manufacture year:     2018,  Week: 17
   VCP version:         2.1

Display 2
   I2C bus:  /dev/i2c-7
   DRM connector:           card0-DP-2
   EDID synopsis:
      Mfg id:               ACR - Acer Technologies
      Model:                ED273UR
      Product code:         1682  (0x0692)
      Serial number:        
      Binary serial number: 0 (0x00000000)
      Manufacture year:     2018,  Week: 17
   VCP version:         2.1
```
But wluma only sees:
```
[2023-03-07T19:16:56Z DEBUG wluma::brightness::ddcutil] Discovered displays: ["ED273UR ", "ED273UR "]
```

This makes it impossible to initialize both displays or even one of them.

maximbaz, 2024-10-21:
Closing due to lack of activity, please leave a comment or open a new issue if this is still relevant.

closed 2024-10-21 by maximbaz