max-baz/wluma #48: [Feature request]: Allow specifying the physical device for Vulkan

ilkecan, 2022-02-28:
### Please describe your feature request

What do you think about adding an option to specify the physical device for Vulkan? I have two on my system and the first one doesn't have the required extensions, so Vulkan can't initialize. It works fine if I change the code to use the second one.

https://github.com/maximbaz/wluma/blob/0c49755fdcf19b22841844df6508f29c31975ce2/src/frame/vulkan.rs#L55-L58

maximbaz, 2022-02-28:
Thanks, truth be told I cut corners, hardcoded `0` and was waiting until someone asked for this, as I don't have a way to test it 😁 

The correct approach is to query device capabilities and select the best device that supports everything we need.

Some examples:

- [inspiration / general concept](https://vulkan-tutorial.com/Drawing_a_triangle/Setup/Physical_devices_and_queue_families#page_Base-device-suitability-checks)
- [ash code example, probably we need something very similar](https://github.com/MaikKlein/ash/blob/3c7615b3bb11099e982e5248228c2b77aca17352/examples/src/lib.rs#L276-L300)

If you would like to experiment and make a PR, let me know, it's very appreciated!

ilkecan, 2022-02-28:
I will look into this. Thanks for the pointers.