max-baz/wluma #51: [Bug]: wlroots capture shows NoCompositorListening, Sway on Void Linux, from WezTerm terminal

JJGadgets, 2022-05-28:
### Steps for reproducing the issue

Hardware: ThinkPad T14 Gen 1 (**AMD**), Chicony webcam, AMDGPU
Software: Void Linux, SwayWM, `wluma` run from WezTerm

1. Compile from source (`make build && sudo make install`)
2. Run `wluma`
3. Error log:

```
❯ RUST_BACKTRACE=full wluma
thread 'predictor-eDP-1' panicked at 'called `Result::unwrap()` on an `Err` value: NoCompositorListening', src/frame/capturer/wlroots.rs:70:49
stack backtrace:
[2022-05-28T20:45:08Z INFO  wluma] Continue adjusting brightness and wluma will learn your preference over time.
   0:     0x55e2748c66b0 - <std::sys_common::backtrace::_print::DisplayBacktrace as core::fmt::Display>::fmt::h18674753585b8cc5
   1:     0x55e2748e916c - core::fmt::write::h650970577346edc1
   2:     0x55e2748c2fb5 - std::io::Write::write_fmt::hcdbe4458fe2ebdfb
   3:     0x55e2748c84cb - std::panicking::default_hook::{{closure}}::h2b3ca2e1a25274db
   4:     0x55e2748c8043 - std::panicking::default_hook::hb3d80776b693aaeb
   5:     0x55e27475b33e - wluma::main::{{closure}}::hac5ad5b4efe56de1
   6:     0x55e2748c8bc9 - std::panicking::rust_panic_with_hook::habc6079310c0728a
   7:     0x55e2748c8670 - std::panicking::begin_panic_handler::{{closure}}::ha449aee990d62948
   8:     0x55e2748c6b54 - std::sys_common::backtrace::__rust_end_short_backtrace::h54cc540f2a5a6bf2
   9:     0x55e2748c85d9 - rust_begin_unwind
  10:     0x55e274735231 - core::panicking::panic_fmt::he85288327cd30385
  11:     0x55e274735323 - core::result::unwrap_failed::ha180eafd08eaf142
  12:     0x55e27475f857 - <wluma::frame::capturer::wlroots::Capturer as core::default::Default>::default::hf2d4c68db8240d60
  13:     0x55e274760d9b - std::sys_common::backtrace::__rust_begin_short_backtrace::h1b6ace0267256e64
  14:     0x55e274784a58 - core::ops::function::FnOnce::call_once{{vtable.shim}}::h2d19e92af92472d4
  15:     0x55e2748cbf43 - std::sys::unix::thread::Thread::new::thread_start::h2582dd3e3de2d3c5
  16:     0x7f45fa2f3eae - start_thread
                               at /builddir/glibc-2.32/nptl/pthread_create.c:463:8
  17:     0x7f45fa0de2ff - __GI___clone
                               at /builddir/glibc-2.32/misc/../sysdeps/unix/sysv/linux/x86_64/clone.S:95
  18:                0x0 - <unknown>
```

### What is the buggy behavior?

wlroots capture panics `wluma` on launch, with error "NoCompositorListening".

### What is the expected behavior?

I can use wlroots capture without errors.

### Logs

```shell
[2022-05-28T20:47:12Z DEBUG wluma] Using Config {
        als: Webcam {
            video: 0,
            thresholds: {
                60: "bright",
                45: "normal",
                0: "night",
                30: "dim",
                15: "dark",
                75: "outdoors",
            },
        },
        output: [
            Backlight(
                BacklightOutput {
                    name: "eDP-1",
                    path: "/sys/class/backlight/amdgpu_bl0",
                    capturer: Wlroots,
                    min_brightness: 1,
                },
            ),
            Backlight(
                BacklightOutput {
                    name: "keyboard-thinkpad",
                    path: "/sys/bus/platform/devices/thinkpad_acpi/leds/tpacpi::kbd_backlight",
                    capturer: None,
                    min_brightness: 0,
                },
            ),
        ],
    }
thread 'predictor-eDP-1' panicked at 'called `Result::unwrap()` on an `Err` value: NoCompositorListening', src/frame/capturer/wlroots.rs:70:49
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

### Version

Latest commit from master, which should also mean 4.1.2?

### Environment

```shell
Hardware: ThinkPad T14 Gen 1 (**AMD**), Chicony webcam, AMDGPU
Software: Void Linux, SwayWM, `wluma` run from WezTerm

`lsb_release -a; uname -a; pgrep -l sway;`

LSB Version:    1.0
Distributor ID: VoidLinux
Description:    Void Linux
Release:        rolling
Codename:       void
Linux [HOSTNAME REDACTED] #1 SMP 1653511930 x86_64 GNU/Linux
12018 swayidle
12213 swaybg
12218 swaybar
20880 sway

`xbps-query -l | egrep "(wlroots|vulkan|sway|clang|rust)"`

ii autotiling-1.6_1                                            Automatically switch horizontal/vertical split orientation in i3/sway
ii clang-12.0.1_2                                              Low Level Virtual Machine - C language family frontend
ii dnssec-anchors-20181012_2                                   DNSSEC trust anchors for the root zone
ii fuzzel-1.7.0_1                                              Application launcher for wlroots based Wayland compositors
ii grimshot-1.7_20220521                                       Helper for screenshots within sway
ii i3status-rust-0.21.9_1                                      Replacement for i3status, written in Rust
ii libclang-12.0.1_2                                           Low Level Virtual Machine - C frontend runtime library
ii libclang-cpp-12.0.1_2                                       Low Level Virtual Machine - C frontend runtime library (C++ interface)
ii mesa-vulkan-radeon-21.3.7_1                                 Mesa Radeon Vulkan driver
ii mesa-vulkan-radeon-32bit-21.3.7_1                           Mesa Radeon Vulkan driver (32bit)
ii python3-i3ipc-2.2.1_5                                       Improved Python library to control i3wm and sway
ii rust-1.57.0_1                                               Safe, concurrent, practical systems language
ii rust-std-1.57.0_1                                           Safe, concurrent, practical systems language - standard library
ii sway-1.7_20220521                                           Tiling Wayland compositor compatible with i3
ii sway-audio-idle-inhibit-0.1.1_1                             Prevent swayidle from sleeping when audio is playing
ii swaybg-1.1_1                                                Wallpaper tool for Wayland compositors
ii swayidle-1.7.1_1                                            Idle management daemon for Wayland
ii swaylock-1.5_1                                              Screen locker for Wayland
ii trousers-0.3.14_11                                          Trusted Computing Software Stack for the TPM
ii vulkan-loader-1.3.204.1_1                                   Vulkan Installable Client Driver (ICD) loader
ii vulkan-loader-32bit-1.3.204.1_1                             Vulkan Installable Client Driver (ICD) loader (32bit)
ii wayvnc-0.4.1_1                                              VNC server for wlroots based Wayland compositors
ii wlr-randr-0.2.0_1                                           Wayland clone of xrandr for wlroots compositors
ii wlroots-0.15.1_1                                            Modular Wayland compositor library
ii wlroots-devel-0.15.1_1                                      Modular Wayland compositor library - development files
ii xdg-desktop-portal-wlr-0.5.0_1                              Backend of xdg-desktop-portal for wlroots
```

cyrinux, 2022-05-28:
Hi,
Thanks for the report.

Do you have this env variable set ? `WAYLAND_DISPLAY` ? With what value ?

JJGadgets, 2022-05-29:
> Hi, Thanks for the report.
> 
> Do you have this env variable set: `WAYLAND_DISPLAY` ? With what value ?

`WAYLAND_DISPLAY` is `wayland-1`, set using `exec dbus-update-activation-environment DISPLAY SWAYSOCK WAYLAND_DISPLAY XDG_CURRENT_DESKTOP=sway` in Sway's config. Tried to unset it, doesn't change anything.

JJGadgets, 2022-05-29:
Restarting Sway seems to have fixed it (I had unsaved work prior), but now I get a new error.

```
thread 'predictor-eDP-1' panicked at 'Unable to compute luma percent: TIMEOUT', src/frame/capturer/wlroots.rs:128:26
stack backtrace:
   0:     0x557ddbec46b0 - <std::sys_common::backtrace::_print::DisplayBacktrace as core::fmt::Display>::fmt::h18674753585b8cc5
   1:     0x557ddbee716c - core::fmt::write::h650970577346edc1
   2:     0x557ddbec0fb5 - std::io::Write::write_fmt::hcdbe4458fe2ebdfb
   3:     0x557ddbec64cb - std::panicking::default_hook::{{closure}}::h2b3ca2e1a25274db
   4:     0x557ddbec6043 - std::panicking::default_hook::hb3d80776b693aaeb
   5:     0x557ddbd5933e - wluma::main::{{closure}}::hac5ad5b4efe56de1
   6:     0x557ddbec6bc9 - std::panicking::rust_panic_with_hook::habc6079310c0728a
   7:     0x557ddbec6670 - std::panicking::begin_panic_handler::{{closure}}::ha449aee990d62948
   8:     0x557ddbec4b54 - std::sys_common::backtrace::__rust_end_short_backtrace::h54cc540f2a5a6bf2
   9:     0x557ddbec65d9 - rust_begin_unwind
  10:     0x557ddbd33231 - core::panicking::panic_fmt::he85288327cd30385
  11:     0x557ddbd33323 - core::result::unwrap_failed::ha180eafd08eaf142
  12:     0x557ddbd7ae2e - wayland_client::proxy::Main<I>::quick_assign::{{closure}}::h5d0332f5ba01d6f6
  13:     0x557ddbd7b886 - wayland_commons::filter::Filter<E>::send::h4188a2956c92af27
  14:     0x557ddbd6a3a5 - wayland_client::imp::proxy::ProxyInner::assign::{{closure}}::h894514efdf4d0a49
  15:     0x557ddbe3eafe - scoped_tls::ScopedKey<T>::with::h18d79b63d6ea52a1
  16:     0x557ddbe3d3df - wayland_client::imp::proxy::proxy_dispatcher::{{closure}}::h3a388c2a97c405fe
  17:     0x557ddbe40d0a - wayland_client::imp::proxy::proxy_dispatcher::h59caa90d4624e586
  18:     0x7f1dfce4b04c - <unknown>
  19:     0x7f1dfce4b2cc - wl_display_dispatch_queue_pending
  20:     0x557ddbd7a433 - scoped_tls::ScopedKey<T>::set::h2c340a82916e5f54
  21:     0x557ddbd5cc09 - <wluma::frame::capturer::wlroots::Capturer as wluma::frame::capturer::Capturer>::run::h3171b8c751a86480
  22:     0x557ddbd5ee46 - std::sys_common::backtrace::__rust_begin_short_backtrace::h1b6ace0267256e64
  23:     0x557ddbd82a58 - core::ops::function::FnOnce::call_once{{vtable.shim}}::h2d19e92af92472d4
  24:     0x557ddbec9f43 - std::sys::unix::thread::Thread::new::thread_start::h2582dd3e3de2d3c5
  25:     0x7f1dfda9aeae - start_thread
                               at /builddir/glibc-2.32/nptl/pthread_create.c:463:8
  26:     0x7f1dfd8852ff - __GI___clone
                               at /builddir/glibc-2.32/misc/../sysdeps/unix/sysv/linux/x86_64/clone.S:95
  27:                0x0 - <unknown>
```

Relevant config:
```toml
[[output.backlight]]
name = "eDP-1"
path = "/sys/class/backlight/amdgpu_bl0"
capturer = "wlroots"
```

Additionally, if `MANGOHUD=1` env var is present, MangoHud will attempt to hook into wluma's Vulkan capture and will confuse wluma on where to find config files:
```
[2022-05-29T09:38:29Z INFO  wluma] Continue adjusting brightness and wluma will learn your preference over time.
skipping config: /usr/bin/MangoHud.conf [ not found ]
skipping config: /home/jj-void/.config/MangoHud/wluma.conf [ not found ]
parsing config: /home/jj-void/.config/MangoHud/MangoHud.conf [ ok ]
Unknown option 'procmem'
thread 'predictor-eDP-1' panicked at 'Unable to compute luma percent: TIMEOUT', src/frame/capturer/wlroots.rs:128:26
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

Would you like me to move these issues into their own issues for tracking, or leave them in this issue?

maximbaz, 2022-05-29:
Hello! Thanks for the reports!

> I believe the MangoHud messages come from MangoHud itself, not wluma, so I don't think it's actually an issue?

Good catch, indeed this is not our errors 🙂 

> Unable to compute luma percent: TIMEOUT

Now this is a issue we should try to get fixed... To be honest I don't really know where to get started, as I can't reproduce on my hardware and there isn't much in the stack trace that could give us details... Overall, we can see that the error happens here:

https://github.com/maximbaz/wluma/blob/0c49755fdcf19b22841844df6508f29c31975ce2/src/frame/capturer/wlroots.rs#L125-L128

This function should get the contents of the screen using Vulkan and evaluate how much "light" the screen is currently displaying.

The function is here:

https://github.com/maximbaz/wluma/blob/0c49755fdcf19b22841844df6508f29c31975ce2/src/frame/vulkan.rs#L144

But as we don't get any error from it, just `TIMEOUT`, I don't quite know where it gets stuck..

Would you be able to try to debug this a little on your end? Set `RUST_LOG=trace` for example and run `cargo run`, and maybe try to add some log prints to see how far the code reaches before it gets stuck with the timeout? It should hopefully give us some hints!

JJGadgets, 2022-05-29:
Sure, I'm comfortable with debugging (but have no Rust experience lmao), only issue is that I'm actually trying out `wluma` when I was procrastinating, and I have an exam and some projects upcoming in the next week, so I hope you don't mind if replies are delayed.

```shell
❯ RUST_LOG=trace RUST_BACKTRACE=full MANGOHUD=0 cargo run
    Finished dev [unoptimized + debuginfo] target(s) in 0.05s
     Running `target/debug/wluma`
[2022-05-29T16:28:42Z DEBUG wluma] Using Config {
        als: Webcam {
            video: 0,
            thresholds: {
                20: "dim",
                75: "outdoors",
                10: "dark",
                0: "night",
                50: "bright",
                33: "normal",
            },
        },
        output: [
            Backlight(
                BacklightOutput {
                    name: "eDP-1",
                    path: "/sys/class/backlight/amdgpu_bl0",
                    capturer: Wlroots,
                    min_brightness: 1,
                },
            ),
            Backlight(
                BacklightOutput {
                    name: "keyboard-thinkpad",
                    path: "/sys/bus/platform/devices/thinkpad_acpi/leds/tpacpi::kbd_backlight",
                    capturer: None,
                    min_brightness: 0,
                },
            ),
        ],
    }
[2022-05-29T16:28:42Z INFO  wluma] Continue adjusting brightness and wluma will learn your preference over time.
[2022-05-29T16:28:42Z TRACE wluma::als::webcam] ALS (webcam): outdoors (100)
[2022-05-29T16:28:42Z TRACE wluma::predictor::controller] Prediction: 2 (lux: outdoors, luma: 0)
[2022-05-29T16:28:42Z DEBUG wluma::frame::capturer::wlroots] Using output 'Chimei Innolux Corporation 0x14F2 0x00000000 (eDP-1)' for config 'eDP-1'
[2022-05-29T16:28:42Z TRACE wluma::als::webcam] ALS (webcam): outdoors (100)
[2022-05-29T16:28:42Z TRACE wluma::predictor::controller] Prediction: 2 (lux: outdoors, luma: 0)
[2022-05-29T16:28:42Z TRACE wluma::als::webcam] ALS (webcam): outdoors (100)
[2022-05-29T16:28:42Z TRACE wluma::als::webcam] ALS (webcam): night (7)
[2022-05-29T16:28:42Z TRACE wluma::predictor::controller] Prediction: 2 (lux: outdoors, luma: 0)
[2022-05-29T16:28:42Z TRACE wluma::als::webcam] ALS (webcam): night (7)
[2022-05-29T16:28:42Z TRACE wluma::als::webcam] ALS (webcam): night (7)
[2022-05-29T16:28:43Z TRACE wluma::predictor::controller] Prediction: 2 (lux: outdoors, luma: 0)
[2022-05-29T16:28:43Z TRACE wluma::als::webcam] ALS (webcam): night (7)
[2022-05-29T16:28:43Z TRACE wluma::als::webcam] ALS (webcam): night (7)
[2022-05-29T16:28:43Z TRACE wluma::predictor::controller] Prediction: 2 (lux: outdoors, luma: 0)
[2022-05-29T16:28:43Z TRACE wluma::als::webcam] ALS (webcam): night (7)
[2022-05-29T16:28:43Z TRACE wluma::als::webcam] ALS (webcam): night (7)
[2022-05-29T16:28:43Z TRACE wluma::predictor::controller] Prediction: 2 (lux: outdoors, luma: 0)
[2022-05-29T16:28:43Z TRACE wluma::als::webcam] ALS (webcam): night (7)
thread 'predictor-eDP-1' panicked at 'Unable to compute luma percent: TIMEOUT', src/frame/capturer/wlroots.rs:128:26
stack backtrace:
   0:     0x5654a9c69970 - <std::sys_common::backtrace::_print::DisplayBacktrace as core::fmt::Display>::fmt::h18674753585b8cc5
   1:     0x5654a9c8c42c - core::fmt::write::h650970577346edc1
   2:     0x5654a9c66275 - std::io::Write::write_fmt::hcdbe4458fe2ebdfb
   3:     0x5654a9c6b78b - std::panicking::default_hook::{{closure}}::h2b3ca2e1a25274db
   4:     0x5654a9c6b303 - std::panicking::default_hook::hb3d80776b693aaeb
   5:     0x5654a9746ab3 - <alloc::boxed::Box<F,A> as core::ops::function::Fn<Args>>::call::h267ec243cda9bc74
                               at /builddir/rustc-1.57.0-src/library/alloc/src/boxed.rs:1705:9
   6:     0x5654a978094b - wluma::main::{{closure}}::hdb3f5ec7d8d1fdbf
                               at /home/jjgadgets/GitRepos/Others/wluma/src/main.rs:14:9
   7:     0x5654a9c6be89 - std::panicking::rust_panic_with_hook::habc6079310c0728a
   8:     0x5654a9c6b930 - std::panicking::begin_panic_handler::{{closure}}::ha449aee990d62948
   9:     0x5654a9c69e14 - std::sys_common::backtrace::__rust_end_short_backtrace::h54cc540f2a5a6bf2
[... 46 lines cut ...]
                               at /home/jjgadgets/GitRepos/Others/wluma/src/main.rs:87:29
  36:     0x5654a974f25c - std::sys_common::backtrace::__rust_begin_short_backtrace::hb09d013859fddd38
                               at /builddir/rustc-1.57.0-src/library/std/src/sys_common/backtrace.rs:123:18
  37:     0x5654a9792bd0 - std::thread::Builder::spawn_unchecked::{{closure}}::{{closure}}::hdfdac7fca1df38a5
                               at /builddir/rustc-1.57.0-src/library/std/src/thread/mod.rs:483:17
  38:     0x5654a97e8250 - <core::panic::unwind_safe::AssertUnwindSafe<F> as core::ops::function::FnOnce<()>>::call_once::h6c0cebfcdde1e9b7
                               at /builddir/rustc-1.57.0-src/library/core/src/panic/unwind_safe.rs:271:9
  39:     0x5654a971081a - std::panicking::try::do_call::h81534e8a3112760c
                               at /builddir/rustc-1.57.0-src/library/std/src/panicking.rs:403:40
  40:     0x5654a9710d3d - __rust_try
  41:     0x5654a9710451 - std::panicking::try::hdd0ba3b931e62611
                               at /builddir/rustc-1.57.0-src/library/std/src/panicking.rs:367:19
  42:     0x5654a96cb520 - std::panic::catch_unwind::h65f38094707f0e98
[... cut ...]
                               at /builddir/rustc-1.57.0-src/library/core/src/ops/function.rs:227:5
  45:     0x5654a9c6f203 - std::sys::unix::thread::Thread::new::thread_start::h2582dd3e3de2d3c5
  46:     0x7ff2b14a3eae - start_thread
                               at /builddir/glibc-2.32/nptl/pthread_create.c:463:8
  47:     0x7ff2b128e2ff - __GI___clone
                               at /builddir/glibc-2.32/misc/../sysdeps/unix/sysv/linux/x86_64/clone.S:95
  48:                0x0 - <unknown>
```

cyrinux, 2022-05-29:
Ok no problem, take your time. You can try to add some more debug log in this luma_percent function and give the logs? 
Thanks! (exams first please)

tpeacock19, 2022-12-11:
I'm experiencing the same issue. Here are the details of my environment and some logs associated. If it helps.

Hardware: ThinkPad T14 Gen 1 (AMD), Aukey PC-W3 webcam, AMDGPU
Software: Arch Linux, River, wluma built from source

<details open>
<summary>wluma.log</summary>

```sh
src/archlinux/wluma
 ❯ RUST_LOG=trace RUST_BACKTRACE=full vk_radv cargo run                 main 306d ⬡
    Finished dev [unoptimized + debuginfo] target(s) in 0.10s
     Running `target/debug/wluma`
[2022-12-11T03:19:51Z DEBUG wluma] Using Config {
        als: Webcam {
            video: 4,
            thresholds: {
                75: "outdoors",
                0: "night",
                60: "bright",
                30: "dim",
                45: "normal",
                15: "dark",
            },
        },
        output: [
            Backlight(
                BacklightOutput {
                    name: "eDP-1",
                    path: "/sys/class/backlight/amdgpu_bl0",
                    capturer: Wlroots,
                    min_brightness: 1,
                },
            ),
        ],
    }
[2022-12-11T03:19:51Z INFO  wluma] Continue adjusting brightness and wluma will learn your preference over time.
[2022-12-11T03:19:51Z TRACE wluma::als::webcam] ALS (webcam): outdoors (100)
[2022-12-11T03:19:51Z DEBUG wluma::frame::capturer::wlroots] Using output 'IVO 0x057D (eDP-1)' for config 'eDP-1'
[2022-12-11T03:19:51Z TRACE wluma::als::webcam] ALS (webcam): outdoors (100)
[2022-12-11T03:19:51Z TRACE wluma::als::webcam] ALS (webcam): outdoors (100)
[2022-12-11T03:19:51Z TRACE wluma::als::webcam] ALS (webcam): outdoors (100)
[2022-12-11T03:19:52Z TRACE wluma::als::webcam] ALS (webcam): outdoors (100)
[2022-12-11T03:19:52Z TRACE wluma::als::webcam] ALS (webcam): outdoors (100)
[2022-12-11T03:19:52Z TRACE wluma::als::webcam] ALS (webcam): night (7)
[2022-12-11T03:19:52Z TRACE wluma::als::webcam] ALS (webcam): night (7)
[2022-12-11T03:19:52Z TRACE wluma::als::webcam] ALS (webcam): night (7)
[2022-12-11T03:19:52Z TRACE wluma::als::webcam] ALS (webcam): night (7)
[2022-12-11T03:19:52Z TRACE wluma::als::webcam] ALS (webcam): night (7)
thread 'predictor-eDP-1' panicked at 'Unable to compute luma percent: TIMEOUT', src/frame/capturer/wlroots.rs:128:26
stack backtrace:
   0:     0x55e2c939e1ad - std::backtrace_rs::backtrace::libunwind::trace::h8217d0a8f3fd2f41
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library/std/src/../../backtrace/src/backtrace/libunwind.rs:93:5
   1:     0x55e2c939e1ad - std::backtrace_rs::backtrace::trace_unsynchronized::h308103876b3af410
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library/std/src/../../backtrace/src/backtrace/mod.rs:66:5
   2:     0x55e2c939e1ad - std::sys_common::backtrace::_print_fmt::hc208018c6153605e
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library/std/src/sys_common/backtrace.rs:66:5
   3:     0x55e2c939e1ad - <std::sys_common::backtrace::_print::DisplayBacktrace as core::fmt::Display>::fmt::hf89a7ed694dfb585
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library/std/src/sys_common/backtrace.rs:45:22
   4:     0x55e2c93c11dc - core::fmt::write::h21038c1382fe4264
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library/core/src/fmt/mod.rs:1197:17
   5:     0x55e2c939b441 - std::io::Write::write_fmt::h7dbb1c9a3c254aef
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library/std/src/io/mod.rs:1672:15
   6:     0x55e2c939fb85 - std::sys_common::backtrace::_print::h4e8889719c9ddeb8
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library/std/src/sys_common/backtrace.rs:48:5
   7:     0x55e2c939fb85 - std::sys_common::backtrace::print::h1506fe2cb3022667
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library/std/src/sys_common/backtrace.rs:35:9
   8:     0x55e2c939fb85 - std::panicking::default_hook::{{closure}}::hd9d7ce2a8a782440
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library/std/src/panicking.rs:295:22
   9:     0x55e2c939f8a6 - std::panicking::default_hook::h5b16ec25444b1b5d
[... 166 lines cut ...]
Dec 10 20:19:51 baalsecundus kernel: [30707.226695] amdgpu 0000:06:00.0: amdgpu:     PERMISSION_FAULTS: 0x0
Dec 10 20:19:51 baalsecundus kernel: [30707.226696] amdgpu 0000:06:00.0: amdgpu:     MAPPING_ERROR: 0x0
Dec 10 20:19:51 baalsecundus kernel: [30707.226698] amdgpu 0000:06:00.0: amdgpu:     RW: 0x0
Dec 10 20:19:51 baalsecundus kernel: [30707.226700] amdgpu 0000:06:00.0: amdgpu: [gfxhub0] no-retry page fault (src_id:0 ring:24 vmid:6 pasid:32778, for process wluma pid 731265 thread predictor-eDP-1 pid 731268)
Dec 10 20:19:51 baalsecundus kernel: [30707.226703] amdgpu 0000:06:00.0: amdgpu:   in page starting at address 0x0000800100de9000 from IH client 0x1b (UTCL2)
Dec 10 20:19:51 baalsecundus kernel: [30707.226707] amdgpu 0000:06:00.0: amdgpu: VM_L2_PROTECTION_FAULT_STATUS:0x00000000
Dec 10 20:19:51 baalsecundus kernel: [30707.226708] amdgpu 0000:06:00.0: amdgpu:     Faulty UTCL2 client ID: CB (0x0)
Dec 10 20:19:51 baalsecundus kernel: [30707.226710] amdgpu 0000:06:00.0: amdgpu:     MORE_FAULTS: 0x0
Dec 10 20:19:51 baalsecundus kernel: [30707.226712] amdgpu 0000:06:00.0: amdgpu:     WALKER_ERROR: 0x0
[... cut ...]
Dec 10 20:19:51 baalsecundus kernel: [30707.226730] amdgpu 0000:06:00.0: amdgpu:     WALKER_ERROR: 0x0
Dec 10 20:19:51 baalsecundus kernel: [30707.226732] amdgpu 0000:06:00.0: amdgpu:     PERMISSION_FAULTS: 0x0
Dec 10 20:19:51 baalsecundus kernel: [30707.226734] amdgpu 0000:06:00.0: amdgpu:     MAPPING_ERROR: 0x0
Dec 10 20:19:51 baalsecundus kernel: [30707.226736] amdgpu 0000:06:00.0: amdgpu:     RW: 0x0
Dec 10 20:20:02 baalsecundus kernel: [30717.636063] [drm:amdgpu_job_timedout [amdgpu]] *ERROR* ring gfx timeout, but soft recovered
```

<details>

maximbaz, 2022-12-11:
Thanks for reporting, to be honest it doesn't immediately strike me as the same issue, perhaps file it as a new one, so that its not forgotten in the shadow of this one? 

By the way, do you have only integrated gpu, or also a separate one? (you can answer in the new ticket so that we move the discussion in one place)

maximbaz, 2024-10-21:
Closing due to lack of activity, please leave a comment or open a new issue if this is still relevant.

closed 2024-10-21 by maximbaz