max-baz/wluma #52: [Feature request]: Wildcard support in backlight path.

Mic1e, 2022-07-03:
### Please describe your feature request

I have an amd cpu in my laptop and the path to the backlight device changes from boot to boot. The device changes names from "amdgpu_bl0" to "amdgpu_bl1" and so on. This makes it impossible to configure wluma as it cannot find the device after a reboot.

Therefore I think it would be nice to be able to use a wildcard in the path such as "amdgpu_bl*" and have wluma use the first directory that matches this but it does not seem to work/be implemented.

maximbaz, 2022-07-04:
This sounds like a good idea, in fact we have this implemented for [iio light sensor](https://github.com/maximbaz/wluma/blob/df8cf1f05ace838460296542e832224ad90a2a26/src/als/iio.rs#L30) (exactly because of this reason), but you are right not for anything else.

In the iio the way it is implemented is not using wildcard, but configuring parent folder (which is always static), and it is then a hardcoded logic of the app to enumerate child folders.

We can try the same, or we can switch to wildcards in both places, whatever works best, as long as it's consistent 👍

MarkGhanz, 2023-11-10:
Same issue here. How's the status?

maximbaz, 2023-11-10:
Nobody is working on it at the moment, if you'd like to try your hand at it, you are most welcome - reach out if you have any questions, here or in a PR 😉