max-baz/wluma #54: [Bug]: error: failed to run custom build command for `v4l-sys v0.2.0`

phqzgunsfjror, 2022-08-20:
### Steps for reproducing the issue

```
#https://www.rust-lang.org/tools/install
curl --proto '=https' --tlsv1.2 -sSf https://sh.rustup.rs | sh
source $HOME/.cargo/env

sudo apt-get install libudev-dev
sudo apt-get install -y clang

git clone https://github.com/maximbaz/wluma.git
cd wluma

make build
```

rust version:
```
rustup
rustup 1.25.1 (bb60b1e89 2022-07-12)
```

### What is the buggy behavior?

```
error: failed to run custom build command for `v4l-sys v0.2.0`
```

### What is the expected behavior?

make build to be successful

### Logs

```shell
RUST_LOG=debug wluma
```

### Version

4.1.2 and main

### Environment

```shell
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.1 LTS
Release:	22.04
Codename:	jammy
Linux yogao 5.15.0-46-generic #49-Ubuntu SMP Thu Aug 4 18:03:25 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
Command 'pacman' not found, but can be installed with:
sudo apt install pacman
ii  clang                                                       1:14.0-55~exp2                          amd64        C, C++ and Objective-C compiler (LLVM based), clang binary
ii  clang-14                                                    1:14.0.0-1ubuntu1                       amd64        C, C++ and Objective-C compiler
ii  libclang-common-14-dev                                      1:14.0.0-1ubuntu1                       amd64        Clang library - Common development package
ii  libclang-cpp14                                              1:14.0.0-1ubuntu1                       amd64        C++ interface to the Clang library
ii  libclang1-14                                                1:14.0.0-1ubuntu1                       amd64        C interface to the Clang library
ii  libvulkan-dev:amd64                                         1.3.204.1-2                             amd64        Vulkan loader library -- development files
ii  libvulkan1:amd64                                            1.3.204.1-2                             amd64        Vulkan loader library
ii  mesa-vulkan-drivers:amd64                                   22.0.5-0ubuntu0.1                       amd64        Mesa Vulkan graphics drivers
ii  p11-kit-modules:amd64                                       0.24.0-6build1                          amd64        p11-glue proxy and trust modules
```

maximbaz, 2022-08-26:
Hello, is there more errors in the build logs perhaps?

The latest CI (which also runs on ubuntu) failed for a different reason, but it actually succeeds in building this part 🤔 

I wonder if you perhaps miss some of these dependencies?

https://github.com/maximbaz/wluma/blob/b9a36723bc54bd3296364bb92387181b5f892e17/.github/workflows/ci.yml#L15

phqzgunsfjror, 2022-08-28:
@maximbaz, thank you! That helped and fixed the build.

Is it worth to mention it in the README.md under the "Build" section?

maximbaz, 2022-08-28:
Awesome! 

Sure, if you have time, please submit PR 😉

phqzgunsfjror, 2022-08-28:
All right. Here it is: https://github.com/maximbaz/wluma/pull/55
If it should be different, just let me know =)

maximbaz, 2022-08-28:
It's fine, thanks!

closed 2022-08-28 by phqzgunsfjror