max-baz/wluma #56: [Bug]: wluma writing "1" to /sys/class/backlight/intel_backlight/brightness *really* fast

kelvie, 2022-09-23:
### Steps for reproducing the issue

Run wluma with this config:

```toml
[als.iio]
path = "/sys/bus/iio/devices"
thresholds = { 0 = "night", 20 = "dark", 80 = "dim", 250 = "normal", 500 = "bright", 800 = "outdoors" }

# [als.webcam]
# video = 0
# thresholds = { 0 = "night", 15 = "dark", 30 = "dim", 45 = "normal", 60 = "bright", 75 = "outdoors" }

# [als.time]
# thresholds = { 0 = "night", 7 = "dark", 9 = "dim", 11 = "normal", 13 = "bright", 16 = "normal", 18 = "dark", 20 = "night" }

# [als.none]

[[output.backlight]]
name = "eDP-1"
path = "/sys/class/backlight/intel_backlight"
capturer = "none"
```

It went away after I set the brightness manually once, so I'm not sure how to reproduce it (may look at the code later, though).

### What is the buggy behavior?

I remember doing an strace and it kept writing "1" to /sys/class/backlight/intel_backlight/brightness

To the point that `systemd-udevd` was using 100% CPU and a `udevadm monitor` would spam the console with brightness changes

### What is the expected behavior?

That it set it to `1` once, and leave it there.

### Logs

```shell
Can't reproduce it anymore, but:

> cat ~/.local/share/wluma/eDP-1.yaml
---
output_name: eDP-1
entries:
  - lux: dark
    luma: 0
    brightness: 10561
  - lux: night
    luma: 0
    brightness: 1
  - lux: normal
    luma: 0
    brightness: 22081
```
```

### Version

4.1.2.r1.g0c49755-1

### Environment

```shell
lsb_release -a; uname -a; pgrep -l sway; pacman -Q | egrep "(wlroots|vulkan|sway|clang|rust)";

LSB Version:    n/a
Distributor ID: Arch
Description:    Arch Linux
Release:        rolling
Codename:       n/a
Linux kelvie-fw 5.19.10-arch1-1 #1 SMP PREEMPT_DYNAMIC Tue, 20 Sep 2022 15:17:59 +0000 x86_64 GNU/Linux
egrep: warning: egrep is obsolescent; using grep -E
clang 14.0.6-2
rust 1:1.63.0-1
rust-analyzer 20220919-1
rust-src 1:1.63.0-1
vulkan-icd-loader 1.3.226-1
vulkan-tools 1.3.226-1
```
```

kelvie, 2022-09-24:
I suspect this happens when the illuminance is 0?

kelvie, 2022-09-24:
Nope, that's not it. With my hand over the ALS, it doesn't go in the same loop, only when I close the lid.

kelvie, 2022-09-24:
Confirmed that it writes '1" in a really tight loop to the brightness when the ALS reports 0.

kelvie, 2022-09-24:
With RUST_LOG=trace, it seems like the predictor wants it to set it to 0 for some reason, perhaps the KDE / plasma brightness settings are interfering with wluma, by setting the night mode to 0

kelvie, 2022-10-03:
From quickly reading the code on my break:

https://github.com/maximbaz/wluma/blob/main/src/brightness/controller.rs#L106 

This is being run constantly (with only a 1ms sleep) because current is 1, and that means that `step` must be 0, which checks because here:

https://github.com/maximbaz/wluma/blob/main/src/brightness/controller.rs#L26

If `step` is zero will always return false, it assumes step is nonzero.

I'm not sure I have time to try to reproduce how `step` got set to 0 in the first place.

kelvie, 2022-10-03:
OK, I think it's this:

https://github.com/maximbaz/wluma/blob/main/src/brightness/controller.rs#L90

If desired is `1` and current is `0`, this will end up with a step of `0` which would cause this (due to your `div_ceil` function).

cyrinux, 2022-10-11:
@kelvie @maximbaz and I did a big PR to fix this, could you try and tell us if it's fix please ? 🍰

maximbaz, 2022-10-11:
In fact it doesn't seem to be the case, I added a unit test for this theory, when `desired = 1` and `current = 0`, the `desired > current` would be `true`, `div_ceil` would be called on `div_ceil(1-0, 200)` and it would compute `(1+200-1/200)` which is `1`. More research is needed 😉

kelvie, 2022-10-11:
> In fact it doesn't seem to be the case, I added a unit test for this theory, when `desired = 1` and `current = 0`, the `desired > current` would be `true`, `div_ceil` would be called on `div_ceil(1-0, 200)` and it would compute `(1+200-1/200)` which is `1`. More research is needed wink

Oh you're right, I read it wrong and thought `0` and `1` were `x` and `y`, but it's numerator and denominator. Hm.

In either case I can't reproduce it after turning off KDE's screen power control.

maximbaz, 2022-10-11:
Ok, since neither of the 3 of us can repro, I suggest we close the bug until we find a reliable way to repro and solve it 👍

closed 2022-10-11 by maximbaz