max-baz/wluma #60: [Feature request]: Dim screen after inactivity

pollen00, 2022-11-01:
### Please describe your feature request

Some config option to dim the screen after some amount of time of inactivity would be very useful. Since I am already running wluma to control brightness, I'd prefer not to have to run another daemon just to lower brightness after some time.

maximbaz, 2022-11-02:
Interesting scenario to consider, many thanks for sharing! I can't say at the moment whether I think it's the right move to incorporate it in wluma or to let two daemons do one thing right, but in any case, we should have a decent support for this, wluma should not conflict with lowering brightness on inactivity, it should neither learn the lowered brightness as the data point (which would negatively affect the prediction algorithm).

pollen00, 2022-11-15:
Being able to control this with tools such as xss or xautolock would also be helpful.

Atemu, 2024-05-22:
> it should neither learn the lowered brightness as the data point (which would negatively affect the prediction algorithm).

On GNOME, it appears to be doing this. If you enable GNOME's auto-dimming in power settings and let it sit for a while, you'll see wluma learning the dimmed brightness as a data point and then the actual one again when you interact with the machine again.

kcoderhtml, 2024-07-26:
It does this with hyperidle too; i'm not sure whether it would be possible to specify over dbus whether a brightness change was initiated by a program or not but that might be worth looking into

max-baz, 2026-08-15:
This is now implemented on `main` branch and will be part of the next release. It is generally impossible to detect who initiated a brightness change, therefore it is recommended to disable any other auto-dimming and use wluma's own dimming on idle mechanism.

closed 2026-08-15 by max-baz