max-baz/wluma #64: [Bug]:

coffebar, 2022-11-26:
### Steps for reproducing the issue

- add user to **video** group
- reboot
- Install wluma from AUR (on Manjaro i3)
- run **wluma**

### What is the buggy behavior?

```
thread 'predictor-eDP-1' panicked at 'called `Result::unwrap()` on an `Err` value: NoCompositorListening', src/frame/capturer/wlroots.rs:70:49
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

### What is the expected behavior?

running process without errors, adjusting brightness

### Logs

```shell
[2022-11-26T08:44:09Z DEBUG wluma] Using Config {
        als: Iio {
            path: "/sys/bus/iio/devices",
            thresholds: {
                250: "normal",
                80: "dim",
                20: "dark",
                0: "night",
                800: "outdoors",
                500: "bright",
            },
        },
        output: [
            Backlight(
                BacklightOutput {
                    name: "eDP-1",
                    path: "/sys/class/backlight/intel_backlight",
                    capturer: Wlroots,
                    min_brightness: 1,
                },
            ),
            Backlight(
                BacklightOutput {
                    name: "keyboard-dell",
                    path: "/sys/bus/platform/devices/dell-laptop/leds/dell::kbd_backlight",
                    capturer: None,
                    min_brightness: 0,
                },
            ),
        ],
    }
thread 'predictor-eDP-1' panicked at 'called `Result::unwrap()` on an `Err` value: NoCompositorListening', src/frame/capturer/wlroots.rs:70:49
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

### Version

4.1.2-1

### Environment

```shell
LSB Version:	n/a
Distributor ID:	ManjaroLinux
Description:	Manjaro Linux
Release:	22.0.0
Codename:	Sikaris
Linux kindle 5.15.76-1-MANJARO #1 SMP PREEMPT Sat Oct 29 14:22:16 UTC 2022 x86_64 GNU/Linux
egrep: warning: egrep is obsolescent; using grep -E
clang 14.0.6-3
lib32-vulkan-icd-loader 1.3.226-1
lib32-vulkan-intel 22.2.3-1
lib32-vulkan-radeon 22.2.3-1
rust 1:1.65.0-1
rust-analyzer 20221107-1
rust-src 1:1.65.0-1
vulkan-headers 1:1.3.227-1
vulkan-icd-loader 1.3.226-1
vulkan-intel 22.2.3-1
zsh: command not found: dpkg
egrep: warning: egrep is obsolescent; using grep -E
```

maximbaz, 2022-11-26:
Hello, at the moment the app supports running only under wlroots-based compositors such as sway, for supporting other desktop environments please follow #5

coffebar, 2022-11-26:
👍

closed 2022-11-26 by maximbaz