max-baz/wluma #65: [Bug]: Percent calculation timeout error

tpeacock19, 2022-12-13:
### Steps for reproducing the issue

Start wluma on my machine with the following config:

```toml
[als.webcam]
video = 4
thresholds = { 0 = "night", 15 = "dark", 30 = "dim", 45 = "normal", 60 = "bright", 75 = "outdoors" }

[[output.backlight]]
name = "eDP-1"
path = "/sys/class/backlight/amdgpu_bl0"
capturer = "wlroots"

```

### What is the buggy behavior?

wluma does not run after identifying the correct output.

### What is the expected behavior?

wluma should run without error.

### Logs

```shell
<details open>
<summary>wluma.log</summary>

src/archlinux/wluma
 ❯ RUST_LOG=trace RUST_BACKTRACE=full vk_radv cargo run                 main 306d ⬡
    Finished dev [unoptimized + debuginfo] target(s) in 0.10s
     Running `target/debug/wluma`
[2022-12-11T03:19:51Z DEBUG wluma] Using Config {
        als: Webcam {
            video: 4,
            thresholds: {
                75: "outdoors",
                0: "night",
                60: "bright",
                30: "dim",
                45: "normal",
                15: "dark",
            },
        },
        output: [
            Backlight(
                BacklightOutput {
                    name: "eDP-1",
                    path: "/sys/class/backlight/amdgpu_bl0",
                    capturer: Wlroots,
                    min_brightness: 1,
                },
            ),
        ],
    }
[2022-12-11T03:19:51Z INFO  wluma] Continue adjusting brightness and wluma will learn your preference over time.
[2022-12-11T03:19:51Z TRACE wluma::als::webcam] ALS (webcam): outdoors (100)
[2022-12-11T03:19:51Z DEBUG wluma::frame::capturer::wlroots] Using output 'IVO 0x057D (eDP-1)' for config 'eDP-1'
[2022-12-11T03:19:51Z TRACE wluma::als::webcam] ALS (webcam): outdoors (100)
[2022-12-11T03:19:51Z TRACE wluma::als::webcam] ALS (webcam): outdoors (100)
[2022-12-11T03:19:51Z TRACE wluma::als::webcam] ALS (webcam): outdoors (100)
[2022-12-11T03:19:52Z TRACE wluma::als::webcam] ALS (webcam): outdoors (100)
[2022-12-11T03:19:52Z TRACE wluma::als::webcam] ALS (webcam): outdoors (100)
[2022-12-11T03:19:52Z TRACE wluma::als::webcam] ALS (webcam): night (7)
[2022-12-11T03:19:52Z TRACE wluma::als::webcam] ALS (webcam): night (7)
[2022-12-11T03:19:52Z TRACE wluma::als::webcam] ALS (webcam): night (7)
[2022-12-11T03:19:52Z TRACE wluma::als::webcam] ALS (webcam): night (7)
[2022-12-11T03:19:52Z TRACE wluma::als::webcam] ALS (webcam): night (7)
[... 222 lines cut ...]
Release:	rolling
Codename:	n/a
Linux baalsecundus 6.0.12-zen1-1-zen #1 ZEN SMP PREEMPT_DYNAMIC Thu, 08 Dec 2022 11:03:40 +0000 x86_64 GNU/Linux
amd-vulkan-prefixes 1-1
clang 14.0.6-4
clang13 13.0.1-4
lib32-clang 14.0.6-2
lib32-vulkan-amdgpu-pro 22.20.4_1498766-2
lib32-vulkan-icd-loader 1.3.235-1
lib32-vulkan-mesa-layers 22.2.3-1
lib32-vulkan-radeon 22.2.3-1
rust-analyzer 20221205-1
rustup 1.25.1-2
swayidle-git 1.8.0.r8.g4b9e70c-1
vulkan-amdgpu-pro 22.20.5_1511376-2
vulkan-headers 1:1.3.235-1
vulkan-icd-loader 1.3.235-1
vulkan-mesa-layers 22.2.3-1
vulkan-radeon 22.2.3-1
vulkan-swrast 22.2.3-1
vulkan-tools 1.3.226-1
vulkan-validation-layers 1.3.231.1-1
wlroots 0.15.1-6
river-0.2.0-dev.138+e603c54
```

maximbaz, 2022-12-13:
Could you please confirm if you have one or two GPUs, and in the latter case, which one do you use? The `VM_L2_PROTECTION_FAULT_STATUS` issue seems to be quite frequent if I google for it (e.g. [1](https://bbs.archlinux.org/viewtopic.php?id=266358&p=5), [2](https://gitlab.freedesktop.org/drm/amd/-/issues/1609#note_1369839)), so I wonder if it's either a driver problem, or on the other hand a wluma's problem of using discrete GPU (we have [at least one](https://github.com/maximbaz/wluma/issues/63) other issue for this case, btw, not necessarily a duplicate, but could be relevant), or something else entirely.

tpeacock19, 2022-12-13:
Oh sorry, I just have the amd T14 Gen 1 laptop with AMD Ryzen 7 Pro 4750U. I believe it is just Radeon integrated graphics. I came across those same issues so it may not be a wluma thing. It's the only program I experience this with, but I don't play a lot of games.

maximbaz, 2024-10-21:
Hey, I know this is super old issue but since then I learned a bit more about how to debug Vulkan issues, would you be interested in debugging this further, or this is no longer relevant for you?

maximbaz, 2024-12-25:
Closing as not actionable, please do comment here or in a new issue if someone is able to reproduce and willing to help debugging :pray:

closed 2024-12-25 by maximbaz