max-baz/wluma #68: [Feature request]: Can we have a windows version ?

dngnm, 2023-02-19:
### Please describe your feature request

Hi,
This is a great tool thanks a lot
Can we have a windows version?
Regards.

maximbaz, 2023-02-19:
Hello,

Thanks for the kind words and the request.

In theory it's possible, but it's a huge task, and I won't be able to do it myself. I can keep the issue open, in case it inspires anyone to take a stab at it 😉 I'd be very happy to help with guidance and ideas, so if someone decides to try this out, please do post here in the thread!

The good news is that the code is quite modular, so there shouldn't be any problem integrating the windows bits, the challenge is purely in figuring out how to capture screen contents and get average color on Windows, and optionally how to get ambient light on Windows (this can be tackled separately).

#5 might be relevant to look at, in a sense that it's also about integrating support for a different way of capturing screen contents - implementing Windows support would "look" somewhat like that, but more, and with Windows specifics 🙂

Finally, I would also suggest to search for similar projects, if nothing else, to verify that the idea is possible, and maybe to get some inspiration from them for the implementation.

https://github.com/vinaykumar7686/Smart-Adaptive-Brightness is one result I found after a quick search, judging by the code, it takes screenshot and processes it, so it will be slow / battery-intensive, but it could be a start!

Hope this helps to make it sound less scary! 😁

maximbaz, 2024-10-21:
I'll close the ticket because I think it's unlikely to ever happen, but if anyone actually wants to take a stab at it, ping me here or in a new issue and let's discuss!

closed 2024-10-21 by maximbaz