max-baz/wluma #71: [Bug]: wluma does not build on 4.2.0

GallaisPoutine, 2023-04-23:
### Steps for reproducing the issue

`git clone git@github.com:maximbaz/wluma.git /tmp/wluma`
`cd /tmp/wluma`
`cargo build --locked --release`

### What is the buggy behavior?

Does not build using tag 4.2.0

### What is the expected behavior?

Should build

### Logs

```shell
`cargo build --locked --release`

    Updating crates.io index
error: failed to select a version for the requirement `predicates = "^2.0.1"`
candidate versions found which didn't match: 2.1.1, 2.1.0, 2.0.3, ...
location searched: crates.io index
required by package `mockall v0.11.3`
    ... which satisfies dependency `mockall = "^0.11.2"` (locked to 0.11.3) of package `wluma v4.2.0 (/tmp/wluma)`
```
```

### Version

`git log --oneline --decorate -n5`

```
1f4f3f9 (HEAD -> main, tag: 4.2.0, origin/main, origin/HEAD) Bump version, update deps
d64d06d Add support for system-level config in /etc/xdg/wluma/ (#66)
89fb8c3 chore: fix clippy warnings (#67)
d13f50a chore: dependencies update (#59)
e2350be Revert a1c69bd, add test for current=0 desired=1
```

### Environment

```shell
lsb_release -a; uname -a; pgrep -l sway; pacman -Q | egrep "(wlroots|vulkan|sway|clang|rust)"; dpkg -l | egrep "(wlroots|vulkan|sway|clang|rust)"

bash: lsb_release : commande introuvable
Linux archlinux 6.2.12-arch1-1 #1 SMP PREEMPT_DYNAMIC Thu, 20 Apr 2023 16:11:55 +0000 x86_64 GNU/Linux
egrep: warning: egrep is obsolescent; using grep -E
clang 15.0.7-2
i3status-rust 0.30.7-1
rustup 1.25.2-1
vulkan-icd-loader 1.3.245-1
bash: dpkg : commande introuvable
egrep: warning: egrep is obsolescent; using grep -E
```

maximbaz, 2023-04-23:
Hello! To be honest I can't quite reproduce, but I ran `cargo upgrade` to bump some deps, including the offending mockall, does this help?

GallaisPoutine, 2023-04-23:
Hello !

Yup, dumb me forgot to `rustup update` and `cargo update` before building the sources files.

I figured that earlier when after `cargo update` I had these
```
cargo build
error: package `toml v0.5.9 (https://github.com/HarveyHunt/toml?branch=dotted-table-parsing-toml#7db18be3)` cannot be built because it requires rustc 1.60.0 or newer, while the currently active rustc version is 1.59.0
```
I guess it is resolved now, thank you for your time.

closed 2023-04-23 by GallaisPoutine