max-baz/wluma #73: [Bug]: Crash on AMD APU (Laptop)

XenHat, 2023-05-02:
### Steps for reproducing the issue

Run wluma on a laptop with an AMD APU
Environment: 
WM: `qtile -b wayland`
wluma config:
```toml
[als.webcam]
video = 0
thresholds = { 0 = "night", 15 = "dark", 30 = "dim", 45 = "normal", 60 = "bright", 75 = "outdoors" }

[[output.backlight]]
name = "eDP-1"
path = "/sys/class/backlight/amdgpu_bl0"
capturer = "wlroots"
```

### What is the buggy behavior?

wluma crashes with `ERROR_EXTENSION_NOT_PRESENT`

### What is the expected behavior?

wluma should run properly

### Logs

```shell
RUST_BACKTRACE=full RUST_LOG=debug wluma
[2023-05-02T02:56:31Z DEBUG wluma] Using Config {
        als: Webcam {
            video: 0,
            thresholds: {
                0: "night",
                15: "dark",
                45: "normal",
                60: "bright",
                75: "outdoors",
                30: "dim",
            },
        },
        output: [
            Backlight(
                BacklightOutput {
                    name: "eDP-1",
                    path: "/sys/class/backlight/amdgpu_bl0",
                    capturer: Wlroots,
                    min_brightness: 1,
                },
            ),
        ],
    }
[2023-05-02T02:56:31Z INFO  wluma] Continue adjusting brightness and wluma will learn your preference over time.
thread 'predictor-eDP-1' panicked at 'Unable to initialize Vulkan: ERROR_EXTENSION_NOT_PRESENT', src/frame/capturer/wlroots.rs:86:44
stack backtrace:
   0:     0x55a852ffd07a - <unknown>
   1:     0x55a85301fb1e - <unknown>
   2:     0x55a852ffa945 - <unknown>
   3:     0x55a852ffce45 - <unknown>
   4:     0x55a852ffe94f - <unknown>
   5:     0x55a852ffe68b - <unknown>
   6:     0x55a852ea9cee - <unknown>
   7:     0x55a852fff07d - <unknown>
   8:     0x55a852ffedf9 - <unknown>
   9:     0x55a852ffd4e6 - <unknown>
  10:     0x55a852ffeb02 - <unknown>
  11:     0x55a852e60dd3 - <unknown>
  12:     0x55a852e61283 - <unknown>
  13:     0x55a852ebed5f - <unknown>
  14:     0x55a852e852c3 - <unknown>
  15:     0x55a852ea4c00 - <unknown>
  16:     0x55a853001bb3 - <unknown>
  17:     0x7f5391841bb5 - <unknown>
  18:     0x7f53918c3d90 - <unknown>
  19:                0x0 - <unknown>
```
```

### Version

`aur/wluma 4.2.0-1`

### Environment

```shell
LSB Version:	n/a
Distributor ID:	Artix
Description:	Artix Linux
Release:	rolling
Codename:	n/a
Linux LNV21LT001 6.2.13-artix1-1 #1 SMP PREEMPT_DYNAMIC Thu, 27 Apr 2023 19:59:36 +0000 x86_64 GNU/Linux
egrep: warning: egrep is obsolescent; using grep -E
lib32-vulkan-icd-loader 1.3.245-1
python-pywlroots 0.15.24-3
vulkan-headers 1:1.3.248-1
vulkan-icd-loader 1.3.245-1
vulkan-tools 1.3.245-1
vulkan-validation-layers 1.3.243.0-1
wlroots0.15 0.15.1-2
zsh: command not found: dpkg
egrep: warning: egrep is obsolescent; using grep -E
```
```

maximbaz, 2023-05-02:
Hello! Thanks for reporting. 

This cryptic error means that unfortunately your GPU doesn't support the necessary features ("vulkan extensions") that wluma is using to detect the average lightness of the current contents of the screen. 

I think it's worth searching if you can find anything on internet about Vulkan and your GPU, for example if there maybe is a dependency to install or update, or if the development is in progress, or if there are other interesting learnings. 

Otherwise, your best bet unfortunately is to subscribe to https://github.com/maximbaz/wluma/issues/62 to see if it ever gets implemented (or help out with it 😉).

XenHat, 2023-05-02:
Turns out I was missing `vulkan-radeon`! oops!
Doesn't seem to work on Hyprland due to #8 but I can close this.

closed 2023-05-02 by XenHat