max-baz/wluma #74: [Feature request]: ALS IIO sensor not found. Change path to sensor?

clear-lag, 2023-05-12:
### Please describe your feature request

Trying to use wluma on a MacBook Pro 11,x on Arch. The ALS output is at 
`/sys/devices/platform/applesmc.768/light`

Error shown is `thread 'als' panicked at 'Unable to initialize ALS IIO sensor: "No iio device found"', src/main.rs:112:26`

wluma will not launch.

maximbaz, 2023-05-12:
Hello! Exciting to get it it working on MacBook! 

To clarify, you changed this config line, and you get the crash, is that right? 

https://github.com/maximbaz/wluma/blob/191e98499262db17e0327e90a6dbac1eaeddb865/config.toml#L2

Could you show me the contents of the folder? 

ls -al /sys/devices/platform/applesmc.768/light

Thanks

podbyte, 2023-05-15:
> Hello! Exciting to get it it working on MacBook!
>
> To clarify, you changed this config line, and you get the crash, is that right? [...]

Hi, only just realised there was a config.toml, I pulled it from the AUR and didn't manually build it.
`/sys/devices/platform/applesmc.768/light` is a file, it reads `(x,0)` (x is the light level).
I don't know rust but I made a fork and I'm trying to see if I can get it to work.

maximbaz, 2023-05-16:
Lovely! Allow me to give you some general pointers, and feel free to ask for clarification :slightly_smiling_face: 

In this project, the architecture is already prepared to be plug-and-play for different hardware. `iio` is just one of many light sensors, and you are the first person to desire to have `applesmc` type.

First, create a file next to [`iio.rs`](https://github.com/maximbaz/wluma/blob/main/src/als/iio.rs) and call it `applesmc.rs`. Inside you will implement the logic that is responsible for letting the rest of the app know about this particular driver.

At the very minimum, your file must contain the implementation for this trait:

https://github.com/maximbaz/wluma/blob/191e98499262db17e0327e90a6dbac1eaeddb865/src/als/mod.rs#L11-L13

In `iio.rs`, this looks like so:

https://github.com/maximbaz/wluma/blob/191e98499262db17e0327e90a6dbac1eaeddb865/src/als/iio.rs#L72-L80

and in `time.rs`, it looks like so:

https://github.com/maximbaz/wluma/blob/191e98499262db17e0327e90a6dbac1eaeddb865/src/als/time.rs#L15-L23

Most likely, you can have exactly the same function, with substituted "raw" value read from your sensor.

What are those "thresholds": your sensor will give some integer values, and we need to convert them to some logical groups. I recommend trying to have the same groups, as here:

https://github.com/maximbaz/wluma/blob/191e98499262db17e0327e90a6dbac1eaeddb865/config.toml#L3

Don't worry too much about getting the perfect values, it's just a default config. But it would be nice for them to be reasonable for most people :wink: How you find them is an art... Go in a dark room, see what value your sensor shows, go in a sunny day outside and see the values, etc...

Once you do this, just fill out the rest of the glue code, like defining config like so:

https://github.com/maximbaz/wluma/blob/191e98499262db17e0327e90a6dbac1eaeddb865/src/config/app.rs#L10-L14

Extending `config.toml` with your example commented out, etc, whatever is needed to make the app run :slightly_smiling_face: 

And good luck, enjoy this exploration!