max-baz/wluma #76: [Bug]: not version

newyorkthink, 2023-07-07:
### Steps for reproducing the issue

[image]

### What is the buggy behavior?

1

### What is the expected behavior?

1

### Logs

```shell
RUST_LOG=debug wluma
```

### Version

1

### Environment

```shell
lsb_release -a; uname -a; pgrep -l sway; pacman -Q | egrep "(wlroots|vulkan|sway|clang|rust)"; dpkg -l | egrep "(wlroots|vulkan|sway|clang|rust)"
```

maximbaz, 2023-07-07:
I have no idea what you wanted to express.

newyorkthink, 2023-07-08:
> I have no idea what you wanted to express.我不知道你想表达什么。

is not new package, is three line, but new version package, is not show

maximbaz, 2023-07-08:
Unfortunately it's hard to understand still, can you please try to explain in as much details as you can, what is wrong and what you expected, using specific examples if possible?

cyrinux, 2023-07-08:
@maximbaz @newyorkthink I understand the issue, the binaries files are missing in the release compared to the previous releases.

maximbaz, 2024-10-21:
Closing as duplicate of https://github.com/maximbaz/wluma/issues/97, will eventually solve :crossed_fingers:

closed 2024-10-21 by maximbaz