max-baz/wluma #77: [Bug]: ./Downloads/wluma: error while loading shared libraries: libv4l2.so.0: cannot open shared object file: No such file or directory

newyorkthink, 2023-07-07:
### Steps for reproducing the issue

[image]

### What is the buggy behavior?

[image]

### What is the expected behavior?

[image]

### Logs

```shell
RUST_LOG=debug wluma
```

### Version

[image]

### Environment

```shell
lsb_release -a; uname -a; pgrep -l sway; pacman -Q | egrep "(wlroots|vulkan|sway|clang|rust)"; dpkg -l | egrep "(wlroots|vulkan|sway|clang|rust)"
```

maximbaz, 2023-07-07:
Based on your report, looks like the "buggy behavior" already matches the "expected behavior"?

cyrinux, 2023-07-08:
@newyorkthink You need, in archlinux, the package `v4l-utils` who provide the v4l library to build the project.
```$ pacman -Ql  v4l-utils | grep /usr/lib/libv4l2.so
v4l-utils /usr/lib/libv4l2.so
v4l-utils /usr/lib/libv4l2.so.0
v4l-utils /usr/lib/libv4l2.so.0.0.0
``` 
Install this lib depending the linux distribution you use.

maximbaz, 2023-07-28:
Thanks, and just to add, the full list of dependencies can also be found in README, in the [Build](https://github.com/maximbaz/wluma#build) section. Will close the ticket as hopefully solved, please feel free to comment if you have more specific questions.

closed 2023-07-28 by maximbaz