max-baz/wluma #78: [Bug]: No such file or directory

newyorkthink, 2023-07-07:
### Steps for reproducing the issue

[image]

### What is the buggy behavior?

[image]

### What is the expected behavior?

[image]

### Logs

```shell
RUST_LOG=debug wluma
```

### Version

[image]

### Environment

```shell
lsb_release -a; uname -a; pgrep -l sway; pacman -Q | egrep "(wlroots|vulkan|sway|clang|rust)"; dpkg -l | egrep "(wlroots|vulkan|sway|clang|rust)"
```

maximbaz, 2023-07-07:
Please fill out the issue template, I have no idea what you mean

cyrinux, 2023-07-08:
@newyorkthink you dont have a `iio` device.

Looks like you don't have a light sensors device. So you hace to use the `als` webcam or none mode.

```
[als.webcam]
video = 0
thresholds = { 0 = "night", 15 = "dark", 30 = "dim", 45 = "normal", 60 = "bright", 75 = "outdoors" }

# [als.none]
```

Please try to be more verbose the next time. And fill well the issue according to the template.

maximbaz, 2023-07-28:
I hope this helped, will close as not-actionable, but do feel free to comment if you have further questions! And thanks @cyrinux for the helpful snippet :wink:

closed 2023-07-28 by maximbaz