max-baz/wluma #84: [Bug]: Panicked when running both with sudo and not

Tim-C1, 2023-07-30:
### Steps for reproducing the issue

after installing wluma, I run `wluma`, and output this
```
$ wluma
[2023-07-30T02:49:33Z WARN  wluma] Skipping 'keyboard-dell' as it might be disconnected: No such file or directory (os error 2)
[2023-07-30T02:49:33Z INFO  wluma] Continue adjusting brightness and wluma will learn your preference over time.
thread 'predictor-eDP-1' panicked at 'Unable to init export_dmabuf_manager: Missing', src/frame/capturer/wlroots.rs:80:14
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace

```
when I run `sudo wluma`, output this
```
$ sudo wluma
[2023-07-30T02:52:13Z WARN  wluma] Skipping 'keyboard-dell' as it might be disconnected: No such file or directory (os error 2)
[2023-07-30T02:52:13Z INFO  wluma] Continue adjusting brightness and wluma will learn your preference over time.
thread 'predictor-eDP-1' panicked at 'called `Result::unwrap()` on an `Err` value: XdgRuntimeDirNotSet', src/frame/capturer/wlroots.rs:70:49
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

### What is the buggy behavior?

there are some error messages when I run wluma

### What is the expected behavior?

I think there should be no error messages

### Logs

```shell
RUST_LOG=debug wluma
RUST_LOG=debug ./target/release/wluma 
[2023-07-30T02:56:53Z DEBUG wluma] Using Config {
        als: Iio {
            path: "/sys/bus/iio/devices",
            thresholds: {
                80: "dim",
                20: "dark",
                800: "outdoors",
                0: "night",
                250: "normal",
                500: "bright",
            },
        },
        output: [
            Backlight(
                BacklightOutput {
                    name: "eDP-1",
                    path: "/sys/class/backlight/intel_backlight",
                    capturer: Wlroots,
                    min_brightness: 1,
                },
            ),
            Backlight(
                BacklightOutput {
                    name: "keyboard-dell",
                    path: "/sys/bus/platform/devices/dell-laptop/leds/dell::kbd_backlight",
                    capturer: None,
                    min_brightness: 0,
                },
            ),
        ],
    }
[2023-07-30T02:56:53Z WARN  wluma] Skipping 'keyboard-dell' as it might be disconnected: No such file or directory (os error 2)
[2023-07-30T02:56:53Z INFO  wluma] Continue adjusting brightness and wluma will learn your preference over time.
thread 'predictor-eDP-1' panicked at 'Unable to init export_dmabuf_manager: Missing', src/frame/capturer/wlroots.rs:80:14
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

### Version

4.3.0

### Environment

```shell
lsb_release -a; uname -a; pgrep -l sway; pacman -Q | egrep "(wlroots|vulkan|sway|clang|rust)"; dpkg -l | egrep "(wlroots|vulkan|sway|clang|rust)"
Linux evael 6.4.3-arch1-1 #1 SMP PREEMPT_DYNAMIC Tue, 11 Jul 2023 05:13:39 +0000 x86_64 GNU/Linux
clang 15.0.7-9
vulkan-headers 1:1.3.256-1
vulkan-icd-loader 1.3.255-1
vulkan-intel 23.1.3-2
vulkan-tools 1.3.255-1
```

maximbaz, 2023-07-30:
Hello, the errors you are getting are in screen capturing part of the app (`wlroots.rs`), which only currently works on wlroots-based Wayland window managers. Looks like you are using some other desktop environment,  is this correct? If so, you'd have to disable that entire functionality in config, using `capturer=none`.

Tim-C1, 2023-07-30:
> Hello, the errors you are getting are in screen capturing part of the app (`wlroots.rs`), which only currently works on wlroots-based Wayland window managers. Looks like you are using some other [...]

Thanks for your timely reply, but I think I am using **Wayland** window manager since I type `echo &XDG_SESSIONTYPE` and it output
```
wayland
```
I don't know if this is the case, since I am using Arch with KDE Plasma default settings.

And when I try to set `capturer=none` and `make build `and run, it output the following
```
./target/release/wluma 
[2023-07-30T08:04:50Z WARN  wluma] Skipping 'keyboard-dell' as it might be disconnected: No such file or directory (os error 2)
[2023-07-30T08:04:50Z INFO  wluma] Continue adjusting brightness and wluma will learn your preference over time.
```
Is this output means that wluma is running normally ?

maximbaz, 2023-07-30:
Correct, it now works normally. KDE is indeed not based on wlroots, so wluma can't take into account what's shown on your screen to decide the best brightness levels, but it still uses ambient light around you. Just continue adjusting brightness to your own preference, and you should expect wluma to pick up on your habits (more or less, light sensors are quite not precise).

Tim-C1, 2023-07-30:
Ok, thank you for your help!

maximbaz, 2023-07-30:
Support for non-wlroots based compositors is tracked in #5 btw, in case you want to subscribe for notifications 😉

Tim-C1, 2023-07-30:
> Support for non-wlroots based compositors is tracked in #5 btw, in case you want to subscribe for notifications wink

Ok, thanks.

closed 2023-07-30 by maximbaz