max-baz/wluma #88: [Feature request]: Support monitor disconnect

riley-martin, 2023-09-08:
### Please describe your feature request

When my laptop screen times out and turns off, wluma crashes with this error:  
`thread 'predictor-eDP-1' panicked at 'Frame was cancelled due to a permanent error. If you just disconnected screen, this is not implemented yet.', src/frame/capturer/wlroots.rs:142:25`.
I understand that this is currently expected, however, it would be nice to not have to restart wluma each time the display is powered on again.  

If this is not feasible, that's fine; but if it is, I would be willing to try to submit a PR. I have some experience with rust, but I have done almost no graphics work so I might need some pointers there.  

BTW, thanks for this great tool!

maximbaz, 2023-09-08:
I think it's very feasible, just not been prioritized because I'm using my laptop without any external screens :sweat_smile: PR would be most appreciated! 

You are more than welcome to have a look at the code, and then if you want we can talk about the approach, hopefully that can help you get onboarded and save some time.

One important thing to consider (for me in particular :sweat_smile:), is this PR: https://github.com/maximbaz/wluma/pull/86

This harmless idea to upgrade deps blew out of proportion, as I've seen that wlroots library has changed a lot in the way you are supposed to use it. I'm done somewhat, but I think it would be most valuable to finish that PR, before building things on top, just because it might significantly impact the approach for detecting screen change.

riley-martin, 2023-09-08:
>  just not been prioritized because I'm using my laptop without any external screens  

I am as well, its just that wluma seems to think that the built-in screen is disconnected when it goes to sleep; or am I misunderstanding and this is a separate problem?  

> I'm done somewhat, but I think it would be most valuable to finish that PR, before building things on top, just because it might significantly impact the approach for detecting screen change.  

Ok, makes sense.

maximbaz, 2023-09-08:
I might be hiding the problem for myself by running wluma as a systemd service with Restart set to always, so even if it crashes on sleep, it just self restarts on resume and I never notice this 😄 How are you running it, manually starting the process?

name-snrl, 2024-01-26:
My `swayidle` config turned off all outputs after 6 min, and sends `idlehint` after 10 minutes, with `IdleActionSec=10min` in logind.conf this suspends my system after 20 minutes, which means only 6 minutes out of 20 my screen is on. This causes wluma to crash. So yeah, I think this is an important issue for anyone using `swayidle` with default settings or something like that.

Hopefully, work on #86 is nearing completion.

> systemd service with Restart set to always

In my case it stops restarting after 6 attempts. There some logs - [journalctl.log](https://github.com/maximbaz/wluma/files/14064502/journalctl.log)

The reason is the default values for the `DefaultStartLimitIntervalSec` and `DefaultStartLimitBurst` options:

<https://www.freedesktop.org/software/systemd/man/latest/systemd-system.conf.html#DefaultStartLimitIntervalSec=>

I think this can be solved with `StartLimitIntervalSec` and `StartLimitBurst`, but I don't see a single solution for all possible cases.

I solved my case with a small change in the swayidle configuration:

```diff
 # DPMS
-timeout 360 'swaymsg "output * power off"' resume 'swaymsg "output * power on"'
+timeout 360 'systemctl --user stop wluma.service; swaymsg "output * power off"' \\
+  resume 'swaymsg "output * power on"; systemctl --user start wluma.service'
 # sleep
 before-sleep 'loginctl lock-session "$XDG_SESSION_ID"'
 idlehint 600 # +IdleActionSec in logind.conf
-after-resume 'swaymsg "output * power on"'
+after-resume 'swaymsg "output * power on"; systemctl --user start wluma.service'
```

max-baz, 2026-08-15:
This is now implemented on latest `main` and will be part of the next release, hotplug is now detected and handled correctly.

closed 2026-08-15 by max-baz