max-baz/wluma #89: [Bug]: Unable to initialize Vulkan: ERROR_EXTENSION_NOT_PRESENT

jschaeff, 2023-11-21:
### Steps for reproducing the issue

Launched wluma the first time.

### What is the buggy behavior?

Wluma panicks because of a missing extension.

### What is the expected behavior?

Wluma should work.

I followed some guides to install Vulkan, so I expect the extension to be available, somehow.

### Logs

```shell
RUST_LOG=debug wluma                                               
[2023-11-21T13:40:58Z DEBUG wluma] Using Config {
        als: Iio {
            path: "/sys/bus/iio/devices",
            thresholds: {
                800: "outdoors",
                20: "dark",
                0: "night",
                250: "normal",
                500: "bright",
                80: "dim",
            },
        },
        output: [
            Backlight(
                BacklightOutput {
                    name: "eDP-1",
                    path: "/sys/class/backlight/intel_backlight",
                    capturer: Wlroots,
                    min_brightness: 1,
                },
            ),
            Backlight(
                BacklightOutput {
                    name: "keyboard-dell",
                    path: "/sys/bus/platform/devices/dell-laptop/leds/dell::kbd_backlight",
                    capturer: None,
                    min_brightness: 0,
                },
            ),
        ],
    }
[2023-11-21T13:40:58Z INFO  wluma] Continue adjusting brightness and wluma will learn your preference over time.
thread 'predictor-eDP-1' panicked at src/frame/capturer/wlroots.rs:86:44:
Unable to initialize Vulkan: ERROR_EXTENSION_NOT_PRESENT
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

### Version

wluma 4.3.0-1 packaged by archlinux.

### Environment

```shell
Linux vogon 6.6.1-arch1-1 #1 SMP PREEMPT_DYNAMIC Wed, 08 Nov 2023 16:05:38 +0000 x86_64 GNU/Linux

Processes:
1958 sway
2031 swaybg
2061 swayrd
2062 swaync
2064 swayidle
2223 swaync-client
33009 swaync-client

Packaged installed: 
clang 16.0.6-1
rust 1:1.73.0-1
sway 1:1.8.1-3
swaybg 1.2.0-1
swayidle 1.8.0-1
swaylock-effects-git r420.7c5681c-1
swaync 0.9.0-1
swayr 0.27.1-1
vulkan-icd-loader 1.3.269-1
vulkan-tools 1.3.269-1
wlroots 0.16.2-2
```

jschaeff, 2023-11-21:
This seems related to a closed issue #38

jschaeff, 2023-11-21:
Yes, I confirm that wluma runs with "capturer=none". Too bad for the feature wich looks super great.

Thanks for your great work !

maximbaz, 2023-11-21:
Happy to help! Subscribe to #5 and #62, if they ever get done, who knows, maybe there will be another pathway for wluma to get screen contents on your hardware! Otherwise do not hesitate to reach out :wink:

closed 2023-11-21 by maximbaz