max-baz/wluma #92: configuration question

name-snrl, 2024-01-25:
### Please describe your feature request

Hi, there was no proper section for the question, so I'm creating feature request.

The only thing I need from wluma is described in the first sentence of the readme:

_The app will automatically brighten the screen when you are looking at a dark window (such as a fullscreen terminal) and darken the screen when you are looking at a bright window (such as web browser)._

I've NixOS on Thinkpad t440s. There are no errors in the log except `MESA-INTEL: warning: Haswell Vulkan support is incomplete`.

The configurations I've tried:

1) ```toml
   [als.none]
   
   [[output.backlight]]
   name = "eDP-1"
   path = "/sys/class/backlight/intel_backlight"
   capturer = "wlroots"
   ```
2) ```toml
   [als.webcam]
   video = 0
   thresholds = { 0 = "night", 15 = "dark", 30 = "dim", 45 = "normal", 60 = "bright", 75 = "outdoors" }
   
   [[output.backlight]]
   name = "eDP-1"
   path = "/sys/class/backlight/intel_backlight"
   capturer = "wlroots"
   ```
   
but nothing works, and with the second cfg I have an annoying webcam light blinking. WHat am I doing wrong?

maximbaz, 2024-01-25:
Hello! The app will learn your preference, but you must train it first 🙂 Open for example a dark app/webpage and increase brightness by hand until you like it, then switch to a white page and decrease brightness until it no longer blinds you - that's the simplest training! Now switch back to the dark page, and brightness should automatically increase. The more you train, the better it will be at predicting your preference.

Webcam is an optional additional data point, which allows you to have more advanced patterns - for example teaching wluma that you prefer startpage of github.com in full screen to be at 90% of brightness during the day (when it's bright around you) and 30% at night, all else being equal.

Hope it helps!

name-snrl, 2024-01-25:
> train it first

Yes, I read about training. And that's the first thing I did, but the only thing `wluma` does is change brightness on startup :c

Where does `wluma` store the trained data? Is `.local/share/wluma/eDP-1.yaml` correct path?

maximbaz, 2024-01-25:
Yeah - try to run it with RUST_LOG=trace, do you spot anything interesting then?

name-snrl, 2024-01-25:
> do you spot anything interesting then?

No, but I assume the problem is in service, when I run `wluma` in terminal everything works. Sorry, I thought I originally checked wluma in terminal and the problem persisted, but it seems that is not the case

closed 2024-01-25 by name-snrl

name-snrl, 2024-01-25:
@maximbaz w8, why do we have an EnvironmentFile option in the service?

https://github.com/maximbaz/wluma/blob/main/wluma.service#L9

The service.conf file is missing from the repo

maximbaz, 2024-01-25:
I believe it's an optional file, to ease configuration