max-baz/wluma #94: [Feature request]: Add gamma control

pyt0xic, 2024-01-27:
### Please describe your feature request

Add the ability to set screen temperature based on sunset/sundown and manually specified times. 

See #91

maximbaz, 2024-01-27:
One of the biggest questions is to figure out what we can do differently and better than gammastep. I don't think we should reimplement and compete with it just for the sake of it.

Our competitive advantage is that we have access to screen content, and optionally als. Is it enough to predict the best possible screen temperature?

pyt0xic, 2024-01-27:
Hmm, I think adjusting the gamma based on light levels would be quite cool.
In terms of screen content, it could maybe adjust it based on the amount of dark/light content. I'm not sure how beneficial that would be. 😅

maximbaz, 2024-01-28:
Maybe... On the other hand, isn't it in some sense preferable to "force" warm colors in the evening, to prepare your brain for sleep in advance, rather than keeping your screen blue just because you keep bright lights on, thereby feeding the vicious cycle that leads to staying awake until much too late? 😅😆

pyt0xic, 2024-01-28:
Very true. 😂
One thing that I would like and I believe @name-snrl would be interested in as well is the ability to specify temperatures for certain times. 

The sun rises very early here, but it becomes significantly brighter around 10am, so it would be nice to have more control over the "transitional" periods. I hope that makes sense...

maximbaz, 2024-01-28:
To continue playing the devil's advocate, wouldn't configuring the curve be better suited for gammastep project? They already have all the code ready and battle-tested, and it should be a small adjustment to teach it how to have a better transition curves? 

By the way, there's lots of project doing that, most recently I was using `wlsunset` instead of `gammastep` (my current hardware does not support `wlr-gamma-control-unstable-v1` :sob: )

name-snrl, 2024-01-29:
> Our competitive advantage is that we have access to screen content, and optionally als. Is it enough to predict the best possible screen temperature?

I'm not sure of the prediction, but I think that by adding gamma functionality to `wluma` we will be able to adjust brightness more finely. e.g. with warm colors, white is less annoying at the same brightness level

maximbaz, 2024-01-29:
Now that is a very interesting and tempting idea to explore, that a screen color would allow for a finer tuning and therefore better predictions. Do you think you would want warmer colors during the day? You lose potentially correctness of colors...

name-snrl, 2024-01-29:
> Do you think you would want warmer colors during the day? 

Nah, I don't need this. All I want I've described here

pyt0xic, 2024-02-03:
> Now that is a very interesting and tempting idea to explore, that a screen color would allow for a finer tuning and therefore better predictions.

I really like this idea!

> Do you think you would want warmer colors during the day? You lose potentially correctness of colors...

I use slightly warmer colors during the day, I stare at a terminal most of the time and I find it helps prevent eye strain.

I think I've worked out how to use the Wayland gamma control protocol, I'll add it to my fork as soon as I am able to set the gamma :laughing:

jficz, 2025-03-16:
I'd be happy with just relative gamma adjustment (i.e. "brightness"), not necessarily color temperature. My use case is an external display which is very bright even at lowest settings and adjusting gamma is the only way to dim it further.

pyt0xic, 2025-03-17:
> I'd be happy with just relative gamma adjustment (i.e. "brightness"), not necessarily color temperature. My use case is an external display which is very bright even at lowest settings and adjusting gamma is the only way to dim it further.
> 
> Think of it as an alternative "fallback" solution to DDC.

Can you not use DDC/CI?
Check out `ddcutil`....

jficz, 2025-03-18:
> Can you not use DDC/CI?

Not really, even with the ddcci driver the interface is buggy. The external monitor is randomly (not)detected by ddcutil and sometimes doesn't even come up when ddcci module is loaded.

jficz, 2025-04-16:
So ddc "somehow" works after applying a bit of dark magic but the problem remains - even the lowest brightness settings of the monitor is too bright.

Also, settings per sun up/down based on (even hardcoded) geolocation would be cool.

max-baz, 2026-08-16:
Gamma control is now implemented in `main` and will be part of the next release. You can control both temperature (e.g. `wluma set temperature eDP-1 3500K`) and dim (e.g. `wluma set dim DP-1 30%`), and then those values are used together to calculate the final gamma adjustment. Brightness, dim and temperature are following the exact same independent learning algorithm, so just like before you continue adjusting those values and wluma should learn your preference over time. Gamma requires exclusive control, so you can't run another app to do it, that's why you need to make these adjustments via wluma cli. Unlike screen brightness and dim, temperature doesn't take screen contents into account, only ambient light.

closed 2026-08-16 by max-baz