max-baz/wluma #100: [Bug]: ALS data not parsed/calculated correctly

RubenKelevra, 2024-03-07:
### Steps for reproducing the issue

I've set up wluma on a system of a friend, which got an ambient light sensor. But it doesn't seem to calculate the data provided correctly.

I've looked at the `src/als/iio.rs` and the values which seems to be read by wluma are the following on the system:

```sh
$ LANG=en_US.UTF-8 grep "" /sys/bus/iio/devices/iio\:device1/*
grep: /sys/bus/iio/devices/iio:device1/buffer: Is a directory
grep: /sys/bus/iio/devices/iio:device1/buffer0: Is a directory
/sys/bus/iio/devices/iio:device1/current_timestamp_clock:realtime
/sys/bus/iio/devices/iio:device1/dev:238:1
grep: /sys/bus/iio/devices/iio:device1/in_illuminance_hysteresis: Invalid argument
/sys/bus/iio/devices/iio:device1/in_illuminance_hysteresis_relative:0.010000
/sys/bus/iio/devices/iio:device1/in_illuminance_offset:0
/sys/bus/iio/devices/iio:device1/in_illuminance_raw:7056
/sys/bus/iio/devices/iio:device1/in_illuminance_sampling_frequency:10.000000
/sys/bus/iio/devices/iio:device1/in_illuminance_scale:0.001000000
/sys/bus/iio/devices/iio:device1/in_intensity_both_raw:7056
grep: /sys/bus/iio/devices/iio:device1/in_intensity_hysteresis: Invalid argument
/sys/bus/iio/devices/iio:device1/in_intensity_hysteresis_relative:0.010000
/sys/bus/iio/devices/iio:device1/in_intensity_offset:0
/sys/bus/iio/devices/iio:device1/in_intensity_sampling_frequency:10.000000
/sys/bus/iio/devices/iio:device1/in_intensity_scale:0.001000000
/sys/bus/iio/devices/iio:device1/name:als
grep: /sys/bus/iio/devices/iio:device1/power: Is a directory
grep: /sys/bus/iio/devices/iio:device1/scan_elements: Is a directory
grep: /sys/bus/iio/devices/iio:device1/subsystem: Is a directory
grep: /sys/bus/iio/devices/iio:device1/trigger: Is a directory
/sys/bus/iio/devices/iio:device1/uevent:MAJOR=238
/sys/bus/iio/devices/iio:device1/uevent:MINOR=1
/sys/bus/iio/devices/iio:device1/uevent:DEVNAME=iio:device1
/sys/bus/iio/devices/iio:device1/uevent:DEVTYPE=iio_device
```

As far as I can tell, the sensor itself works correctly:

```sh
# darkly lit room
$ cat /sys/bus/iio/devices/iio\:device1/in_illuminance_raw 
672
# brightly lit room
$ cat /sys/bus/iio/devices/iio\:device1/in_illuminance_raw 
39648
# normal lit room
$ cat /sys/bus/iio/devices/iio\:device1/in_illuminance_raw 
11088
# lights off (night time)
$ cat /sys/bus/iio/devices/iio\:device1/in_illuminance_raw 
0
```

Settings:

```
[als.iio]
path = "/sys/bus/iio/devices/"
thresholds = { 0 = "night", 20 = "dark", 80 = "dim", 250 = "normal", 500 = "bright", 800 = "outdoors" }

[[output.backlight]]
name = "eDP-1"
path = "/sys/class/backlight/intel_backlight"
capturer = "none"
```

### What is the buggy behavior?

wluma does nothing when started, except setting the brightness once.

```
$ RUST_LOG=trace wluma
[2024-03-07T19:49:00Z DEBUG wluma] Using Config {
        als: Iio {
            path: "/sys/bus/iio/devices/",
            thresholds: {
                500: "bright",
                800: "outdoors",
                250: "normal",
                0: "night",
                20: "dark",
                80: "dim",
            },
        },
        output: [
            Backlight(
                BacklightOutput {
                    name: "eDP-1",
                    path: "/sys/class/backlight/intel_backlight",
                    capturer: None,
                    min_brightness: 1,
                },
            ),
        ],
    }
[2024-03-07T19:49:00Z INFO  wluma] Continue adjusting brightness and wluma will learn your preference over time.
[2024-03-07T19:49:00Z TRACE wluma::als::iio] ALS (iio): night (6)
[2024-03-07T19:49:00Z TRACE wluma::predictor::controller] Prediction: 750 (lux: night, luma: 0)
[2024-03-07T19:49:00Z TRACE wluma::als::iio] ALS (iio): night (6)
[2024-03-07T19:49:00Z TRACE wluma::predictor::controller] Prediction: 750 (lux: night, luma: 0)
[2024-03-07T19:49:00Z TRACE wluma::als::iio] ALS (iio): night (6)
[2024-03-07T19:49:00Z TRACE wluma::predictor::controller] Prediction: 750 (lux: night, luma: 0)
[2024-03-07T19:49:01Z TRACE wluma::als::iio] ALS (iio): night (6)
[2024-03-07T19:49:01Z TRACE wluma::predictor::controller] Prediction: 750 (lux: night, luma: 0)
[2024-03-07T19:49:01Z TRACE wluma::als::iio] ALS (iio): night (6)
[2024-03-07T19:49:01Z TRACE wluma::predictor::controller] Prediction: 750 (lux: night, luma: 0)
[2024-03-07T19:49:01Z TRACE wluma::als::iio] ALS (iio): night (6)
[2024-03-07T19:49:01Z TRACE wluma::predictor::controller] Prediction: 750 (lux: night, luma: 0)
[2024-03-07T19:49:01Z TRACE wluma::als::iio] ALS (iio): night (6)
[2024-03-07T19:49:01Z TRACE wluma::predictor::controller] Prediction: 750 (lux: night, luma: 0)
[2024-03-07T19:49:01Z TRACE wluma::als::iio] ALS (iio): night (6)
[2024-03-07T19:49:01Z TRACE wluma::predictor::controller] Prediction: 750 (lux: night, luma: 0)
[2024-03-07T19:49:02Z TRACE wluma::als::iio] ALS (iio): night (6)
[2024-03-07T19:49:02Z TRACE wluma::predictor::controller] Prediction: 750 (lux: night, luma: 0)
```

From my interpretation wluma seems to read a constant 0 from the sensor, right?

### What is the expected behavior?

Reading different values from the sensor and reacting to it.

### Logs

```shell
Debug log looks uneventful:

$ RUST_LOG=debug wluma
[2024-03-07T19:51:02Z DEBUG wluma] Using Config {
        als: Iio {
            path: "/sys/bus/iio/devices/",
            thresholds: {
                0: "night",
                500: "bright",
                800: "outdoors",
                20: "dark",
                250: "normal",
                80: "dim",
            },
        },
        output: [
            Backlight(
                BacklightOutput {
                    name: "eDP-1",
                    path: "/sys/class/backlight/intel_backlight",
                    capturer: None,
                    min_brightness: 1,
                },
            ),
        ],
    }
[2024-03-07T19:51:02Z INFO  wluma] Continue adjusting brightness and wluma will learn your preference over time.
```
```

### Version

4.3.0 from the AUR

### Environment

```shell
LSB Version:    n/a
Distributor ID: EndeavourOS
Description:    EndeavourOS Linux
Release:        rolling
Codename:       rolling
Linux stylistic-r727 6.7.8-arch1-1 #1 SMP PREEMPT_DYNAMIC Sun, 03 Mar 2024 00:30:36 +0000 x86_64 GNU/Linux
egrep: warning: egrep is obsolescent; using grep -E
vulkan-headers 1:1.3.279-1
vulkan-icd-loader 1.3.279-1
vulkan-tools 1.3.269-1
wlroots 0.17.1-1
bash: dpkg: command not found
egrep: warning: egrep is obsolescent; using grep -E
```

maximbaz, 2024-03-07:
Hello!

Just to confirm something first, have you actually tried to train wluma a bit? Going in a dark place, adjusting the screen brightness by hand, going to a bright environment and increasing the brightness again by hand? It does nothing when it doesn't know your preferences :)

maximbaz, 2024-06-16:
Closing due to inactivity, if you have some updates, please share and let's reopen the thread :+1:

closed 2024-06-16 by maximbaz