max-baz/wluma #104: [Bug]: Unable to start wluma from the command line

joaocandre, 2024-05-02:
### Steps for reproducing the issue

Start `wluma` from the command line.

### What is the buggy behavior?

Error message:

    thread 'predictor-eDP-1' panicked at src/frame/capturer/wlroots.rs:80:14:
    Unable to init export_dmabuf_manager: Missing

### What is the expected behavior?

Display brightness-backlight should be adjusted based on ambient light and screen content; at the very least, `wluma`should have kept running.

### Logs

```shell
[2024-05-02T16:36:26Z DEBUG wluma] Using Config {
        als: Webcam {
            video: 0,
            thresholds: {
                45: "normal",
                30: "dim",
                60: "bright",
                75: "outdoors",
                15: "dark",
                0: "night",
            },
        },
        output: [
            Backlight(
                BacklightOutput {
                    name: "eDP-1",
                    path: "/sys/class/backlight/intel_backlight",
                    capturer: Wlroots,
                    min_brightness: 1,
                },
            ),
            DdcUtil(
                DdcUtilOutput {
                    name: "ASUS VX238H",
                    capturer: Wlroots,
                    min_brightness: 1,
                },
            ),
            Backlight(
                BacklightOutput {
                    name: "keyboard-dell",
                    path: "/sys/bus/platform/devices/dell-laptop/leds/dell::kbd_backlight",
                    capturer: None,
                    min_brightness: 0,
                },
            ),
        ],
    }
thread 'predictor-eDP-1' panicked at src/frame/capturer/wlroots.rs:80:14:
Unable to init export_dmabuf_manager: Missing
stack backtrace:
note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.
```

### Version

`4.3.0-1`,  running on an up-to-date Archlinux machine, installed from the AUR;
User is part of `video`group, and udev rules are installed on `/usr/lib/udev/rules.d`.

### Environment

```shell
LSB Version:	n/a
Distributor ID:	Arch
Description:	Arch Linux
Release:	rolling
Codename:	n/a
Linux joao-laptop 6.8.8-arch1-1 #1 SMP PREEMPT_DYNAMIC Sun, 28 Apr 2024 15:59:47 +0000 x86_64 GNU/Linux
egrep: warning: egrep is obsolescent; using grep -E
clang 17.0.6-2
lib32-vulkan-icd-loader 1.3.279-1
lib32-vulkan-intel 1:24.0.6-2
rustup 1.27.0-1
vulkan-headers 1:1.3.279-1
vulkan-icd-loader 1.3.279-1
vulkan-intel 1:24.0.6-2
bash: dpkg: command not found
egrep: warning: egrep is obsolescent; using grep -E
```

maximbaz, 2024-05-02:
Hello, Gnome is not based on wlroots, so screen capturing cannot be initialized, and that's why the app exists. You should change capturer to "none" in your config. 

https://github.com/maximbaz/wluma?tab=readme-ov-file#displays

However you should be aware that this will mean that wluma will not be able to take into account the content of your screen, when deciding on the ideal screen brightness. I can also see you don't have an ALS sensor, and you are using webcam approximation, which is the only data source for wluma to work with.

joaocandre, 2024-05-02:
Indeed, using `none` fixed the issue. Shame that screen contents can't be captured under Gnome, any plans to support an alternative method?

closed 2024-05-02 by joaocandre

maximbaz, 2024-05-02:
#5, but nobody is working on it at the moment