max-baz/wluma #108: [Feature request]: print more debug info

Atemu, 2024-05-22:
### Please describe your feature request

In the debug log (`RUST_LOG=debug`) it currently only prints the initial config and new data points (lux class, content(?) luma, brightness) which I find a bit lacking.

For debugging, it'd be really useful to also see:

1. Brightness change actions taken
  I had the case of GNOME's auto-brightness feature suddenly appearing and accidentally running two instances of wluma where brightness would change erratically. This would have been a lot easier to debug if the wluma instance I was looking at printed a line of debug info every time it actually took an action.
2. Brightness value read/inferred from the sensor
  When setting up wluma, it's not obvious how to set up the lux "classes" (dim, normal, bright etc.) because you don't have a reference. You could start reading the device's output manually in addition to the debug log but this isn't necessarily trivially possible with webcam sources and even with iio sources it'd be really convenient to just have that info at that point in time.

maximbaz, 2024-05-22:
hello! I'm on a phone now, but I'm pretty sure the `trace` log level is printing exactly the info you are looking for, could you give it a go?

Atemu, 2024-05-22:
Indeed it is. I mistakenly thought debug was the highest level.

For reference, it outputs a constat stream of logs of this pattern

```
[2024-05-22T13:48:56Z TRACE wluma::predictor::controller] Prediction: 627 (lux: normal, luma: 0)
[2024-05-22T13:48:56Z TRACE wluma::als::iio] ALS (iio): normal (8)
[2024-05-22T13:48:56Z TRACE wluma::als::iio] ALS (iio): normal (8)
[2024-05-22T13:48:56Z TRACE wluma::predictor::controller] Prediction: 627 (lux: normal, luma: 0)
[2024-05-22T13:48:56Z TRACE wluma::als::iio] ALS (iio): normal (8)
[2024-05-22T13:48:56Z TRACE wluma::als::iio] ALS (iio): normal (8)
[2024-05-22T13:48:56Z TRACE wluma::predictor::controller] Prediction: 627 (lux: normal, luma: 0)
[2024-05-22T13:48:56Z TRACE wluma::als::iio] ALS (iio): normal (8)
[2024-05-22T13:48:56Z TRACE wluma::als::iio] ALS (iio): normal (8)
```

While trace does print the brightness wluma gets from the sensors, it does not print the actions taken. This would still be useful to have.

maximbaz, 2024-06-16:
Hi again, sorry I forgot to reply, ` Prediction: 627` is the action being taken, wluma is writing the value `627` into your brightness raw file.

maximbaz, 2024-10-21:
Will close, but if you still miss a particular info from the trace log, let me know.

closed 2024-10-21 by maximbaz