max-baz/wluma #110: How do I adjust the brightness

Visual-Dawg, 2024-06-07:
### Steps for reproducing the issue

Probably a stupid question, but how do I change the brightness after running wluma?

### What is the buggy behavior?

-

### What is the expected behavior?

=

### Logs

```shell
-
```

### Version

-

### Environment

```shell
-
```

maximbaz, 2024-06-07:
Before wluma learned your preference, you can adjust brightness using any other tool, like brightnessctl, light or whatever else you were using before you found wluma! With time, as wluma learns more and more about your preferences, you will hopefully find yourself adjusting brightness using those other tools less and less 🙂

Visual-Dawg, 2024-06-11:
Great, thank you! :)

closed 2024-06-11 by Visual-Dawg