max-baz/wluma #111: [Bug]: Unable to keep wlroots for capturer even though i use Hyprland

Rishik-Y, 2024-06-13:
### Steps for reproducing the issue

Running `Wluma` on Hyprland

There is no issue, and the wluma works perfectly fine when keeping `capture = "none"`

but doesn't really work when keeping `capture = "wlroots"`

I was till now stuck in finding out how to set it up exactly properly 

but thanks to https://github.com/maximbaz/wluma/issues/40

It was way easier, but one thing I am stuck on is that https://github.com/vega-d had gnome installed which doesn't have wlroots

While I have hyprland installed which I am pretty sure is based on wlroots Compositor 

I feel like I am missing something since everything works just right when keeping `capture = "none"` and using my webcam as sensor.

Can anybody tell me what exactly is the issue here.

### What is the buggy behavior?

```
❯ wluma
[2024-06-13T13:54:27Z INFO  wluma] Continue adjusting brightness and wluma will learn your preference over time.
thread 'predictor-eDP-1' panicked at src/frame/capturer/wlroots.rs:80:14:
Unable to init export_dmabuf_manager: Missing
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

### What is the expected behavior?

```
❯ wluma
[2024-06-13T13:51:29Z INFO  wluma] Continue adjusting brightness and wluma will learn your preference over time.
```

### Logs

```shell
░▒▓   ~  
❯ RUST_LOG=debug wluma
[2024-06-13T13:58:18Z DEBUG wluma] Using Config {
        als: Webcam {
            video: 0,
            thresholds: {
                30: "dim",
                60: "bright",
                75: "outdoors",
                15: "dark",
                0: "night",
                45: "normal",
            },
        },
        output: [
            Backlight(
                BacklightOutput {
                    name: "eDP-1",
                    path: "/sys/devices/pci0000:00/0000:00:08.1/0000:07:00.0/drm/card2/card2-eDP-1/amdgpu_bl2",
                    capturer: Wlroots,
                    min_brightness: 1,
                },
            ),
            Backlight(
                BacklightOutput {
                    name: "keyboard-dell",
                    path: "/sys/class/net/enp4s0/enp4s0-2::lan",
                    capturer: None,
                    min_brightness: 0,
                },
            ),
        ],
    }
[2024-06-13T13:58:18Z INFO  wluma] Continue adjusting brightness and wluma will learn your preference over time.
thread 'predictor-eDP-1' panicked at src/frame/capturer/wlroots.rs:80:14:
Unable to init export_dmabuf_manager: Missing
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

### Version

4.4.0-1

### Environment

```shell
░▒▓   ~  
❯ lsb_release -a; uname -a; pgrep -l sway; pacman -Q | egrep "(wlroots|vulkan|sway|clang|rust)"; dpkg -l | egrep "(wlroots|vulkan|sway|clang|rust)"
bash: lsb_release: command not found
Linux archlinux 6.9.3-arch1-1 #1 SMP PREEMPT_DYNAMIC Fri, 31 May 2024 15:14:45 +0000 x86_64 GNU/Linux
egrep: warning: egrep is obsolescent; using grep -E
clang 17.0.6-2
rust 1:1.78.0-1
swaylock-effects 1.7.0.0-4
swaylock-effects-debug 1.7.0.0-4
vulkan-headers 1:1.3.285-1
vulkan-icd-loader 1.3.285-1
vulkan-intel 1:24.1.1-1
vulkan-radeon 1:24.1.1-1
vulkan-validation-layers 1.3.275-1
wlroots 0.17.3-1
bash: dpkg: command not found
egrep: warning: egrep is obsolescent; using grep -E
```

maximbaz, 2024-06-13:
Hello! It doesn't sound like you are missing a dependency or something like that, and there have been people writing in this project who were using wluma on hyprland at the time, so it definitely used to work. Hyprland actually runs on a fork of wlroots, so it's possible that it differs too much from vanilla wlroots today, or that there's a bug somewhere (also possibly in wluma code!).

https://github.com/maximbaz/wluma/issues/88#issuecomment-1711797135

I'll try my luck: @riley-martin do you still use wluma and hyprland?

Rishik-Y, 2024-06-13:
Thanks very much for looking into it.
I like using Wluma, but I really don't know where the hiccup I am getting stuck is.
I will post if I do find anything regarding this.

riley-martin, 2024-06-15:
> I'll try my luck: @riley-martin do you still use wluma and hyprland?  

I don't use hyprland much anymore. I'd be glad to test it when I get a chance but it might not happen right away.

riley-martin, 2024-06-16:
I did some testing tonight and it didn't work for me either. Looking at some recent release notes ([0.40.0](https://github.com/hyprwm/Hyprland/releases/tag/v0.40.0) and [0.41.0](https://github.com/hyprwm/Hyprland/releases/tag/v0.41.0)) it appears that hyprland is switching to their own wayland implementation instead of using wlroots, so I guess that would be the problem.

maximbaz, 2024-06-16:
Oh great catch, thanks for pointing it out! 

We can conclude that wluma's `wlroots` capturer is not expected to work on Hyprland anymore.

We can leave this issue open as a feature request, in case someone from community would like to implement a new capturer for Hyprland (perhaps wait until Hyprland completes their transition away from wlroots first).

Thanks @Rishik-Y for spotting the issue and @riley-martin for tracking down the root cause!

Rishik-Y, 2024-06-16:
> I did some testing tonight and it didn't work for me either. Looking at some recent release notes ([0.40.0](https://github.com/hyprwm/Hyprland/releases/tag/v0.40.0) and [...]

Oh, Right, I totally didn't see that,
All this while, I have been breaking myself, thinking I may be bad at understanding or I simply broke stuff somewhere.
It seems the issue was simply something that had been in my blind spot.

Thank you, @riley-martin, for checking up on this issue and @maximbaz, for quickly responding to my issue. 
Finally, I can rest at ease now.

Rishik-Y, 2024-07-10:
Hey @maximbaz, I have been trying to understand the code and how it works for the past few weeks and have  _kind of_ found the exact issue a while back. 

To better give the info _from What I understood_.

The issue shown on my screen is:
`Unable to init export_dmabuf_manager: Missing at line 80 `
Which is in wluma/src/frame/capturer/wlroots.rs.

Which is the error when it executes
```
        let dmabuf_manager = globals
            .instantiate_exact::<ZwlrExportDmabufManagerV1>(1) 
```
Which is at line 78.

I have tried to understand where does the ZwlrExportDmabufManagerV1 initiates from, but it seems that main.rs doesn't have much relation to wlroots, so the first line starts from

```
use wayland_protocols::wlr::unstable::export_dmabuf::v1::client::{
    zwlr_export_dmabuf_frame_v1::{CancelReason, Event},
    zwlr_export_dmabuf_manager_v1::ZwlrExportDmabufManagerV1,
};
```
Which is on line 8 of wlroots.rs.

As @riley-martin stated, Hyprland Had changed the compositor from original wlroots to modified wlroots.

I tried to investigate what exactly was modified in the directories and found out that the directories were changed 
`/usr/include/wlr/types/wlr_export_dmabuf_v1.h` instead of `wlroots` folder
(Which is not that much of an issue...   **yet**) 
but wlr_export_dmabuf_v1.h contains the code very similar to the original wlr-export-dmabuf-unstable-v1.h (_I think?_)
 Where instead of zwlr_export_dmabuf_manager_v1, it contains wlr_export_dmabuf_manager_v1 everywhere, and some minor changes are made.

I have thought about changing the file itself, but I know it will instead break my system apart, so I tried finding ways to change the code for Hyprland Users, but the wluma program uses a crate called wlroots-Protocols which gives the program much of its wlroots functionality. 

I believed that it shouldn't be much of a hassle to fix this, but it turns out there is no crate created for Hyprland Modified wlroots. 

I tried finding out using other crates whether it would work or not by adding the wlroots-protocol-wlr crate into the Cargo.toml and modified a bit 

like in line 8 
I kept,
`use wayland_protocols_wlr::export_dmabuf::v1::client::{`
_(For trial and error)_

And the program seems to detect it by showing the error in other code instead of that line where I need to modify to suit it with the use wayland-protocols-wlr, but even then 
_(from what I think)_ it wont work because even in that the code is written as `zwlr_export_dmabuf_manager_v1`, and i really don't know now what to try next.

I tried asking Chatgpt, and it states that to make a fork of `wayland-protocol-wlr` crate and and modify there, and add that in `cargo.toml`, which seems again very complicated to me.

I Have become very obsessed with trying to make this program run on Hyprland.
I even started learning rust in the past three days just so I could make it work.

I would really appreciate your thoughts on this, where I am assuming something is wrong, and what I should be doing next.

I can understand Rust somewhat since I already know a bit of C/C++, Python, Java and a bit of Haskell. 
But should I complete learning the basics of rust first? (it would probably take a few more days)

**Note: 
I am not asking you to fix it; I am simply asking you to give me any tips on what I can do to contribute here and make wlroots work on my Hyprland. Your expertise in rust would be greatly appreciated.**

_PS: Please don't be too harsh with me; I am just a second-year College student._

maximbaz, 2024-07-10:
Nice work and analysis, @Rishik-Y! 

First of all, it looks like the `dmabuf` protocol [has been marked stable](https://gitlab.freedesktop.org/wayland/wayland-protocols/-/blob/main/unstable/linux-dmabuf/linux-dmabuf-unstable-v1.xml#L94-96), so in `wlroots.rs` we should remove the `::unstable` parts (and probably `cargo update` the corresponding crate). The struct names [still start with `Zwlr`](https://docs.rs/wayland-protocols-wlr/0.3.2/wayland_protocols_wlr/?search=ExportDmabufManager), so perhaps you don't even need to change anything more than the import.

It might be that this change alone fixes wluma for Hyprland, if the only difference they did for now was to get rid of the now obsolete "unstable" version of the `dmabuf` protocol. Before we discuss more ideas, could you try this?

Rishik-Y, 2024-07-10:
Thanks for the suggestion @maximbaz, 
I already tried removing the `::unstable` tag (Through trial and error ☺️)
So unfortunately, it doesn't work.

Also I am sorry for my wording if it was not explained by me properly, but what I meant was 
ZwlrExportDmabufManagerV1 is the same here with wlrExportDmabufManagerV1 instead.

Here is the attachment of the code for `wlr_export_dmabuf_v1.h`
```
/*
 * This an unstable interface of wlroots. No guarantees are made regarding the
 * future consistency of this API.
 */
#ifndef WLR_USE_UNSTABLE
#error "Add -DWLR_USE_UNSTABLE to enable unstable wlroots features"
#endif

#ifndef WLR_TYPES_WLR_EXPORT_DMABUF_V1_H
#define WLR_TYPES_WLR_EXPORT_DMABUF_V1_H

#include <stdbool.h>
#include <wayland-server-core.h>
#include <wlr/render/dmabuf.h>

struct wlr_export_dmabuf_manager_v1 {
	struct wl_global *global;
	struct wl_list frames; // wlr_export_dmabuf_frame_v1.link

	struct wl_listener display_destroy;

	struct {
		struct wl_signal destroy;
	} events;
};

struct wlr_export_dmabuf_frame_v1 {
	struct wl_resource *resource;
	struct wlr_export_dmabuf_manager_v1 *manager;
	struct wl_list link; // wlr_export_dmabuf_manager_v1.frames

	struct wlr_output *output;

	bool cursor_locked;

	struct wl_listener output_commit;
	struct wl_listener output_destroy;
};

struct wlr_export_dmabuf_manager_v1 *wlr_export_dmabuf_manager_v1_create(
	struct wl_display *display);

#endif
```

Here are the files which were supposed to work wluma However I believe is modified by hyprland, Hence not working anymore.
[image]

I don't know whether it's just me making rookie mistakes, but from what I understood these are the modified wlroots used by Hyprland.
Really appreciate any tip.

maximbaz, 2024-07-10:
It looks like the file is actually identical to the wlroots sources :thinking: 

https://gitlab.freedesktop.org/wlroots/wlroots/-/blob/master/include/wlr/types/wlr_export_dmabuf_v1.h

Don't worry this is definitely not rookie mistakes, from what we see so far it looks like the header files of the `dmabuf` did not change between wlroots and Hyprland (yet), so there must be something else that has changed or has broken :thinking: I think the `Z` in `ZwlrExportDmabufManagerV1` means something else during the transition from C to Rust, as you can see `Z` is also absent in the C sources above from wlroots repo.

Rishik-Y, 2024-07-10:
Oh, I totally didn't Know that.
Thanks for Checking up @maximbaz 
I would have wasted a lot of time on this, if it weren't for you.
I will continue to find the actual issue on where its causing.

Rishik-Y, 2024-07-12:
Hello @maximbaz, So sorry that i keep disturbing you again and again.
I just want to ask, 
Was there any specific reason why you decided to use `wayland_protocolsV0.29.5` (Which includes wlr modules)
or 
Will it be fine if I update the version of the `wayland_protocols` to `V0.30.0 or higher` (Which separated `wlr` modules to `wayland_protocols_wlr` extension, Assuming I am able to integrate it in the wlroots.rs file)

maximbaz, 2024-07-12:
Don't worry, ask as much as you need 😁

The reason is that 0.29 was latest when I wrote that code initially, and migrating to 0.30 was a bit tricky so I never succeeded to make that work 😭

https://github.com/maximbaz/wluma/pull/86

maximbaz, 2024-07-12:
That is all to say, if you want to give it a go, please do!

Rishik-Y, 2024-07-12:
> Don't worry, ask as much as you need 😁

Thank you very much. ☺️
> 
> The reason is that 0.29 was latest when I wrote that code initially, and migrating to 0.30 was a bit tricky so I never succeeded to make that work 😭
> 

I had initially assumed that you probably chose that specific version probably due to wlr being incorporated in wayland-protocols or having incompatibility issues in further versions (at least, I hoped That was the case so that I would indirectly have a reason not to touch the complicated stuff)
But it seems that's not the case. 🥲
> That is all to say, if you want to give it a go, please do!

I know I will suffer a lot while figuring things around,
However, I have become somewhat fixated on this project, so I will try migrating it to 0.30 _**or try other ways**_ to see if it will be fixed for my Hyprland. 😁
Again thank you for your response.

daniel-sousa-me, 2024-07-12:
I just found this project after weeks of searching for something like this. I got a new laptop with an oled screen. It is amazing not to have a light-bulb constantly pointed at my face, but now it is extra annoying when I open a bright webpage. The proper way to fix this would probably be on the firmware's brightness control, but it's closed source, so a hack like this is probably required :(

Unfortunately I'm running into this same problem.

@maximbaz Which wm do you use? I don't know rust, but I'd like to try to make this work.

From what I'm reading here, it seems like it would make sense to try to upgrade the wayland_protocols. Even if it doesn't solve our problem with hyprland, at least we'd have a contribution for this project :)

maximbaz, 2024-07-12:
I'm using sway, and yeah I agree finishing #86 would be quite awesome, and help is certainly appreciated as I dont know when I can get to it... Splitting upgrade into wayland and not wayland to make it simpler could be a good way to start

daniel-sousa-me, 2024-07-12:
I'm new to wayland. I thought hyprland and sway were different levels and that hyprland was using sway.

I've found this program is also grabbing frames with the same rust library, but using the new version:
https://github.com/russelltg/wl-screenrec/blob/a3d0b5da69487441ce81ebfc3be8ba8d2255c714/src/main.rs

It works on hyprland.

I'm trying to use it as reference to update wluma's code, but it's going slowly because I have no idea what I'm doing

Rishik-Y, 2024-08-02:
Hi @daniel-sousa-me, Nice to see you here as well trying to help fix the issue.

Also Hello @maximbaz 
It's been a few weeks and I have been learning rust and trying ways to fix the issue but turns out the Hyprland has completely migrated from wlroots-based wayland compositor written over their [website](https://hyprland.org/news/independentHyprland/)

But they do claim that the wlroots apps should continue to work which is believable in my opinion as today again as I was trying and testing wluma the issue of wluma crashing 

AKA:
```
❯ wluma
[2024-06-13T13:54:27Z INFO  wluma] Continue adjusting brightness and wluma will learn your preference over time.
thread 'predictor-eDP-1' panicked at src/frame/capturer/wlroots.rs:80:14:
Unable to init export_dmabuf_manager: Missing
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```
Has now gone 
```

wluma
[2024-08-02T10:38:11Z WARN  wluma] Skipping 'eDP-1' as it might be disconnected: No such file or directory (os error 2)
[2024-08-02T10:38:11Z INFO  wluma] Continue adjusting brightness and wluma will learn your preference over time.

```
(However, another issue has arisen but seems a minor one)

So It seems that the issue will soon be **fixed either by a Hyprland update/by making some small modification in code by us**, or **Wluma will soon be broken completely on Hyprland.**

I can't confirm which will happen as my Hyprland is currently V0.41.2-3 and V0.42 is yet to be released (outside of Git)

In any case, 
I was wondering where the error 
`[2024-08-02T10:38:11Z WARN  wluma] Skipping 'eDP-1' as it might be disconnected: No such file or directory (os error 2)`
came from since i don't believe i made an error in the config file

Here is my config file (I Once configured it but now i totally forgot)
```
#[als.iio]
#path = "/sys/bus/iio/devices"
#thresholds = { 0 = "night", 20 = "dark", 80 = "dim", 250 = "normal", 500 = "bright", 800 = "outdoors" }

[als.webcam]
video = 0
thresholds = { 0 = "night", 15 = "dark", 30 = "dim", 45 = "normal", 60 = "bright", 75 = "outdoors" }

# [als.time]
# thresholds = { 0 = "night", 7 = "dark", 9 = "dim", 11 = "normal", 13 = "bright", 16 = "normal", 18 = "dark", 20 = "night" }

# [als.none]

[[output.backlight]]
name = "eDP-1"
path = "/sys/devices/pci0000:00/0000:00:08.1/0000:07:00.0/drm/card2/card2-eDP-1/amdgpu_bl2"
capturer = "wlroots"

# [[output.ddcutil]]
# name = "Dell Inc. DELL P2415Q"
# capturer = "none"

[[keyboard]]
name = "keyboard-MSI"
path = "/sys/class/net/enp4s0/enp4s0-2::lan"

# [[keyboard]]
# name = "keyboard-thinkpad"
# path = "/sys/bus/platform/devices/thinkpad_acpi/leds/tpacpi::kbd_backlight"
```

Rishik-Y, 2024-08-02:
Hi @daniel-sousa-me, Nice to see you here as well trying to help fix the issue.

Also Hello @maximbaz 
It's been a few weeks and I have been learning rust and trying ways to fix the issue but turns out the Hyprland has completely migrated from wlroots-based wayland compositor written over their [website](https://hyprland.org/news/independentHyprland/)
But they do claim that the wlroots apps should continue to work.

In any case, The issue is still persistent, and I am still continuing to find a fix for my laptop.

I was just wondering whether updating the version of the toml files truly be a fix for it or should I go in some other route trying other ways instead?

wolf-yuan-6115, 2024-08-02:
> But they do claim that the wlroots apps should continue to work.
> 

I think it's not referring to everything depending on wlroots work, they are talking about "apps" here, it might only mention apps that use wlroots to display content. (Not sure, just my guess)

wluma doesn't display any GUI, it's capturing the display, so it's probably not in that scope.

maximbaz, 2024-08-02:
I'm reasonably sure the primary reason for it not working on Hyprland is that the current code uses "unstable" dmabuf protocol (because it was unstable at the time of writing it), and Hyprland after the rewrite only announces the support for the stable one:

https://github.com/hyprwm/Hyprland/blob/1fa4b7d79baaad47fde8e72221cd77f569fbfe35/CMakeLists.txt#L307

We should really upgrade `wayland_protocols` in this repo, and then hopefully it will work, unless then we discover some other incompatibility issues between wlroots and hyprland :slightly_smiling_face:

maximbaz, 2024-09-18:
This file `brightness_hw_changed` may or may not exist, when it does, it can indicate that brightness was changed by hardware rather than software. The file name is decided by the Linux kernel.

That file establishes `inotify` watches on several files, which allows us to run some code when any of those files get changed. Meaning that whenever brightness was changed (using software or hardware), we want to execute `update` function, which reads the current brightness value:

https://github.com/maximbaz/wluma/blob/08b48960cbb8325a8546d51aadc3ed5461747e74/src/brightness/backlight.rs#L103-L111

This is to avoid having to periodically check for the current brightness value, which is not very battery-efficient.

Hope that helps!

maximbaz, 2024-09-18:
> Can you clarify whether there may be some people who may get "brightness_hw_changed" and its just in my system that the file "brightness_hw_changed" doesn't appear?

Correct, some people may have this file, and others may not. My current system doesn't have that file either. 

> Also if the File name is decided by the linux kernal, then why did we hardcode it there with brightness_hw_changed in the code?

If it will be present, it will be called `brightness_hw_changed`, i.e. linux has decided to call that file `brightness_hw_changed` and nothing else. Similarly it decided to call the file that contains the brightness value `brightness`, and we also hardcode its name:

https://github.com/maximbaz/wluma/blob/08b48960cbb8325a8546d51aadc3ed5461747e74/src/brightness/backlight.rs#L28

And similarly it decided to name the file that contains maximum brightness value `max_brightness`, which is also hardcoded:

https://github.com/maximbaz/wluma/blob/08b48960cbb8325a8546d51aadc3ed5461747e74/src/brightness/backlight.rs#L68

I hope I understood the questions correctly?

Rishik-Y, 2024-09-19:
Ohh! I see,
It makes a lot of sense now. 
Sorry for taking your time, and thank you for your explanation.

Rishik-Y, 2024-09-28:
Hey @maximbaz, Nothing of a doubt right now here,
I was simply searching for what I could do to replace the quick_assign function, and I was surprised to come across your old chats randomly,
It was interesting for me, so I just wanted to share it with you! 😁
https://github.com/Smithay/wayland-rs/issues/358

Rishik-Y, 2024-10-03:
Hey @maximbaz, Sorry to ask you this,
But I am again stuck and was restarting from scratch using my previous written code as reference. 
(The main issue being I forget what why or what i tried to do by changing the code)

Can you just look into this [Wlroots.rs](https://github.com/Rishik-Y/wluma/blob/main/src/frame/capturer/wlroots.rs)

This is not the final code or anything! 
The goal is just to try to make it work somehow and then improve the code later on...

but I can't seem to know whether I am even doing it the right way

Here are the errors in case it helps (Since you stated you can't use Vulkan right now)
```

❯ cargo test
   Compiling wluma v4.4.1 (/home/kihsir/Downloads/wluma-main)
error: expected expression, found `.`
   --> src/frame/capturer/wlroots.rs:142:13
    |
142 |             .quick_assign(move |data, event, _| match event {
    |             ^ expected expression

warning: unused import: `std::os::fd::IntoRawFd`
 --> src/frame/capturer/wlroots.rs:1:5
  |
1 | use std::os::fd::IntoRawFd;
  |     ^^^^^^^^^^^^^^^^^^^^^^
  |
  = note: `#[warn(unused_imports)]` on by default

warning: unused import: `thread`
 --> src/frame/capturer/wlroots.rs:4:34
  |
4 | use std::{cell::RefCell, rc::Rc, thread, time::Duration};
  |                                  ^^^^^^

warning: unused imports: `Global` and `registry_queue_init`
 --> src/frame/capturer/wlroots.rs:6:15
  |
6 |     globals::{registry_queue_init, Global, GlobalList, GlobalListContents},
  |               ^^^^^^^^^^^^^^^^^^^  ^^^^^^

warning: unused imports: `CancelReason` and `Event`
  --> src/frame/capturer/wlroots.rs:13:35
   |
13 |     zwlr_export_dmabuf_frame_v1::{CancelReason, Event},
   |                                   ^^^^^^^^^^^^  ^^^^^

error[E0599]: no method named `quick_assign` found for struct `ZxdgOutputV1` in the current scope
  --> src/frame/capturer/wlroots.rs:62:22
   |
60 | /                 self.xdg_output_manager
61 | |                     .get_xdg_output(&output, queue_handle, unwraped_list)
62 | |                     .quick_assign(move |_, event, _| match event {
   | |                     -^^^^^^^^^^^^ method not found in `ZxdgOutputV1`
   | |_____________________|
   |

error[E0599]: no method named `dispatch` found for struct `RefMut<'_, EventQueue<wlroots::Capturer>>` in the current scope
   --> src/frame/capturer/wlroots.rs:80:18
    |
78  | /             self.event_queue
79  | |                 .borrow_mut()
80  | |                 .dispatch(&mut (), |_, _, _| {})
    | |_________________-^^^^^^^^
    |
help: there is a method `dispatch_pending` with a similar name, but with different arguments
   --> /home/kihsir/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wayland-client-0.31.6/src/event_queue.rs:387:5
    |
387 |     pub fn dispatch_pending(&mut self, data: &mut State) -> Result<usize, DispatchError> {
    |     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

error[E0599]: no function or associated item named `new` found for struct `GlobalList` in the current scope
  --> src/frame/capturer/wlroots.rs:96:35
   |
96 |         let globals = GlobalList::new(&attached_display);
   |                                   ^^^ function or associated item not found in `GlobalList`

For more information about this error, try `rustc --explain E0599`.
warning: `wluma` (bin "wluma" test) generated 4 warnings
error: could not compile `wluma` (bin "wluma" test) due to 4 previous errors; 4 warnings emitted
```

I am planning to restart the wlroots.rs code again anyway since I believe that would be for the best, however, I would appreciate any faults or pointers you give me that I wouldn't have found otherwise.

maximbaz, 2024-10-03:
This is a tough one, as I remember the way we have to use the new version of the lib has changed quite a lot, that is the reason for some of those errors like "dispatch" or "quick_assign" not existing anymore... If I recall correctly I had some decent progress in #86 on upgrading to the "new" way of using the lib, but I think I did not reach the state where it would work, so you can take it as an inspiration or a starting point, but not as a complete solution...

Rishik-Y, 2024-10-04:
Hi, thanks for your suggestion, But could you umm...
Sorry to ask, but could you give me the link to the code where you wrote it? (I can't seem to find it 😅)

maximbaz, 2024-10-04:
I was just referring to `wlroots.rs` in that PR, e.g. here :sweat_smile: 
https://github.com/maximbaz/wluma/pull/86/files#diff-68e993e6d37850f27f1c82bc616a17284d0ad42115aee63e2d559d9544f1a736

maximbaz, 2024-10-21:
@Rishik-Y Vulkan driver got released for my hardware, which means I can run and test wluma again 🥳 Have you made some progress on upgrading wayland libs? Otherwise I can now try to take it over 💪

maximbaz, 2024-10-21:
I'll close this in favor of https://github.com/maximbaz/wluma/issues/121. Hyprland may or may not support the `wlr-screencopy-unstable-v1` after its migration away from wlroots, but it will most likely support the new `ext-image-*` protocols (at some point).

closed 2024-10-21 by maximbaz

Rishik-Y, 2024-10-22:
> @Rishik-Y Vulkan driver got released for my hardware, which means I can run and test wluma again 🥳 Have you made some progress on upgrading wayland libs? Otherwise I can now try to take it over 💪

Its Nice that you can finally test wluma again 😊 
But sorry to say I didn't make much progress since past 2 weeks 😞 (Due to having Semester exams going on right now)
You can take it over if you would like. 👍️

maximbaz, 2024-10-23:
@Rishik-Y please run the latest `main` branch with `RUST_LOG=trace` and share what you see :wink: 

I'm curious whether your hyprland supports another protocol that I just implemented (it's a bit buggy still).

Rishik-Y, 2024-10-23:
Yep, So As you asked i just ran the new wluma..
I Haven't checked the code yet on what changes you made (Will do soon 😁)
I am a bit confused in some cases, but I will send every scenario nonetheless...

I dont know whether i need to send the logs here or https://github.com/maximbaz/wluma/issues/121 
So for now, i will send all log in the new https://github.com/maximbaz/wluma/issues/121 as its related to the new/**old** protocol which is `wlr-screencopy-unstable-v1`
(Do tell me if you would like to me to continue here instead)

maximbaz, 2024-10-23:
Here is fine, #121 is about a different protocol, not `wlr-screencopy` :)

Rishik-Y, 2024-10-23:
I just ran the latest wluma-main code in my latest Hyprland
Here are some of the logs from when I ran with the config wlr-export-dmabuf-unstable-v

```
❯ cargo run
    Finished `dev` profile [unoptimized + debuginfo] target(s) in 0.16s
warning: the following packages contain code that will be rejected by a future version of Rust: nom v3.2.1
note: to see what the problems were, use the option `--future-incompat-report`, or run `cargo report future-incompatibilities --id 1`
     Running `target/debug/wluma`
[2024-10-23T21:15:26Z INFO  wluma] Continue adjusting brightness and wluma will learn your preference over time.
thread 'predictor-eDP-1' panicked at src/frame/capturer/wayland.rs:105:21:
Requested to use wlr-export-dmabuf-unstable-v1 protocol, but it's not available
stack backtrace:
   0: rust_begin_unwind
   1: core::panicking::panic_fmt
   2: <wluma::frame::capturer::wayland::Capturer as wluma::frame::capturer::Capturer>::run
             at ./src/frame/capturer/wayland.rs:105:21
   3: wluma::main::{{closure}}::{{closure}}
             at ./src/main.rs:87:29
note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.
```

maximbaz, 2024-10-23:
Run it with config `capturer=wlroots`, it will then detect all :+1:

Rishik-Y, 2024-10-23:
Yep just ran it with wlroots 👍️
I was confused for a few minutes thinking it started working but it wasn't changing based on screen content so just went and tried using `Rust_Log=trace`
and its showing only for als.webcam
So I am assuming you fixed it so that at least the program won't crash 🙂

maximbaz, 2024-10-23:
Cool, but just to be sure I understand correctly what you experience, could you please share the logs when you run it with `capturer=wlroots` and `RUST_LOG=trace`?

Rishik-Y, 2024-10-23:
> Here is fine, #121 is about a different protocol, not `wlr-screencopy` :)
> 
> I'm just curious if hyprland supports it at all, or not - in the beginning wluma will now trace-log `Detected support for ....`.

Oh sorry I just took a quick glance at the code here in [app.rs](https://github.com/maximbaz/wluma/blob/main/src/config/app.rs)
and saw WlrScreencopyUnstableV1, so i simply assumed you used it here.
I presume you probably were simply adding it for now as a template.

Rishik-Y, 2024-10-23:
Also here are the logs for `capturer=wlroots` using `RUST_LOG=trace`

```
cargo run
    Finished `dev` profile [unoptimized + debuginfo] target(s) in 0.13s
warning: the following packages contain code that will be rejected by a future version of Rust: nom v3.2.1
note: to see what the problems were, use the option `--future-incompat-report`, or run `cargo report future-incompatibilities --id 1`
     Running `target/debug/wluma`
[2024-10-23T21:32:17Z DEBUG wluma] Using Config {
        als: Webcam {
            video: 0,
            thresholds: {
                30: "dim",
                60: "bright",
                15: "dark",
                0: "night",
                45: "normal",
                75: "outdoors",
            },
        },
        output: [
            Backlight(
                BacklightOutput {
                    name: "eDP-1",
                    path: "/sys/devices/pci0000:00/0000:00:08.1/0000:07:00.0/drm/card2/card2-eDP-1/amdgpu_bl2",
                    capturer: Wayland(
                        Any,
                    ),
                    min_brightness: 1,
                },
            ),
            Backlight(
                BacklightOutput {
                    name: "keyboard-MSI",
                    path: "/sys/class/net/enp4s0/enp4s0-2::lan",
                    capturer: None,
                    min_brightness: 0,
                },
            ),
        ],
    }
[2024-10-23T21:32:17Z INFO  wluma] Continue adjusting brightness and wluma will learn your preference over time.
[2024-10-23T21:32:17Z TRACE wluma::als::webcam] ALS (webcam): outdoors (100)
[2024-10-23T21:32:17Z TRACE wluma::predictor::controller] Prediction: 0 (lux: outdoors, luma: 0)
[2024-10-23T21:32:17Z TRACE wluma::frame::capturer::wayland] Detected support for wlr-screencopy-unstable-v1 protocol
[2024-10-23T21:32:17Z DEBUG wluma::frame::capturer::wayland] Using output 'AU Optronics 0xAF90  (eDP-1)' for config 'eDP-1'
[2024-10-23T21:32:17Z TRACE wluma::frame::capturer::wayland] Using wlr-screencopy-unstable-v1 protocol to request frames
[2024-10-23T21:32:17Z TRACE wluma::als::webcam] ALS (webcam): outdoors (100)
[2024-10-23T21:32:17Z TRACE wluma::als::webcam] ALS (webcam): outdoors (100)
[2024-10-23T21:32:17Z TRACE wluma::predictor::controller] Prediction: 0 (lux: outdoors, luma: 0)
[2024-10-23T21:32:17Z TRACE wluma::als::webcam] ALS (webcam): outdoors (100)
[2024-10-23T21:32:18Z TRACE wluma::predictor::controller] Prediction: 0 (lux: outdoors, luma: 0)
[2024-10-23T21:32:18Z TRACE wluma::als::webcam] ALS (webcam): outdoors (100)
[2024-10-23T21:32:18Z TRACE wluma::als::webcam] ALS (webcam): outdoors (100)
[2024-10-23T21:32:18Z TRACE wluma::predictor::controller] Prediction: 0 (lux: outdoors, luma: 0)
[2024-10-23T21:32:18Z TRACE wluma::als::webcam] ALS (webcam): outdoors (100)
[2024-10-23T21:32:18Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T21:32:18Z TRACE wluma::predictor::controller] Prediction: 0 (lux: outdoors, luma: 0)
[2024-10-23T21:32:18Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T21:32:18Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T21:32:18Z TRACE wluma::predictor::controller] Prediction: 0 (lux: outdoors, luma: 0)
[2024-10-23T21:32:18Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T21:32:18Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T21:32:18Z TRACE wluma::predictor::controller] Prediction: 0 (lux: outdoors, luma: 0)
[2024-10-23T21:32:18Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T21:32:18Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T21:32:19Z TRACE wluma::predictor::controller] Prediction: 0 (lux: outdoors, luma: 0)
[2024-10-23T21:32:19Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T21:32:19Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T21:32:19Z TRACE wluma::predictor::controller] Prediction: 0 (lux: outdoors, luma: 0)
[... 81 lines cut ...]
[2024-10-23T21:32:23Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T21:32:23Z TRACE wluma::predictor::controller] Prediction: 35 (lux: night, luma: 25)
[2024-10-23T21:32:23Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T21:32:23Z TRACE wluma::predictor::controller] Prediction: 35 (lux: night, luma: 25)
[2024-10-23T21:32:23Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T21:32:23Z TRACE wluma::predictor::controller] Prediction: 35 (lux: night, luma: 31)
[2024-10-23T21:32:23Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T21:32:23Z TRACE wluma::predictor::controller] Prediction: 35 (lux: night, luma: 25)
[2024-10-23T21:32:23Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T21:32:23Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T21:32:23Z TRACE wluma::predictor::controller] Prediction: 35 (lux: night, luma: 0)
[2024-10-23T21:32:23Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T21:32:23Z TRACE wluma::predictor::controller] Prediction: 35 (lux: night, luma: 5)
[2024-10-23T21:32:24Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T21:32:24Z TRACE wluma::predictor::controller] Prediction: 35 (lux: night, luma: 6)
[2024-10-23T21:32:24Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T21:32:24Z TRACE wluma::predictor::controller] Prediction: 35 (lux: night, luma: 26)
[2024-10-23T21:32:24Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T21:32:24Z TRACE wluma::predictor::controller] Prediction: 35 (lux: night, luma: 6)
[2024-10-23T21:32:24Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T21:32:24Z TRACE wluma::predictor::controller] Prediction: 35 (lux: night, luma: 27)
[2024-10-23T21:32:24Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T21:32:24Z TRACE wluma::predictor::controller] Prediction: 35 (lux: night, luma: 24)
^C
```

maximbaz, 2024-10-23:
Awesome, thanks! So we can conclude that Hyprland didn't implement `wlr-export-dmabuf-unstable-v1` when moved away from wlroots, but they did implement `wlr-screencopy-unstable-v1` one - which is also wlroots-specific, but perhaps a lot more widely used, so they wanted to ensure compatibility with most apps.

There's unfortunately a memory leak right now, which I can't catch, so be careful with running `wluma` from main branch for long period of time, my RAM gets eaten up very fast. I'll ping you when it's usable for you :+1: 

Thanks for testing and sharing the logs!

Rishik-Y, 2024-10-23:
Very Glad to be useful! 😁
And really sorry you had to, in the end, do it yourself...
I really wanted to fix it, unfortunately, it seems that i still don't have enough skills and knowledge to fix it **yet**
Do tell me if you need any help!

maximbaz, 2024-10-23:
I think I fixed the memory leak :partying_face: I think it's safe to run wluma for longer periods of time, so it would actually be helpful if you could use the latest code from main branch and report if you stumble upon any issue!

maximbaz, 2024-10-23:
One particular issue to keep an eye on, is whether you get occasional unexpected zeros in `luma` calculation, like so:

```
[2024-10-23T21:48:43Z TRACE wluma::predictor::controller] Prediction: 180 (lux: none, luma: 17)
[2024-10-23T21:48:43Z TRACE wluma::predictor::controller] Prediction: 181 (lux: none, luma: 16)
[2024-10-23T21:48:43Z TRACE wluma::predictor::controller] Prediction: 181 (lux: none, luma: 16)
[2024-10-23T21:48:43Z TRACE wluma::predictor::controller] Prediction: 181 (lux: none, luma: 16)
[2024-10-23T21:48:44Z TRACE wluma::predictor::controller] Prediction: 174 (lux: none, luma: 0)  <-------------
[2024-10-23T21:48:44Z TRACE wluma::predictor::controller] Prediction: 180 (lux: none, luma: 17)
[2024-10-23T21:48:44Z TRACE wluma::predictor::controller] Prediction: 181 (lux: none, luma: 16)
[2024-10-23T21:48:44Z TRACE wluma::predictor::controller] Prediction: 181 (lux: none, luma: 16)
[2024-10-23T21:48:44Z TRACE wluma::predictor::controller] Prediction: 181 (lux: none, luma: 16)
[2024-10-23T21:48:44Z TRACE wluma::predictor::controller] Prediction: 181 (lux: none, luma: 16)
[2024-10-23T21:48:44Z TRACE wluma::predictor::controller] Prediction: 181 (lux: none, luma: 16)
[2024-10-23T21:48:44Z TRACE wluma::predictor::controller] Prediction: 174 (lux: none, luma: 0)  <-------------
[2024-10-23T21:48:44Z TRACE wluma::predictor::controller] Prediction: 181 (lux: none, luma: 16)
[2024-10-23T21:48:45Z TRACE wluma::predictor::controller] Prediction: 174 (lux: none, luma: 0)  <-------------
[2024-10-23T21:48:45Z TRACE wluma::predictor::controller] Prediction: 181 (lux: none, luma: 16)
[2024-10-23T21:48:45Z TRACE wluma::predictor::controller] Prediction: 181 (lux: none, luma: 16)
```

I can't still find what I am doing wrong, and there's a dirty workaround that should prevent some/most of them...

https://github.com/maximbaz/wluma/blob/19ec1690cca3f33dfb70af35184397c00cf45d9b/src/frame/capturer/wayland.rs#L429-L430

Rishik-Y, 2024-10-23:
Hey @maximbaz, 
I was running wluma again and it wasnt changing brightness at all so ran the logs and
I am indeed getting those occasional unexpected zeros except its not occasional but always

Here are the logs:
```
❯ cargo run
    Finished `dev` profile [unoptimized + debuginfo] target(s) in 0.06s
warning: the following packages contain code that will be rejected by a future version of Rust: nom v3.2.1
note: to see what the problems were, use the option `--future-incompat-report`, or run `cargo report future-incompatibilities --id 1`
     Running `target/debug/wluma`
[2024-10-23T22:01:32Z DEBUG wluma] Using Config {
        als: Webcam {
            video: 0,
            thresholds: {
                45: "normal",
                30: "dim",
                0: "night",
                60: "bright",
                15: "dark",
                75: "outdoors",
            },
        },
        output: [
            Backlight(
                BacklightOutput {
                    name: "eDP-1",
                    path: "/sys/devices/pci0000:00/0000:00:08.1/0000:07:00.0/drm/card2/card2-eDP-1/amdgpu_bl2",
                    capturer: Wayland(
                        Any,
                    ),
                    min_brightness: 1,
                },
            ),
            Backlight(
                BacklightOutput {
                    name: "keyboard-MSI",
                    path: "/sys/class/net/enp4s0/enp4s0-2::lan",
                    capturer: None,
                    min_brightness: 0,
                },
            ),
        ],
    }
[2024-10-23T22:01:32Z INFO  wluma] Continue adjusting brightness and wluma will learn your preference over time.
[2024-10-23T22:01:32Z TRACE wluma::als::webcam] ALS (webcam): outdoors (100)
[2024-10-23T22:01:32Z TRACE wluma::frame::capturer::wayland] Detected support for wlr-screencopy-unstable-v1 protocol
[2024-10-23T22:01:32Z DEBUG wluma::frame::capturer::wayland] Using output 'AU Optronics 0xAF90  (eDP-1)' for config 'eDP-1'
[2024-10-23T22:01:32Z TRACE wluma::frame::capturer::wayland] Using wlr-screencopy-unstable-v1 protocol to request frames
[2024-10-23T22:01:32Z TRACE wluma::predictor::controller] Prediction: 0 (lux: outdoors, luma: 0)
[2024-10-23T22:01:32Z TRACE wluma::als::webcam] ALS (webcam): outdoors (100)
[2024-10-23T22:01:32Z TRACE wluma::als::webcam] ALS (webcam): outdoors (100)
[2024-10-23T22:01:32Z TRACE wluma::predictor::controller] Prediction: 0 (lux: outdoors, luma: 0)
[2024-10-23T22:01:32Z TRACE wluma::als::webcam] ALS (webcam): outdoors (100)
[2024-10-23T22:01:32Z TRACE wluma::als::webcam] ALS (webcam): outdoors (100)
[2024-10-23T22:01:32Z TRACE wluma::predictor::controller] Prediction: 0 (lux: outdoors, luma: 0)
[2024-10-23T22:01:32Z TRACE wluma::als::webcam] ALS (webcam): outdoors (100)
[2024-10-23T22:01:33Z TRACE wluma::predictor::controller] Prediction: 0 (lux: outdoors, luma: 0)
[2024-10-23T22:01:33Z TRACE wluma::als::webcam] ALS (webcam): outdoors (100)
[2024-10-23T22:01:33Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T22:01:33Z TRACE wluma::predictor::controller] Prediction: 0 (lux: outdoors, luma: 0)
[2024-10-23T22:01:33Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T22:01:33Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T22:01:33Z TRACE wluma::predictor::controller] Prediction: 0 (lux: outdoors, luma: 0)
[2024-10-23T22:01:33Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T22:01:33Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T22:01:33Z TRACE wluma::predictor::controller] Prediction: 0 (lux: outdoors, luma: 0)
[2024-10-23T22:01:33Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T22:01:33Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T22:01:33Z TRACE wluma::predictor::controller] Prediction: 0 (lux: outdoors, luma: 0)
[... 341 lines cut ...]
[2024-10-23T22:02:06Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T22:02:07Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T22:02:07Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T22:02:07Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T22:02:07Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T22:02:07Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T22:02:07Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T22:02:07Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T22:02:07Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T22:02:07Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T22:02:07Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T22:02:08Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T22:02:08Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T22:02:08Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T22:02:08Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T22:02:08Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T22:02:08Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T22:02:08Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T22:02:08Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T22:02:08Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T22:02:08Z TRACE wluma::als::webcam] ALS (webcam): night (8)
^C
```

I waited for a few seconds hoping it would change but nothing happened

maximbaz, 2024-10-23:
Very interesting, hmmmm :thinking: Thanks for sharing, will ping you if I get any idea as to what to try...

maximbaz, 2024-10-23:
Interestingly in [your previous log](https://github.com/maximbaz/wluma/issues/111#issuecomment-2433509184), you did get `luma` value. Could you maybe just try a couple of times latest main (to be sure you still only getting zeros) and then a couple of times a previous commit (to be sure you get non-zeros sometimes). If that's the case, it's a clear indication that my memory leak cleanup is wrong

Rishik-Y, 2024-10-23:
Ok i ran the Latest Commit a couple of times,
and i noticed 2 things...
One is prediction values are still always zero
Second is als.webcam is still as usual detecting (as there are changes with swaps of 8 and 9)

Log:

```
cargo run
    Finished `dev` profile [unoptimized + debuginfo] target(s) in 0.06s
warning: the following packages contain code that will be rejected by a future version of Rust: nom v3.2.1
note: to see what the problems were, use the option `--future-incompat-report`, or run `cargo report future-incompatibilities --id 1`
     Running `target/debug/wluma`
[2024-10-23T22:12:37Z DEBUG wluma] Using Config {
        als: Webcam {
            video: 0,
            thresholds: {
                60: "bright",
                15: "dark",
                30: "dim",
                0: "night",
                45: "normal",
                75: "outdoors",
            },
        },
        output: [
            Backlight(
                BacklightOutput {
                    name: "eDP-1",
                    path: "/sys/devices/pci0000:00/0000:00:08.1/0000:07:00.0/drm/card2/card2-eDP-1/amdgpu_bl2",
                    capturer: Wayland(
                        Any,
                    ),
                    min_brightness: 1,
                },
            ),
            Backlight(
                BacklightOutput {
                    name: "keyboard-MSI",
                    path: "/sys/class/net/enp4s0/enp4s0-2::lan",
                    capturer: None,
                    min_brightness: 0,
                },
            ),
        ],
    }
[2024-10-23T22:12:37Z INFO  wluma] Continue adjusting brightness and wluma will learn your preference over time.
[2024-10-23T22:12:37Z TRACE wluma::als::webcam] ALS (webcam): outdoors (100)
[2024-10-23T22:12:37Z TRACE wluma::predictor::controller] Prediction: 0 (lux: outdoors, luma: 0)
[2024-10-23T22:12:37Z TRACE wluma::frame::capturer::wayland] Detected support for wlr-screencopy-unstable-v1 protocol
[2024-10-23T22:12:37Z DEBUG wluma::frame::capturer::wayland] Using output 'AU Optronics 0xAF90  (eDP-1)' for config 'eDP-1'
[2024-10-23T22:12:37Z TRACE wluma::frame::capturer::wayland] Using wlr-screencopy-unstable-v1 protocol to request frames
[2024-10-23T22:12:37Z TRACE wluma::als::webcam] ALS (webcam): outdoors (100)
[2024-10-23T22:12:37Z TRACE wluma::predictor::controller] Prediction: 0 (lux: outdoors, luma: 0)
[2024-10-23T22:12:37Z TRACE wluma::als::webcam] ALS (webcam): outdoors (100)
[2024-10-23T22:12:37Z TRACE wluma::als::webcam] ALS (webcam): outdoors (100)
[2024-10-23T22:12:37Z TRACE wluma::predictor::controller] Prediction: 0 (lux: outdoors, luma: 0)
[2024-10-23T22:12:37Z TRACE wluma::als::webcam] ALS (webcam): outdoors (100)
[2024-10-23T22:12:37Z TRACE wluma::als::webcam] ALS (webcam): outdoors (100)
[2024-10-23T22:12:38Z TRACE wluma::predictor::controller] Prediction: 0 (lux: outdoors, luma: 0)
[2024-10-23T22:12:38Z TRACE wluma::als::webcam] ALS (webcam): outdoors (100)
[2024-10-23T22:12:38Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T22:12:38Z TRACE wluma::predictor::controller] Prediction: 0 (lux: outdoors, luma: 0)
[2024-10-23T22:12:38Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T22:12:38Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T22:12:38Z TRACE wluma::predictor::controller] Prediction: 0 (lux: outdoors, luma: 0)
[2024-10-23T22:12:38Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T22:12:38Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T22:12:38Z TRACE wluma::predictor::controller] Prediction: 0 (lux: outdoors, luma: 0)
[2024-10-23T22:12:38Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[... 167 lines cut ...]
[2024-10-23T22:12:54Z TRACE wluma::als::webcam] ALS (webcam): night (9)
[2024-10-23T22:12:54Z TRACE wluma::als::webcam] ALS (webcam): night (9)
[2024-10-23T22:12:54Z TRACE wluma::als::webcam] ALS (webcam): night (9)
[2024-10-23T22:12:54Z TRACE wluma::als::webcam] ALS (webcam): night (9)
[2024-10-23T22:12:54Z TRACE wluma::als::webcam] ALS (webcam): night (9)
[2024-10-23T22:12:54Z TRACE wluma::als::webcam] ALS (webcam): night (9)
[2024-10-23T22:12:54Z TRACE wluma::als::webcam] ALS (webcam): night (9)
[2024-10-23T22:12:54Z TRACE wluma::als::webcam] ALS (webcam): night (9)
[2024-10-23T22:12:55Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T22:12:55Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T22:12:55Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T22:12:55Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T22:12:55Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T22:12:55Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T22:12:55Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T22:12:55Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T22:12:55Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T22:12:55Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T22:12:56Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T22:12:56Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T22:12:56Z TRACE wluma::als::webcam] ALS (webcam): night (8)
^C
```

But yes one thing is clear prediction value is always zero (tested 10 times)

Rishik-Y, 2024-10-23:
Here is again the log of previous commit code:

Log:

```
❯ cargo run
    Finished `dev` profile [unoptimized + debuginfo] target(s) in 0.08s
warning: the following packages contain code that will be rejected by a future version of Rust: nom v3.2.1
note: to see what the problems were, use the option `--future-incompat-report`, or run `cargo report future-incompatibilities --id 1`
     Running `target/debug/wluma`
[2024-10-23T22:18:04Z DEBUG wluma] Using Config {
        als: Webcam {
            video: 0,
            thresholds: {
                60: "bright",
                15: "dark",
                0: "night",
                75: "outdoors",
                45: "normal",
                30: "dim",
            },
        },
        output: [
            Backlight(
                BacklightOutput {
                    name: "eDP-1",
                    path: "/sys/devices/pci0000:00/0000:00:08.1/0000:07:00.0/drm/card2/card2-eDP-1/amdgpu_bl2",
                    capturer: Wayland(
                        Any,
                    ),
                    min_brightness: 1,
                },
            ),
            Backlight(
                BacklightOutput {
                    name: "keyboard-MSI",
                    path: "/sys/class/net/enp4s0/enp4s0-2::lan",
                    capturer: None,
                    min_brightness: 0,
                },
            ),
        ],
    }
[2024-10-23T22:18:04Z INFO  wluma] Continue adjusting brightness and wluma will learn your preference over time.
[2024-10-23T22:18:04Z TRACE wluma::als::webcam] ALS (webcam): outdoors (100)
[2024-10-23T22:18:04Z TRACE wluma::predictor::controller] Prediction: 0 (lux: outdoors, luma: 0)
[2024-10-23T22:18:04Z TRACE wluma::frame::capturer::wayland] Detected support for wlr-screencopy-unstable-v1 protocol
[2024-10-23T22:18:04Z DEBUG wluma::frame::capturer::wayland] Using output 'AU Optronics 0xAF90  (eDP-1)' for config 'eDP-1'
[2024-10-23T22:18:04Z TRACE wluma::frame::capturer::wayland] Using wlr-screencopy-unstable-v1 protocol to request frames
[2024-10-23T22:18:04Z TRACE wluma::predictor::controller] Prediction: 135 (lux: outdoors, luma: 64)
[2024-10-23T22:18:04Z TRACE wluma::als::webcam] ALS (webcam): outdoors (100)
[2024-10-23T22:18:04Z TRACE wluma::predictor::controller] Prediction: 135 (lux: outdoors, luma: 35)
[2024-10-23T22:18:04Z TRACE wluma::predictor::controller] Prediction: 0 (lux: outdoors, luma: 0)
[2024-10-23T22:18:04Z TRACE wluma::als::webcam] ALS (webcam): outdoors (100)
[2024-10-23T22:18:04Z TRACE wluma::predictor::controller] Prediction: 135 (lux: outdoors, luma: 59)
[2024-10-23T22:18:04Z TRACE wluma::als::webcam] ALS (webcam): outdoors (100)
[2024-10-23T22:18:04Z TRACE wluma::predictor::controller] Prediction: 135 (lux: outdoors, luma: 63)
[2024-10-23T22:18:04Z TRACE wluma::predictor::controller] Prediction: 0 (lux: outdoors, luma: 0)
[2024-10-23T22:18:04Z TRACE wluma::als::webcam] ALS (webcam): outdoors (100)
[2024-10-23T22:18:04Z TRACE wluma::predictor::controller] Prediction: 135 (lux: outdoors, luma: 65)
[2024-10-23T22:18:04Z TRACE wluma::als::webcam] ALS (webcam): outdoors (100)
[2024-10-23T22:18:04Z TRACE wluma::predictor::controller] Prediction: 135 (lux: outdoors, luma: 58)
[2024-10-23T22:18:04Z TRACE wluma::predictor::controller] Prediction: 0 (lux: outdoors, luma: 0)
[2024-10-23T22:18:04Z TRACE wluma::als::webcam] ALS (webcam): outdoors (100)
[2024-10-23T22:18:05Z TRACE wluma::predictor::controller] Prediction: 135 (lux: outdoors, luma: 36)
[2024-10-23T22:18:05Z TRACE wluma::als::webcam] ALS (webcam): night (9)
[2024-10-23T22:18:05Z TRACE wluma::predictor::controller] Prediction: 135 (lux: outdoors, luma: 65)
[2024-10-23T22:18:05Z TRACE wluma::predictor::controller] Prediction: 0 (lux: outdoors, luma: 0)
[2024-10-23T22:18:05Z TRACE wluma::als::webcam] ALS (webcam): night (9)
[2024-10-23T22:18:05Z TRACE wluma::predictor::controller] Prediction: 135 (lux: outdoors, luma: 59)
[... 56 lines cut ...]
[2024-10-23T22:18:07Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T22:18:07Z TRACE wluma::predictor::controller] Prediction: 35 (lux: night, luma: 50)
[2024-10-23T22:18:07Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T22:18:07Z TRACE wluma::predictor::controller] Prediction: 35 (lux: night, luma: 66)
[2024-10-23T22:18:07Z TRACE wluma::predictor::controller] Prediction: 0 (lux: outdoors, luma: 0)
[2024-10-23T22:18:07Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T22:18:07Z TRACE wluma::predictor::controller] Prediction: 35 (lux: night, luma: 66)
[2024-10-23T22:18:07Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T22:18:07Z TRACE wluma::predictor::controller] Prediction: 35 (lux: night, luma: 46)
[2024-10-23T22:18:07Z TRACE wluma::predictor::controller] Prediction: 0 (lux: outdoors, luma: 0)
[2024-10-23T22:18:07Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T22:18:08Z TRACE wluma::predictor::controller] Prediction: 35 (lux: night, luma: 37)
[2024-10-23T22:18:08Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T22:18:08Z TRACE wluma::predictor::controller] Prediction: 35 (lux: night, luma: 65)
[2024-10-23T22:18:08Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T22:18:08Z TRACE wluma::predictor::controller] Prediction: 35 (lux: night, luma: 63)
[2024-10-23T22:18:08Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T22:18:08Z TRACE wluma::predictor::controller] Prediction: 35 (lux: night, luma: 32)
[2024-10-23T22:18:08Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T22:18:08Z TRACE wluma::predictor::controller] Prediction: 35 (lux: night, luma: 19)
[2024-10-23T22:18:08Z TRACE wluma::als::webcam] ALS (webcam): night (8)
^C
```

Which runs (However with memory leaks)

maximbaz, 2024-10-23:
Okay thanks, and how about with the latest commit that I just pushed?

Rishik-Y, 2024-10-23:
Here the is Logs of previous commit code where there is no such issue (except memory leak)

❯ cargo run
    Finished `dev` profile [unoptimized + debuginfo] target(s) in 0.08s
warning: the following packages contain code that will be rejected by a future version of Rust: nom v3.2.1
note: to see what the problems were, use the option `--future-incompat-report`, or run `cargo report future-incompatibilities --id 1`
     Running `target/debug/wluma`
[2024-10-23T22:18:04Z DEBUG wluma] Using Config {
        als: Webcam {
            video: 0,
            thresholds: {
                60: "bright",
                15: "dark",
                0: "night",
                75: "outdoors",
                45: "normal",
                30: "dim",
            },
        },
        output: [
            Backlight(
                BacklightOutput {
                    name: "eDP-1",
                    path: "/sys/devices/pci0000:00/0000:00:08.1/0000:07:00.0/drm/card2/card2-eDP-1/amdgpu_bl2",
                    capturer: Wayland(
                        Any,
                    ),
                    min_brightness: 1,
                },
            ),
            Backlight(
                BacklightOutput {
                    name: "keyboard-MSI",
                    path: "/sys/class/net/enp4s0/enp4s0-2::lan",
                    capturer: None,
                    min_brightness: 0,
                },
            ),
        ],
    }
[2024-10-23T22:18:04Z INFO  wluma] Continue adjusting brightness and wluma will learn your preference over time.
[2024-10-23T22:18:04Z TRACE wluma::als::webcam] ALS (webcam): outdoors (100)
[2024-10-23T22:18:04Z TRACE wluma::predictor::controller] Prediction: 0 (lux: outdoors, luma: 0)
[2024-10-23T22:18:04Z TRACE wluma::frame::capturer::wayland] Detected support for wlr-screencopy-unstable-v1 protocol
[2024-10-23T22:18:04Z DEBUG wluma::frame::capturer::wayland] Using output 'AU Optronics 0xAF90  (eDP-1)' for config 'eDP-1'
[2024-10-23T22:18:04Z TRACE wluma::frame::capturer::wayland] Using wlr-screencopy-unstable-v1 protocol to request frames
[2024-10-23T22:18:04Z TRACE wluma::predictor::controller] Prediction: 135 (lux: outdoors, luma: 64)
[2024-10-23T22:18:04Z TRACE wluma::als::webcam] ALS (webcam): outdoors (100)
[2024-10-23T22:18:04Z TRACE wluma::predictor::controller] Prediction: 135 (lux: outdoors, luma: 35)
[2024-10-23T22:18:04Z TRACE wluma::predictor::controller] Prediction: 0 (lux: outdoors, luma: 0)
[2024-10-23T22:18:04Z TRACE wluma::als::webcam] ALS (webcam): outdoors (100)
[2024-10-23T22:18:04Z TRACE wluma::predictor::controller] Prediction: 135 (lux: outdoors, luma: 59)
[2024-10-23T22:18:04Z TRACE wluma::als::webcam] ALS (webcam): outdoors (100)
[2024-10-23T22:18:04Z TRACE wluma::predictor::controller] Prediction: 135 (lux: outdoors, luma: 63)
[2024-10-23T22:18:04Z TRACE wluma::predictor::controller] Prediction: 0 (lux: outdoors, luma: 0)
[2024-10-23T22:18:04Z TRACE wluma::als::webcam] ALS (webcam): outdoors (100)
[2024-10-23T22:18:04Z TRACE wluma::predictor::controller] Prediction: 135 (lux: outdoors, luma: 65)
[2024-10-23T22:18:04Z TRACE wluma::als::webcam] ALS (webcam): outdoors (100)
[2024-10-23T22:18:04Z TRACE wluma::predictor::controller] Prediction: 135 (lux: outdoors, luma: 58)
[2024-10-23T22:18:04Z TRACE wluma::predictor::controller] Prediction: 0 (lux: outdoors, luma: 0)
[2024-10-23T22:18:04Z TRACE wluma::als::webcam] ALS (webcam): outdoors (100)
[2024-10-23T22:18:05Z TRACE wluma::predictor::controller] Prediction: 135 (lux: outdoors, luma: 36)
[2024-10-23T22:18:05Z TRACE wluma::als::webcam] ALS (webcam): night (9)
[2024-10-23T22:18:05Z TRACE wluma::predictor::controller] Prediction: 135 (lux: outdoors, luma: 65)
[2024-10-23T22:18:05Z TRACE wluma::predictor::controller] Prediction: 0 (lux: outdoors, luma: 0)
[2024-10-23T22:18:05Z TRACE wluma::als::webcam] ALS (webcam): night (9)
[2024-10-23T22:18:05Z TRACE wluma::predictor::controller] Prediction: 135 (lux: outdoors, luma: 59)
[2024-10-23T22:18:05Z TRACE wluma::als::webcam] ALS (webcam): night (9)
[2024-10-23T22:18:05Z TRACE wluma::predictor::controller] Prediction: 135 (lux: outdoors, luma: 66)
[2024-10-23T22:18:05Z TRACE wluma::predictor::controller] Prediction: 0 (lux: outdoors, luma: 0)
[... 50 lines cut ...]
[2024-10-23T22:18:07Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T22:18:07Z TRACE wluma::predictor::controller] Prediction: 35 (lux: night, luma: 65)
[2024-10-23T22:18:07Z TRACE wluma::predictor::controller] Prediction: 0 (lux: outdoors, luma: 0)
[2024-10-23T22:18:07Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T22:18:07Z TRACE wluma::predictor::controller] Prediction: 35 (lux: night, luma: 50)
[2024-10-23T22:18:07Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T22:18:07Z TRACE wluma::predictor::controller] Prediction: 35 (lux: night, luma: 66)
[2024-10-23T22:18:07Z TRACE wluma::predictor::controller] Prediction: 0 (lux: outdoors, luma: 0)
[2024-10-23T22:18:07Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T22:18:07Z TRACE wluma::predictor::controller] Prediction: 35 (lux: night, luma: 66)
[2024-10-23T22:18:07Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T22:18:07Z TRACE wluma::predictor::controller] Prediction: 35 (lux: night, luma: 46)
[2024-10-23T22:18:07Z TRACE wluma::predictor::controller] Prediction: 0 (lux: outdoors, luma: 0)
[2024-10-23T22:18:07Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[... cut ...]
[2024-10-23T22:18:08Z TRACE wluma::predictor::controller] Prediction: 35 (lux: night, luma: 63)
[2024-10-23T22:18:08Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T22:18:08Z TRACE wluma::predictor::controller] Prediction: 35 (lux: night, luma: 32)
[2024-10-23T22:18:08Z TRACE wluma::als::webcam] ALS (webcam): night (8)
[2024-10-23T22:18:08Z TRACE wluma::predictor::controller] Prediction: 35 (lux: night, luma: 19)
[2024-10-23T22:18:08Z TRACE wluma::als::webcam] ALS (webcam): night (8)
^C

Rishik-Y, 2024-10-23:
I have tried it as well a couple of times and there is no such issue in this one where its showing 0 as prediction value except at the start of process

maximbaz, 2024-10-23:
Okay, cool, so we declare this as fixed? Thanks for spotting! If you spot more issues, please report, it's very helpful :pray:

maximbaz, 2024-10-24:
Actually one more request @Rishik-Y, could you comment this line out and confirm whether those occasional zeros appear for you or not?

https://github.com/maximbaz/wluma/blob/8e81a8f0f3bd0c33d5b368bf909e18caea60b2a1/src/frame/capturer/wayland.rs#L431-L432

```
$ RUST_LOG=trace cargo run
...
[2024-10-23T21:48:43Z TRACE wluma::predictor::controller] Prediction: 180 (lux: none, luma: 17)
[2024-10-23T21:48:43Z TRACE wluma::predictor::controller] Prediction: 181 (lux: none, luma: 16)
[2024-10-23T21:48:43Z TRACE wluma::predictor::controller] Prediction: 181 (lux: none, luma: 16)
[2024-10-23T21:48:43Z TRACE wluma::predictor::controller] Prediction: 181 (lux: none, luma: 16)
[2024-10-23T21:48:44Z TRACE wluma::predictor::controller] Prediction: 174 (lux: none, luma: 0)  <-------------
[2024-10-23T21:48:44Z TRACE wluma::predictor::controller] Prediction: 180 (lux: none, luma: 17)
[2024-10-23T21:48:44Z TRACE wluma::predictor::controller] Prediction: 181 (lux: none, luma: 16)
[2024-10-23T21:48:44Z TRACE wluma::predictor::controller] Prediction: 181 (lux: none, luma: 16)
[2024-10-23T21:48:44Z TRACE wluma::predictor::controller] Prediction: 181 (lux: none, luma: 16)
[2024-10-23T21:48:44Z TRACE wluma::predictor::controller] Prediction: 181 (lux: none, luma: 16)
[2024-10-23T21:48:44Z TRACE wluma::predictor::controller] Prediction: 181 (lux: none, luma: 16)
[2024-10-23T21:48:44Z TRACE wluma::predictor::controller] Prediction: 174 (lux: none, luma: 0)  <-------------
[2024-10-23T21:48:44Z TRACE wluma::predictor::controller] Prediction: 181 (lux: none, luma: 16)
[2024-10-23T21:48:45Z TRACE wluma::predictor::controller] Prediction: 174 (lux: none, luma: 0)  <-------------
[2024-10-23T21:48:45Z TRACE wluma::predictor::controller] Prediction: 181 (lux: none, luma: 16)
[2024-10-23T21:48:45Z TRACE wluma::predictor::controller] Prediction: 181 (lux: none, luma: 16)
```

In other words, is that `thread::sleep()` necessary for you or not?

Rishik-Y, 2024-10-24:
Yep!
As you asked just compared between commenting it out and not,
Both have exactly same output.

Where at start the prediction would predict zero for 3 seconds would swap between 0 and 135 (idk why 135) as the prediction value

Next for 3 seconds occasional zero's appear and disappear (Swapping between predicted value and 0)

And later (Which is after 6 seconds)
Proper prediction which doesn't show any occasional zero.

Commenting and not commenting doesn't seem to make any difference whatsoever. 👍️

maximbaz, 2024-10-25:
@Rishik-Y just so that I have more data, could I ask you to confirm whether the latest `main` still works for you? (see #124 with Hyprland-related issue)

Rishik-Y, 2024-10-25:
Yep just tested the latest main,
And i don't seem to face such issue unfortunately

However while running the `export RUST_LOG=TRACE`

There is some repeatition over here 

```
[2024-10-25T17:08:34Z TRACE wluma::als::webcam] ALS (webcam): night (13)
[2024-10-25T17:08:34Z TRACE wluma::als::webcam] ALS (webcam): night (13)
```

I don't think it is a problem as of now but just letting you know that every 3 seconds, in trace its seems to print double ALS however there is no print of double prediction.
But other than that there is not much of an issue on my side...

Here is the LOG:
```
cargo run
    Finished `dev` profile [unoptimized + debuginfo] target(s) in 0.12s
warning: the following packages contain code that will be rejected by a future version of Rust: nom v3.2.1
note: to see what the problems were, use the option `--future-incompat-report`, or run `cargo report future-incompatibilities --id 1`
     Running `target/debug/wluma`
[2024-10-25T17:08:32Z DEBUG wluma] Using Config {
        als: Webcam {
            video: 0,
            thresholds: {
                60: "bright",
                0: "night",
                45: "normal",
                75: "outdoors",
                15: "dark",
                30: "dim",
            },
        },
        output: [
            Backlight(
                BacklightOutput {
                    name: "eDP-1",
                    path: "/sys/devices/pci0000:00/0000:00:08.1/0000:07:00.0/drm/card2/card2-eDP-1/amdgpu_bl2",
                    capturer: Wayland(
                        Any,
                    ),
                    min_brightness: 1,
                },
            ),
            Backlight(
                BacklightOutput {
                    name: "keyboard-MSI",
                    path: "/sys/class/net/enp4s0/enp4s0-2::lan",
                    capturer: None,
                    min_brightness: 0,
                },
            ),
        ],
    }
[2024-10-25T17:08:32Z DEBUG wluma::brightness::backlight] Using direct write on /sys/devices/pci0000:00/0000:00:08.1/0000:07:00.0/drm/card2/card2-eDP-1/amdgpu_bl2 to change brightness value
[2024-10-25T17:08:32Z DEBUG wluma::brightness::backlight] Using direct write on /sys/class/net/enp4s0/enp4s0-2::lan to change brightness value
[2024-10-25T17:08:32Z INFO  wluma] Continue adjusting brightness and wluma will learn your preference over time.
[2024-10-25T17:08:32Z TRACE wluma::als::webcam] ALS (webcam): outdoors (100)
[2024-10-25T17:08:32Z TRACE wluma::predictor::controller] Prediction: 0 (lux: outdoors, luma: 0)
[2024-10-25T17:08:32Z DEBUG wluma::frame::capturer::wayland] Detected support for wlr-screencopy-unstable-v1 protocol
[2024-10-25T17:08:32Z DEBUG wluma::frame::capturer::wayland] Using output 'AU Optronics 0xAF90  (eDP-1)' for config 'eDP-1'
[2024-10-25T17:08:32Z DEBUG wluma::frame::capturer::wayland] Using wlr-screencopy-unstable-v1 protocol to request frames
[2024-10-25T17:08:32Z TRACE wluma::als::webcam] ALS (webcam): outdoors (100)
[2024-10-25T17:08:32Z TRACE wluma::predictor::controller] Prediction: 0 (lux: outdoors, luma: 0)
[2024-10-25T17:08:32Z TRACE wluma::als::webcam] ALS (webcam): outdoors (100)
[2024-10-25T17:08:32Z TRACE wluma::als::webcam] ALS (webcam): outdoors (100)
[2024-10-25T17:08:33Z TRACE wluma::predictor::controller] Prediction: 0 (lux: outdoors, luma: 0)
[2024-10-25T17:08:33Z TRACE wluma::als::webcam] ALS (webcam): outdoors (100)
[2024-10-25T17:08:33Z TRACE wluma::als::webcam] ALS (webcam): outdoors (100)
[... 142 lines cut ...]
[2024-10-25T17:08:40Z TRACE wluma::predictor::controller] Prediction: 35 (lux: night, luma: 3)
[2024-10-25T17:08:40Z TRACE wluma::als::webcam] ALS (webcam): night (13)
[2024-10-25T17:08:40Z TRACE wluma::predictor::controller] Prediction: 35 (lux: night, luma: 10)
[2024-10-25T17:08:40Z TRACE wluma::als::webcam] ALS (webcam): night (13)
[2024-10-25T17:08:40Z TRACE wluma::predictor::controller] Prediction: 35 (lux: night, luma: 3)
[2024-10-25T17:08:40Z TRACE wluma::als::webcam] ALS (webcam): night (13)
[2024-10-25T17:08:41Z TRACE wluma::predictor::controller] Prediction: 35 (lux: night, luma: 12)
[2024-10-25T17:08:41Z TRACE wluma::als::webcam] ALS (webcam): night (13)
[2024-10-25T17:08:41Z TRACE wluma::als::webcam] ALS (webcam): night (13)
[2024-10-25T17:08:41Z TRACE wluma::predictor::controller] Prediction: 35 (lux: night, luma: 10)
[2024-10-25T17:08:41Z TRACE wluma::als::webcam] ALS (webcam): night (13)
[2024-10-25T17:08:41Z TRACE wluma::predictor::controller] Prediction: 35 (lux: night, luma: 10)
[2024-10-25T17:08:41Z TRACE wluma::als::webcam] ALS (webcam): night (13)
[2024-10-25T17:08:41Z TRACE wluma::predictor::controller] Prediction: 35 (lux: night, luma: 2)
[2024-10-25T17:08:41Z TRACE wluma::als::webcam] ALS (webcam): night (13)
[2024-10-25T17:08:41Z TRACE wluma::predictor::controller] Prediction: 35 (lux: night, luma: 19)
[2024-10-25T17:08:41Z TRACE wluma::als::webcam] ALS (webcam): night (13)
[2024-10-25T17:08:41Z TRACE wluma::predictor::controller] Prediction: 35 (lux: night, luma: 3)
[2024-10-25T17:08:41Z TRACE wluma::als::webcam] ALS (webcam): night (13)
[2024-10-25T17:08:41Z TRACE wluma::predictor::controller] Prediction: 35 (lux: night, luma: 2)
[2024-10-25T17:08:41Z TRACE wluma::als::webcam] ALS (webcam): night (13)
[2024-10-25T17:08:41Z TRACE wluma::predictor::controller] Prediction: 35 (lux: night, luma: 11)
[2024-10-25T17:08:41Z TRACE wluma::als::webcam] ALS (webcam): night (13)
^C
```

Rishik-Y, 2025-05-23:
Hello @maximbaz,

It's been a while — a *really* long while. The last time I worked with you on wluma was during last year’s summer vacation, and I’ve been meaning to reach out to you for a while.

I just got selected for Google Summer of Code 2025 with Waycrate!

[image]

You can see my name over [here](https://summerofcode.withgoogle.com/programs/2025/organizations/waycrate) too 😀

Not sure if you remember, but back when I first started contributing to wluma, it was my very first real open-source project. At the time, I just thought wluma was like any average open-source tool. Looking back now, I realize — **it absolutely wasn’t beginner-friendly at all**. Honestly, it probably shouldn’t have been my first project.

But the amazing thing is, you never made it feel that way.
You were always patient, always encouraging, and never once made me feel out of place, even when I was asking beginner-level questions. (I still remember asking how to view Rust logs 😅)

If it weren’t for my time working on wluma, I might never have explored low-level programming or Linux internals, infact **had not gotten selected in GSoC in the first place**.

I probably learned way more than I ever gave back — and that early exposure laid the foundation for everything I did after. 

I was even selected as a Teaching Assistant for a master’s-level Computer System Programming course at my university while just a third-year undergrad.

None of this would have happened without the start I had working with you. I’m truly grateful for all the time, effort, and mentorship you gave when I was just getting started.

I’ve been swamped with university work for a while, which is why I didn’t reach out earlier — but now that summer’s begun, and credit is long overdue, I just wanted to say: **thank you**. I truly appreciate it.

I’d love to stay in touch, and maybe even return to wluma someday when I’m better equipped to contribute meaningfully.

[2 more replies]