max-baz/wluma #114: [Bug]: `Unable to get brightness value: DDC/CI error: invalid DDC/CI length`

goekce, 2024-09-06:
### Steps for reproducing the issue

1. Connect three daisy-chained monitors via USB Video
2. Activate `ddcutil` in wluma with name `HP` and capturer `Wlroots`
3. start wluma
4. Logs show: `ERROR wluma::brightness::controller] Unable to get brightness value: DDC/CI error: invalid DDC/CI length`

Setting the brightness via `ddcutil` works.

### What is the buggy behavior?

Wluma cannot get the brightness value

### What is the expected behavior?

Wluma should be able to get the brightness value

### Logs

```shell
RUST_LOG=debug wluma

2024-09-06T19:45:58Z DEBUG wluma] Using Config {
        als: Iio {
            path: "/sys/bus/iio/devices",
            thresholds: {
                800: "outdoors",
                250: "normal",
                500: "bright",
                20: "dark",
                80: "dim",
                0: "night",
            },
        },
        output: [
            Backlight(
                BacklightOutput {
                    name: "eDP-1",
                    path: "/sys/class/backlight/intel_backlight",
                    capturer: Wlroots,
                    min_brightness: 1,
                },
            ),
            DdcUtil(
                DdcUtilOutput {
                    name: "HP",
                    capturer: Wlroots,
                    min_brightness: 1,
                },
            ),
        ],
    }
[2024-09-06T19:46:24Z DEBUG wluma::brightness::ddcutil] Discovered displays (check_caps=true): ["HP Z24u G3 ...", "HP Z24u G3 ...", "HP Z24u G3 ...", "HP Z24n G2 ..."]
[2024-09-06T19:46:24Z DEBUG wluma::brightness::ddcutil] Using display 'HP Z24u G3 CN43520628' for config 'HP' (check_caps=true)
[2024-09-06T19:46:24Z INFO  wluma] Continue adjusting brightness and wluma will learn your preference over time.
[2024-09-06T19:46:24Z DEBUG wluma::frame::capturer::wlroots] Using output 'HP Inc. HP Z24u G3 ... (DP-5)' for config 'HP'
[2024-09-06T19:46:24Z DEBUG wluma::frame::capturer::wlroots] Using output 'HP Inc. HP Z24u G3 ... (DP-7)' for config 'HP'
[2024-09-06T19:46:24Z DEBUG wluma::frame::capturer::wlroots] Using output 'HP Inc. HP Z24n G2 ... (DP-8)' for config 'HP'
[2024-09-06T19:46:25Z ERROR wluma::brightness::controller] Unable to get brightness value: DDC/CI error: invalid DDC/CI length
```
```

### Version

Using Archlinux package with version 4.4.0

### Environment

```shell
Linux m 6.10.8-arch1-1 #1 SMP PREEMPT_DYNAMIC Wed, 04 Sep 2024 15:16:37 +0000 x86_64 GNU/Linux
853 sway
914 swayidle
egrep: warning: egrep is obsolescent; using grep -E
sway 1:1.9-5
sway-contrib 1.9.0-2
swayidle 1.8.0-2
swaylock 1.8.0-1
vulkan-headers 1:1.3.285-1
vulkan-icd-loader 1.3.285-1
vulkan-intel 1:24.2.1-1
wlroots0.17 0.17.4-3
zsh: command not found: dpkg
egrep: warning: egrep is obsolescent; using grep -E
```
```

goekce, 2024-09-06:
I believe the problem is related to the phantom display (with serial `xxx` below) which has the same name as the real display. The phantom display probably does not deliver a brightness value.
```
["HP Z24u G3 xxx", "HP Z24u G3 xxx", "HP Z24u G3 yyy", "HP Z24n G2 zzz"]
```
```
$ ddcutil detect
Invalid display
   I2C bus:  /dev/i2c-11
   DRM connector:           card1-eDP-1
   EDID synopsis:
      Mfg id:               CMN - Chimei Innolux Corporation
      Model:                
      Product code:         5173  (0x1435)
      Serial number:        
      Binary serial number: 0 (0x00000000)
      Manufacture year:     2021,  Week: 13
   This is a laptop display.  Laptop displays do not support DDC/CI

Phantom display
   Associated non-phantom display: bus /dev/i2c-16
   I2C bus:  /dev/i2c-12
   DRM connector:           card1-DP-1
   EDID synopsis:
      Mfg id:               HPN - UNK
      Model:                HP Z24u G3
      Product code:         14026  (0x36ca)
      Serial number:        xxx
      Binary serial number: 0 (0x00000000)
      Manufacture year:     2023,  Week: 52
   DDC communication failed
   Use non-phantom device bus /dev/i2c-16

Display 1
   I2C bus:  /dev/i2c-16
   DRM connector:           card1-DP-5
   EDID synopsis:
      Mfg id:               HPN - UNK
      Model:                HP Z24u G3
      Product code:         14026  (0x36ca)
      Serial number:        xxx
      Binary serial number: 0 (0x00000000)
      Manufacture year:     2023,  Week: 52
   VCP version:         Detection failed

Display 2
   I2C bus:  /dev/i2c-17
   DRM connector:           card1-DP-7
   EDID synopsis:
      Mfg id:               HPN - UNK
      Model:                HP Z24u G3
      Product code:         14027  (0x36cb)
      Serial number:        yyy
      Binary serial number: 0 (0x00000000)
      Manufacture year:     2023,  Week: 52
   VCP version:         2.2

Display 3
   I2C bus:  /dev/i2c-18
   DRM connector:           card1-DP-8
   EDID synopsis:
      Mfg id:               HPN - UNK
      Model:                HP Z24n G2
      Product code:         13445  (0x3485)
      Serial number:        zzz
      Binary serial number: ...
      Manufacture year:     2019,  Week: 12
   VCP version:         2.2
```

maximbaz, 2024-09-06:
Hello! I believe you can match on the serial number of your display in wluma config, in case the serial number doesn't change across reboots. Otherwise it's a bit tough to find a solution to skip phantom displays without necessary hardware, what worked so far was to filter displays only for those that can successfully respond to "update capabilities" request, but clearly in your case the phantom display is able to respond to that :thinking: If you'd like to tinker with the code and see if you can identify another condition that can distinguish a phantom from a real display in your setup, the necessary parts are here, and any findings would be very appreciated: https://github.com/maximbaz/wluma/blob/8df0b8e2c04c9e7bf0a5f560d59e29786cc5be8e/src/brightness/ddcutil.rs#L71-L73

goekce, 2024-09-06:
Actually the serials are the same, so I get two exact strings for phantom and nonphantom.

Thanks for the idea! However I will unlikely will have time to look into this, but someone probably will 🙂

maximbaz, 2024-09-09:
Sure thing! Your output of `ddcutil detect` has an interesting extra line for the phantom display: "DDC communication failed". Hopefully we'll be able to find a command that can tell us this, and use this as a filter.

Let's keep it open for some time and see if anyone is able to help. I do think it's a bit hard to investigate this without having the hardware to test against, but maybe there will be someone else having the exact same problem.

EvanSunde, 2024-11-04:
I have have same problem but in my case there is no phantom display 

```
RUST_LOG=debug wluma
[2024-11-04T02:23:20Z DEBUG wluma] Using Config {
        als: Webcam {
            video: 0,
            thresholds: {
                60: "bright",
                15: "dark",
                0: "night",
                30: "dim",
                45: "normal",
                75: "outdoors",
            },
        },
        output: [
            Backlight(
                BacklightOutput {
                    name: "eDP-1",
                    path: "/sys/class/backlight/intel_backlight",
                    capturer: Wayland(
                        Any,
                    ),
                    min_brightness: 1,
                },
            ),
            DdcUtil(
                DdcUtilOutput {
                    name: "K3VT99AT0PLL",
                    capturer: Wayland(
                        Any,
                    ),
                    min_brightness: 1,
                },
            ),
            Backlight(
                BacklightOutput {
                    name: "keyboard-dell",
                    path: "/sys/bus/platform/devices/dell-laptop/leds/dell::kbd_backlight",
                    capturer: None,
                    min_brightness: 0,
                },
            ),
        ],
    }
[2024-11-04T02:23:20Z DEBUG wluma::brightness::backlight] Using direct write on /sys/class/backlight/intel_backlight to change brightness value
[2024-11-04T02:23:20Z DEBUG wluma::frame::capturer::wayland] Detected support for wlr-screencopy-unstable-v1 protocol
[2024-11-04T02:23:20Z DEBUG wluma::frame::capturer::wayland] Detected support for linux-dmabuf-v1 protocol
[2024-11-04T02:23:20Z DEBUG wluma::frame::capturer::wayland] Using output 'AU Optronics 0x10ED  (eDP-1)' for config 'eDP-1'
[2024-11-04T02:23:20Z DEBUG wluma::frame::capturer::wayland] Using wlr-screencopy-unstable-v1 protocol to request frames
[2024-11-04T02:23:22Z DEBUG wluma::brightness::ddcutil] Discovered displays (check_caps=true): ["DELL E2418HN K3VT99AT0PLL"]
[2024-11-04T02:23:22Z DEBUG wluma::brightness::ddcutil] Using display 'DELL E2418HN K3VT99AT0PLL' for config 'K3VT99AT0PLL' (check_caps=true)
[2024-11-04T02:23:22Z DEBUG wluma::frame::capturer::wayland] Detected support for wlr-screencopy-unstable-v1 protocol
[2024-11-04T02:23:22Z DEBUG wluma::frame::capturer::wayland] Detected support for linux-dmabuf-v1 protocol
[2024-11-04T02:23:22Z DEBUG wluma::frame::capturer::wayland] Using output 'Dell Inc. DELL E2418HN K3VT99AT0PLL (HDMI-A-1)' for config 'K3VT99AT0PLL'
[2024-11-04T02:23:22Z DEBUG wluma::frame::capturer::wayland] Using wlr-screencopy-unstable-v1 protocol to request frames
[2024-11-04T02:23:22Z DEBUG wluma::brightness::backlight] Using direct write on /sys/bus/platform/devices/dell-laptop/leds/dell::kbd_backlight to change brightness value
[2024-11-04T02:23:22Z INFO  wluma] Continue adjusting brightness and wluma will learn your preference over time.
[2024-11-04T02:27:11Z ERROR wluma::brightness::controller] Unable to get brightness value: DDC/CI error: Expected DDC/CI length bit
[2024-11-04T02:27:11Z ERROR wluma::brightness::controller] Unable to get brightness value: DDC/CI error: Expected DDC/CI length bit
[2024-11-04T02:27:12Z ERROR wluma::brightness::controller] Unable to get brightness value: DDC/CI error: Unsupported VCP code
[2024-11-04T02:27:12Z ERROR wluma::brightness::controller] Unable to get brightness value: DDC/CI error: Unsupported VCP code
[2024-11-04T02:27:12Z ERROR wluma::brightness::controller] Unable to get brightness value: DDC/CI error: Expected DDC/CI length bit
[2024-11-04T02:27:13Z ERROR wluma::brightness::controller] Unable to get brightness value: DDC/CI error: Expected DDC/CI length bit
[2024-11-04T02:27:13Z ERROR wluma::brightness::controller] Unable to get brightness value: DDC/CI error: Expected DDC/CI length bit
[2024-11-04T02:27:14Z ERROR wluma::brightness::controller] Unable to get brightness value: DDC/CI error: Expected DDC/CI length bit
[2024-11-04T02:27:15Z ERROR wluma::brightness::controller] Unable to get brightness value: DDC/CI error: Expected DDC/CI length bit
[2024-11-04T02:27:15Z ERROR wluma::brightness::controller] Unable to get brightness value: DDC/CI error: Expected DDC/CI length bit
[2024-11-04T02:27:17Z DEBUG wluma::predictor::controller] [K3VT99AT0PLL] Learning Entry { lux: "bright", luma: 8, brightness: 7 }
[... 20 lines cut ...]
   I2C bus:  /dev/i2c-4
   DRM connector:           card1-HDMI-A-1
   EDID synopsis:
      Mfg id:               DEL - Dell Inc.
      Model:                DELL E2418HN
      Product code:         41221  (0xa105)
      Serial number:        K3VT99AT0PLL
      Binary serial number: 810568780 (0x30504c4c)
      Manufacture year:     2019,  Week: 44
   VCP version:         Detection failed

Invalid display
   I2C bus:  /dev/i2c-6
   DRM connector:           card1-eDP-1
   EDID synopsis:
      Mfg id:               AUO - UNK
      Model:                
      Product code:         4333  (0x10ed)
      Serial number:        
      Binary serial number: 0 (0x00000000)
      Manufacture year:     2016,  Week: 0
   This is a laptop display.  Laptop displays do not support DDC/CI
```

so when i manually try to change brightness it takes some time (a second or 2) each time the key is pressed which is very frustrating.

maximbaz, 2024-12-26:
Hi all, thanks again for reporting but it looks like it's an unsolvable problem. Unfortunately I don't think it's actionable for me at this point, I just have no idea how to investigate this further. I'm therefore going to close this issue. Any findings, ideas or PRs are always welcome, do not hesitate to share here or in a new issue.

I will also explicitly add https://gitlab.com/ddcci-driver-linux/ddcci-driver-linux to README, as it sounds like more people generally had success with this rather than using DDC directly.

closed 2024-12-26 by maximbaz