max-baz/wluma #115: [Feature request]: Pass config file as a commandline argument

aktaboot, 2024-09-07:
### Please describe your feature request

Having this feature would make it more convenient to run it as a system-wide service.

maximbaz, 2024-09-09:
Just to add that in order to achieve running it as a system-wide service, more changes would be needed, e.g. wluma stores training data in `$XDG_DATA_HOME`.

https://github.com/maximbaz/wluma/blob/8df0b8e2c04c9e7bf0a5f560d59e29786cc5be8e/src/predictor/data.rs#L58-L59

May I ask what is your motivation for running it as a system-wide service?

maximbaz, 2024-12-25:
Not planned for now, as I'm not really seeing a benefit, but any new arguments in favor of this request are very welcome, here or in a new issue.

closed 2024-12-25 by maximbaz