max-baz/wluma #120: [Bug]: Unable to compute luma percent: ERROR_INVALID_EXTERNAL_HANDLE

nitrotol, 2024-10-18:
### Steps for reproducing the issue

1. start _wluma_
2. Logs show: Unable to compute luma percent: ERROR_INVALID_EXTERNAL_HANDLE

### What is the buggy behavior?

App chashes

### What is the expected behavior?

Wluma should be fully functional

### Logs

```shell
RUST_LOG=debug wluma
[2024-10-18T19:31:34Z DEBUG wluma] Using Config {
        als: Iio {
            path: "/sys/bus/iio/devices",
            thresholds: {
                250: "normal",
                0: "night",
                800: "outdoors",
                80: "dim",
                500: "bright",
                20: "dark",
            },
        },
        output: [
            Backlight(
                BacklightOutput {
                    name: "eDP-1",
                    path: "/sys/class/backlight/intel_backlight",
                    capturer: Wlroots,
                    min_brightness: 1,
                },
            ),
            Backlight(
                BacklightOutput {
                    name: "keyboard-dell",
                    path: "/sys/bus/platform/devices/dell-laptop/leds/dell::kbd_backlight",
                    capturer: None,
                    min_brightness: 0,
                },
            ),
        ],
    }
[2024-10-18T19:31:34Z WARN  wluma] Skipping 'keyboard-dell' as it might be disconnected: No such file or directory (os error 2)
[2024-10-18T19:31:34Z INFO  wluma] Continue adjusting brightness and wluma will learn your preference over time.
[2024-10-18T19:31:34Z DEBUG wluma::frame::capturer::wlroots] Using output 'Samsung Display Corp. 0x419D (eDP-1)' for config 'eDP-1'
thread 'predictor-eDP-1' panicked at src/frame/capturer/wlroots.rs:128:26:
Unable to compute luma percent: ERROR_INVALID_EXTERNAL_HANDLE
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

### Version

4.4.0-1

### Environment

```shell
lsb_release -a; uname -a; pgrep -l sway; pacman -Q | egrep "(wlroots|vulkan|sway|clang|rust)"; dpkg -l | egrep "(wlroots|vulkan|sway|clang|rust)"
zsh: command not found: lsb_release
Linux nitrotol-zenbook 6.11.3-arch1-1 #1 SMP PREEMPT_DYNAMIC Thu, 10 Oct 2024 20:11:06 +0000 x86_64 GNU/Linux
1809 sway
1830 swaybg
1865 swayidle
egrep: warning: egrep is obsolescent; using grep -E
clang 18.1.8-3
rust 1:1.81.0-1
swaybg 1.2.1-1
swayfx 0.4-3
swayfx-debug 0.4-3
swayidle 1.8.0-2
swaylock 1.8.0-1
vulkan-headers 1:1.3.295-1
vulkan-icd-loader 1.3.295-1
vulkan-intel 1:24.2.4-1
vulkan-tools 1.3.269-1
wlroots 0.18.1-1
wlroots0.17 0.17.4-3
zsh: command not found: dpkg
egrep: warning: egrep is obsolescent; using grep -E
```

maximbaz, 2024-10-18:
Interesting! Sounds like Vulkan is unable to import the sway frame. Have you set `WLR_DRM_NO_MODIFIERS=1` before starting sway? That would be the most likely culprit :thinking:

nitrotol, 2024-10-18:
Sure!
Without setting this var I getting same error as described [here](https://github.com/maximbaz/wluma/issues/8) 

Also I set `VK_DRIVER_FILES=/usr/share/vulkan/icd.d/intel_icd.x86_64.json` for working with proper Vulkan drivers.

maximbaz, 2024-10-18:
I see, uff this is a tricky one, but let's see if we can find some solution.
Could you post output of `vulkaninfo --summary` please? I'm wondering if you have e.g. multiple GPUs. 
(I suppose you also need to set your `VK_DRIVER_FILES` before executing that command).

nitrotol, 2024-10-18:
> vulkaninfo --summary

`
==========
VULKANINFO
==========

Vulkan Instance Version: 1.3.295

Instance Extensions: count = 24
-------------------------------
VK_EXT_acquire_drm_display             : extension revision 1
VK_EXT_acquire_xlib_display            : extension revision 1
VK_EXT_debug_report                    : extension revision 10
VK_EXT_debug_utils                     : extension revision 2
VK_EXT_direct_mode_display             : extension revision 1
VK_EXT_display_surface_counter         : extension revision 1
VK_EXT_headless_surface                : extension revision 1
VK_EXT_surface_maintenance1            : extension revision 1
VK_EXT_swapchain_colorspace            : extension revision 4
VK_KHR_device_group_creation           : extension revision 1
VK_KHR_display                         : extension revision 23
VK_KHR_external_fence_capabilities     : extension revision 1
VK_KHR_external_memory_capabilities    : extension revision 1
VK_KHR_external_semaphore_capabilities : extension revision 1
VK_KHR_get_display_properties2         : extension revision 1
VK_KHR_get_physical_device_properties2 : extension revision 2
VK_KHR_get_surface_capabilities2       : extension revision 1
VK_KHR_portability_enumeration         : extension revision 1
VK_KHR_surface                         : extension revision 25
VK_KHR_surface_protected_capabilities  : extension revision 1
VK_KHR_wayland_surface                 : extension revision 6
VK_KHR_xcb_surface                     : extension revision 6
VK_KHR_xlib_surface                    : extension revision 6
VK_LUNARG_direct_driver_loading        : extension revision 1

Instance Layers:
----------------

Devices:
========
GPU0:
	apiVersion         = 1.3.289
	driverVersion      = 24.2.4
	vendorID           = 0x8086
	deviceID           = 0x7d55
	deviceType         = PHYSICAL_DEVICE_TYPE_INTEGRATED_GPU
	deviceName         = Intel(R) Arc(tm) Graphics (MTL)
	driverID           = DRIVER_ID_INTEL_OPEN_SOURCE_MESA
	driverName         = Intel open-source Mesa driver
	driverInfo         = Mesa 24.2.4-arch1.1
	conformanceVersion = 1.3.6.0
	deviceUUID         = 8680557d-0800-0000-0002-000000000000
	driverUUID         = e7ab8bf4-4324-0b47-64da-ea2741aa2562`

maximbaz, 2024-10-18:
Yeah I definitely agree, everything that I know looks good and straightforward, unfortunately I don't have any good ideas :disappointed: 

The error is happening in this function, in case you want to try to debug something: https://github.com/maximbaz/wluma/blob/c82288e8134964cad77644a2101a83e415b29fd9/src/frame/vulkan.rs#L286

One final place that stood out to me is perhaps this line:

https://github.com/maximbaz/wluma/blob/c82288e8134964cad77644a2101a83e415b29fd9/src/frame/vulkan.rs#L337

Could you check in the `vulkaninfo` output (without `--summary`) the section `VkPhysicalDeviceMemoryProperties`, what do you see there? I assume you would just see a single memory type, as you say you have just an integrated GPU, I wouldn't expect anything exotic there... But if you see multiple, maybe that is the issue then?

Maybe share the entire `vulkaninfo` just in case someone more knowledgeable comes along and is able to help us figure out what's wrong...

While my code is most plausible to have the issue than anything else, we cannot rule out that the issue is in the Vulkan driver. It actually is the case for my current laptop I can't run wluma because of what's currently most likely is the bug in the Vulkan driver for my GPU :disappointed:

nitrotol, 2024-10-18:
Here is entire vulkaninfo
[vulkaninfo.txt](https://github.com/user-attachments/files/17440709/vulkaninfo.txt)

```
VkPhysicalDeviceMemoryProperties:
=================================
memoryHeaps: count = 1
        memoryHeaps[0]:
                size   = 16491317248 (0x3d6f58800) (15.36 GiB)
                budget = 14841544704 (0x374a00000) (13.82 GiB)
                usage  = 0 (0x00000000) (0.00 B)
                flags: count = 1
                        MEMORY_HEAP_DEVICE_LOCAL_BIT
memoryTypes: count = 4
        memoryTypes[0]:
                heapIndex     = 0
                propertyFlags = 0x0007: count = 3
                        MEMORY_PROPERTY_DEVICE_LOCAL_BIT
                        MEMORY_PROPERTY_HOST_VISIBLE_BIT
                        MEMORY_PROPERTY_HOST_COHERENT_BIT
                usable for:
                        IMAGE_TILING_OPTIMAL:
                                color images
                                FORMAT_D16_UNORM
                                FORMAT_X8_D24_UNORM_PACK32
                                FORMAT_D32_SFLOAT
                                FORMAT_S8_UINT
                                FORMAT_D24_UNORM_S8_UINT
                                FORMAT_D32_SFLOAT_S8_UINT
                        IMAGE_TILING_LINEAR:
                                color images
        memoryTypes[1]:
                heapIndex     = 0
                propertyFlags = 0x000b: count = 3
                        MEMORY_PROPERTY_DEVICE_LOCAL_BIT
                        MEMORY_PROPERTY_HOST_VISIBLE_BIT
                        MEMORY_PROPERTY_HOST_CACHED_BIT
                usable for:
                        IMAGE_TILING_OPTIMAL:
                                color images
                                FORMAT_D16_UNORM
                                FORMAT_X8_D24_UNORM_PACK32
                                FORMAT_D32_SFLOAT
                                FORMAT_S8_UINT
                                FORMAT_D24_UNORM_S8_UINT
                                FORMAT_D32_SFLOAT_S8_UINT
                        IMAGE_TILING_LINEAR:
                                color images
        memoryTypes[2]:
                heapIndex     = 0
                propertyFlags = 0x0007: count = 3
                        MEMORY_PROPERTY_DEVICE_LOCAL_BIT
                        MEMORY_PROPERTY_HOST_VISIBLE_BIT
                        MEMORY_PROPERTY_HOST_COHERENT_BIT
                usable for:
                        IMAGE_TILING_OPTIMAL:
                                None
                        IMAGE_TILING_LINEAR:
                                None
        memoryTypes[3]:
                heapIndex     = 0
                propertyFlags = 0x000b: count = 3
                        MEMORY_PROPERTY_DEVICE_LOCAL_BIT
                        MEMORY_PROPERTY_HOST_VISIBLE_BIT
                        MEMORY_PROPERTY_HOST_CACHED_BIT
                usable for:
                        IMAGE_TILING_OPTIMAL:
                                None
                        IMAGE_TILING_LINEAR:
                                None
```

3 memory types from my integrated GPU as you can see..

I'll be really apreciated to any help from your side

maximbaz, 2024-10-18:
Wow, cannot believe it was (maybe) a lucky hit!

Would you be able to test some things for me please?

1. Add `println!("memory_type_bits: {:#b}", frame_image_mem_req.memory_requirements.memory_type_bits);`, say after [this line](https://github.com/maximbaz/wluma/blob/c82288e8134964cad77644a2101a83e415b29fd9/src/frame/vulkan.rs#L343)
2. Try to put an index directly, see which one, if any, works:

https://github.com/maximbaz/wluma/blob/c82288e8134964cad77644a2101a83e415b29fd9/src/frame/vulkan.rs#L360

Here, just try with `.memory_type_index(0);`, then with `.memory_type_index(1); `, `.memory_type_index(2);`, etc.

nitrotol, 2024-10-18:
Sure
1. It shows `memory_type_bits: 0b11`
2. I tried `0`,`1`,`2`,`3` and nothing helps

maximbaz, 2024-10-18:
Thanks! I guess I was excited too prematurely :disappointed: It's good to know that the current code (probably) doesn't suffer from the issue with picking among multiple memory types, but with regards to your actual issue we are back to square one, I don't really have any good ideas what might be wrong :disappointed: 

Do you do anything special to your sway setup, do you for example run it with some non-default `WLR_RENDERER` env var? Asking just because the "invalid external handle" implies that vulkan thinks that it got some kind of broken external reference from sway.

nitrotol, 2024-10-18:
No, nothing special. I have a basic sway setup

maximbaz, 2024-10-18:
Could you check if you see something interesting in the output, if you install `vulkan-validation-layers` package, then export these env vars, and then run wluma?

```
export VK_INSTANCE_LAYERS=VK_LAYER_KHRONOS_validation
export VK_LOADER_DEBUG=all
cargo run
```

I get some debug logs, but if I understand correctly, if Vulkan decides that some APIs are not used correctly, you would see some error message there, that might indicate issue in the code.

nitrotol, 2024-10-19:
I see this
`     Running `target/debug/wluma`
INFO:              Vulkan Loader Version 1.3.295
[2024-10-19T12:22:21Z WARN  wluma] Skipping 'keyboard-dell' as it might be disconnected: No such file or directory (os error 2)
[2024-10-19T12:22:21Z INFO  wluma] Continue adjusting brightness and wluma will learn your preference over time.
LAYER:             Searching for implicit layer manifest files
LAYER:                In following locations:
LAYER:                   /home/nitrotol/.config/vulkan/implicit_layer.d
LAYER:                   /etc/xdg/vulkan/implicit_layer.d
LAYER:                   /etc/vulkan/implicit_layer.d
LAYER:                   /home/nitrotol/.local/share/vulkan/implicit_layer.d
LAYER:                   /usr/local/share/vulkan/implicit_layer.d
LAYER:                   /usr/share/vulkan/implicit_layer.d
LAYER:                Found no files
LAYER:             Searching for explicit layer manifest files
LAYER:                In following locations:
LAYER:                   /home/nitrotol/.config/vulkan/explicit_layer.d
LAYER:                   /etc/xdg/vulkan/explicit_layer.d
LAYER:                   /etc/vulkan/explicit_layer.d
LAYER:                   /home/nitrotol/.local/share/vulkan/explicit_layer.d
LAYER:                   /usr/local/share/vulkan/explicit_layer.d
LAYER:                   /usr/share/vulkan/explicit_layer.d
LAYER:                Found the following files:
LAYER:                   /usr/share/vulkan/explicit_layer.d/VkLayer_khronos_validation.json
INFO:              Found manifest file /usr/share/vulkan/explicit_layer.d/VkLayer_khronos_validation.json (file version 1.2.0)
DRIVER:            Searching for driver manifest files
DRIVER:               In following locations:
DRIVER:                  /usr/share/vulkan/icd.d/intel_icd.x86_64.json
DRIVER:               Found the following files:
DRIVER:                  /usr/share/vulkan/icd.d/intel_icd.x86_64.json
DRIVER:            Found ICD manifest file /usr/share/vulkan/icd.d/intel_icd.x86_64.json, version 1.0.0
DEBUG | DRIVER:    Searching for ICD drivers named /usr/lib/libvulkan_intel.so
WARNING | LAYER:   env var 'VK_INSTANCE_LAYERS' defined and adding layers "VK_LAYER_KHRONOS_validation"
WARNING | LAYER:   env var 'VK_INSTANCE_LAYERS' defined and adding layers "VK_LAYER_KHRONOS_validation"
DEBUG | LAYER:     Loading layer library libVkLayer_khronos_validation.so
INFO | LAYER:      Insert instance layer "VK_LAYER_KHRONOS_validation" (libVkLayer_khronos_validation.so)
LAYER:             vkCreateInstance layer callstack setup to:
LAYER:                <Application>
LAYER:                  ||
LAYER:                <Loader>
LAYER:                  ||
LAYER:                VK_LAYER_KHRONOS_validation
LAYER:                        Type: Explicit
LAYER:                        Manifest: /usr/share/vulkan/explicit_layer.d/VkLayer_khronos_validation.json
LAYER:                        Library:  libVkLayer_khronos_validation.so
LAYER:                  ||
LAYER:                <Drivers>
INFO | DRIVER:     linux_read_sorted_physical_devices:
INFO | DRIVER:          Original order:
INFO | DRIVER:                [0] Intel(R) Arc(tm) Graphics (MTL)
INFO | DRIVER:          Sorted order:
INFO | DRIVER:                [0] Intel(R) Arc(tm) Graphics (MTL)
INFO | DRIVER:     linux_read_sorted_physical_devices:
INFO | DRIVER:          Original order:
INFO | DRIVER:                [0] Intel(R) Arc(tm) Graphics (MTL)
INFO | DRIVER:          Sorted order:
INFO | DRIVER:                [0] Intel(R) Arc(tm) Graphics (MTL)
DEBUG | DRIVER:    Copying old device 0 into new device 0
INFO | DRIVER:     linux_read_sorted_physical_devices:
INFO | DRIVER:          Original order:
INFO | DRIVER:                [0] Intel(R) Arc(tm) Graphics (MTL)
INFO | DRIVER:          Sorted order:
INFO | DRIVER:                [0] Intel(R) Arc(tm) Graphics (MTL)
DEBUG | DRIVER:    Copying old device 0 into new device 0
INFO | DRIVER:     linux_read_sorted_physical_devices:
INFO | DRIVER:          Original order:
INFO | DRIVER:                [0] Intel(R) Arc(tm) Graphics (MTL)
INFO | DRIVER:          Sorted order:
INFO | DRIVER:                [0] Intel(R) Arc(tm) Graphics (MTL)
DEBUG | DRIVER:    Copying old device 0 into new device 0
INFO | LAYER:      Inserted device layer "VK_LAYER_KHRONOS_validation" (libVkLayer_khronos_validation.so)
DRIVER | LAYER:    vkCreateDevice layer callstack setup to:
DRIVER | LAYER:       <Application>
DRIVER | LAYER:         ||
DRIVER | LAYER:       <Loader>
DRIVER | LAYER:         ||
LAYER:                VK_LAYER_KHRONOS_validation
LAYER:                        Type: Explicit
LAYER:                        Manifest: /usr/share/vulkan/explicit_layer.d/VkLayer_khronos_validation.json
LAYER:                        Library:  libVkLayer_khronos_validation.so
LAYER:                  ||
DRIVER | LAYER:       <Device>
DRIVER | LAYER:           Using "Intel(R) Arc(tm) Graphics (MTL)" with driver: "/usr/lib/libvulkan_intel.so"
VUID-VkImageCreateInfo-pNext-00990(ERROR / SPEC): msgNum: 1181525669 - Validation Error: [ VUID-VkImageCreateInfo-pNext-00990 ] | MessageID = 0x466ca6a5 | vkCreateImage(): pCreateInfo The handle type (VK_EXTERNAL_MEMORY_HANDLE_TYPE_DMA_BUF_BIT_EXT), format (VK_FORMAT_R8G8B8A8_UNORM), type (VK_IMAGE_TYPE_2D), tiling (VK_IMAGE_TILING_OPTIMAL), usage (VK_IMAGE_USAGE_TRANSFER_SRC_BIT), flags (VkImageCreateFlags(0)) is not supported combination of parameters and vkGetPhysicalDeviceImageFormatProperties2 returned back VK_ERR [... cut ...] n the pNext chain, with a handleType equal to any one of the handle types specified in VkExternalMemoryImageCreateInfo::handleTypes (https://www.khronos.org/registry/vulkan/specs/1.3-extensions/html/vkspec.html#VUID-VkImageCreateInfo-pNext-00990)
    Objects: 0
memory_type_bits: 0b11
thread 'predictor-eDP-1' panicked at src/frame/capturer/wlroots.rs:128:26:
Unable to compute luma percent: ERROR_INVALID_EXTERNAL_HANDLE
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace`

maximbaz, 2024-10-19:
Very interesting! Could you please try `test-formats` branch? Do you get any format that is supported?

nitrotol, 2024-10-19:
I tried `test-formats`
Result: [vk_formats.txt](https://github.com/user-attachments/files/17445462/vk_formats.txt)

unfortunatelly no supported formats as I can see
So this is bug in Vulkan?

maximbaz, 2024-10-19:
I'm not sure yet, because I also don't get any formats supported, but as I mentioned wluma doesn't work on my laptop as well. I'll try to find someone who can run this branch on a laptop where it definitely works, to see if they get any format that is reported as "supported". I'm still hoping it's a bug in the test code, but I can't see right now what could be wrong.

nitrotol, 2024-10-19:
Sure! Enjoy your weekend

maximbaz, 2024-10-21:
Could you please retry the latest `test-formats` branch? I'm interested in both the output (which, if any, formats are supported and which not), and also whether wluma actually works now for you, or you still get the same error as before. 

Do still run with validation layers enabled, in case there's now some other validation being thrown to you.

Thanks!

maximbaz, 2024-10-21:
OMG this fixed wluma on my hardware too!!! Big thank you for reporting, without this I would have thought that the issue is with Vulkan driver! :tada: :rocket:

maximbaz, 2024-10-21:
Okay maybe "fixed" was a bit too early to say, I am getting "some" output from the driver now, but the colors are wrong :joy: But that's something I can work with, at least :grin: So do not trust the "luma" output just yet, but see if wluma runs at all without crashes and without vulkan validations for you, and we take the next step then.

nitrotol, 2024-10-21:
IT WORKS! Awesome!

this is my output 
[vl_out.txt](https://github.com/user-attachments/files/17458112/vl_out.txt)

maximbaz, 2024-10-21:
Cool, I'll ping you for a bit more help with extra debugging when I get the colors right, to make sure I do something that works not only on my hardware but on yours too, but glad to see we are making progress!

nitrotol, 2024-10-21:
Sure. Let me know it you need any additional support with debugging from me

maximbaz, 2024-10-21:
Please try the following test:

Pull latest `test-formats` branch, run it, it will be very high CPU consuming test that occasionally prints some lines.

First, I'm interested in the line that starts with `Frame:`, could you paste it here?

Second, while it's running, go to this website and hit "Start test": https://lcdtech.info/en/tests/dead.pixel.htm

It should open a single color on full-screen. Find red, let it stay on your screen for a few seconds, then go back to wluma and see what it was logging while your screen was fully red. Good news is if you see `255 0 0 0` on repeat, bad news is if you see something else. Try with green (should give `0 255 0 0`), blue (should give `0 0 255 0`).

If you see other values, please share too, color and corresponding values.

Many thanks!

nitrotol, 2024-10-21:
output from the latest `test-formats` build 
Some kind of usual usage:
```
Frame: width: 2880 height: 1800 num_objects: 1 format: 875713112 buffer_flags: 0 flags: Transient mod_high: 0 mod_low: 0
[11, 2, 33, 0, 11, 2, 33, 0, 11, 2, 33, 0]
Frame: width: 2880 height: 1800 num_objects: 1 format: 875713112 buffer_flags: 0 flags: Transient mod_high: 0 mod_low: 0
[11, 2, 33, 0, 11, 2, 33, 0, 11, 2, 33, 0]
Frame: width: 2880 height: 1800 num_objects: 1 format: 875713112 buffer_flags: 0 flags: Transient mod_high: 0 mod_low: 0
[11, 2, 33, 0, 11, 2, 33, 0, 11, 2, 33, 0]
Frame: width: 2880 height: 1800 num_objects: 1 format: 875713112 buffer_flags: 0 flags: Transient mod_high: 0 mod_low: 0
[11, 2, 33, 0, 11, 2, 33, 0, 11, 2, 33, 0]
Frame: width: 2880 height: 1800 num_objects: 1 format: 875713112 buffer_flags: 0 flags: Transient mod_high: 0 mod_low: 0
[11, 2, 33, 0, 11, 2, 33, 0, 11, 2, 33, 0]
Frame: width: 2880 height: 1800 num_objects: 1 format: 875713112 buffer_flags: 0 flags: Transient mod_high: 0 mod_low: 0
[201, 84, 93, 0, 201, 84, 93, 0, 201, 84, 93, 0]
Frame: width: 2880 height: 1800 num_objects: 1 format: 875713112 buffer_flags: 0 flags: Transient mod_high: 0 mod_low: 0
[201, 84, 93, 0, 201, 84, 93, 0, 201, 84, 93, 0]
Frame: width: 2880 height: 1800 num_objects: 1 format: 875713112 buffer_flags: 0 flags: Transient mod_high: 0 mod_low: 0
[201, 84, 93, 0, 201, 84, 93, 0, 201, 84, 93, 0]
Frame: width: 2880 height: 1800 num_objects: 1 format: 875713112 buffer_flags: 0 flags: Transient mod_high: 0 mod_low: 0
[11, 2, 33, 0, 11, 2, 33, 0, 11, 2, 33, 0]
Frame: width: 2880 height: 1800 num_objects: 1 format: 875713112 buffer_flags: 0 flags: Transient mod_high: 0 mod_low: 0
[11, 2, 33, 0, 11, 2, 33, 0, 11, 2, 33, 0]
Frame: width: 2880 height: 1800 num_objects: 1 format: 875713112 buffer_flags: 0 flags: Transient mod_high: 0 mod_low: 0
[11, 2, 33, 0, 11, 2, 33, 0, 11, 2, 33, 0]
Frame: width: 2880 height: 1800 num_objects: 1 format: 875713112 buffer_flags: 0 flags: Transient mod_high: 0 mod_low: 0
[11, 2, 33, 0, 11, 2, 33, 0, 11, 2, 33, 0]
Frame: width: 2880 height: 1800 num_objects: 1 format: 875713112 buffer_flags: 0 flags: Transient mod_high: 0 mod_low: 0
[201, 84, 93, 0, 201, 84, 93, 0, 201, 84, 93, 0]
Frame: width: 2880 height: 1800 num_objects: 1 format: 875713112 buffer_flags: 0 flags: Transient mod_high: 0 mod_low: 0
[201, 84, 93, 0, 201, 84, 93, 0, 201, 84, 93, 0]
Frame: width: 2880 height: 1800 num_objects: 1 format: 875713112 buffer_flags: 0 flags: Transient mod_high: 0 mod_low: 0
[201, 84, 93, 0, 201, 84, 93, 0, 201, 84, 93, 0]
Frame: width: 2880 height: 1800 num_objects: 1 format: 875713112 buffer_flags: 0 flags: Transient mod_high: 0 mod_low: 0
[201, 84, 93, 0, 201, 84, 93, 0, 201, 84, 93, 0]
Frame: width: 2880 height: 1800 num_objects: 1 format: 875713112 buffer_flags: 0 flags: Transient mod_high: 0 mod_low: 0
[201, 84, 93, 0, 201, 84, 93, 0, 201, 84, 93, 0]
Frame: width: 2880 height: 1800 num_objects: 1 format: 875713112 buffer_flags: 0 flags: Transient mod_high: 0 mod_low: 0
[201, 84, 93, 0, 201, 84, 93, 0, 201, 84, 93, 0]
Frame: width: 2880 height: 1800 num_objects: 1 format: 875713112 buffer_flags: 0 flags: Transient mod_high: 0 mod_low: 0
[201, 84, 93, 0, 201, 84, 93, 0, 201, 84, 93, 0]
Frame: width: 2880 height: 1800 num_objects: 1 format: 875713112 buffer_flags: 0 flags: Transient mod_high: 0 mod_low: 0
[11, 2, 33, 0, 11, 2, 33, 0, 11, 2, 33, 0]
Frame: width: 2880 height: 1800 num_objects: 1 format: 875713112 buffer_flags: 0 flags: Transient mod_high: 0 mod_low: 0
```

lcdtech website is down for me, wont oppen(
So I using https://deadpixeltest.org/
Color test:
```
Frame: width: 2880 height: 1800 num_objects: 1 format: 875713112 buffer_flags: 0 flags: Transient mod_high: 0 mod_low: 0
[255, 0, 0, 0, 255, 0, 0, 0, 255, 0, 0, 0]
Frame: width: 2880 height: 1800 num_objects: 1 format: 875713112 buffer_flags: 0 flags: Transient mod_high: 0 mod_low: 0
[255, 0, 0, 0, 255, 0, 0, 0, 255, 0, 0, 0]
Frame: width: 2880 height: 1800 num_objects: 1 format: 875713112 buffer_flags: 0 flags: Transient mod_high: 0 mod_low: 0
[0, 255, 0, 0, 0, 255, 0, 0, 0, 255, 0, 0]
Frame: width: 2880 height: 1800 num_objects: 1 format: 875713112 buffer_flags: 0 flags: Transient mod_high: 0 mod_low: 0
[0, 255, 0, 0, 0, 255, 0, 0, 0, 255, 0, 0]
Frame: width: 2880 height: 1800 num_objects: 1 format: 875713112 buffer_flags: 0 flags: Transient mod_high: 0 mod_low: 0
[0, 255, 0, 0, 0, 255, 0, 0, 0, 255, 0, 0]
Frame: width: 2880 height: 1800 num_objects: 1 format: 875713112 buffer_flags: 0 flags: Transient mod_high: 0 mod_low: 0
[0, 255, 0, 0, 0, 255, 0, 0, 0, 255, 0, 0]
Frame: width: 2880 height: 1800 num_objects: 1 format: 875713112 buffer_flags: 0 flags: Transient mod_high: 0 mod_low: 0
[0, 0, 255, 0, 0, 0, 255, 0, 0, 0, 255, 0]
Frame: width: 2880 height: 1800 num_objects: 1 format: 875713112 buffer_flags: 0 flags: Transient mod_high: 0 mod_low: 0
[0, 0, 255, 0, 0, 0, 255, 0, 0, 0, 255, 0]
Frame: width: 2880 height: 1800 num_objects: 1 format: 875713112 buffer_flags: 0 flags: Transient mod_high: 0 mod_low: 0
[0, 0, 255, 0, 0, 0, 255, 0, 0, 0, 255, 0]
Frame: width: 2880 height: 1800 num_objects: 1 format: 875713112 buffer_flags: 0 flags: Transient mod_high: 0 mod_low: 0
[0, 0, 255, 0, 0, 0, 255, 0, 0, 0, 255, 0]
```
It was on red, than on green and on blue on the end
Looks like it detect color properly now

maximbaz, 2024-10-21:
Awesome, thank you for the test! I'm still struggling because on my own laptop I'm still getting garbage values for the colors :sob: I'll try to struggle a bit more, but even if I fail for my own laptop, I'll make sure to cut a release with the fix for you very soon :pray:

nitrotol, 2024-10-21:
It will be great!
Let me know if I can help you with debug in any way

BTW I don't notice any extra CPU load with latest `test-format` during my tests

maximbaz, 2024-10-21:
In fact I was getting weird colors on lcdtech.info/en/tests/dead.pixel.htm, but on the website you found I am getting correct colors reported! So I think the fix is good after all, I was just confused somewhere during my earlier tests.

The final fix for this issue report is very pleasantly short, just 3 lines of code, and actually makes sense :)

https://github.com/maximbaz/wluma/commit/caf9da6fb23e9005aab4dc7eba0e6624f17c60fd

I'm going to make a release in a moment!

Many thanks again for all your help! :pray: :ukraine:

closed 2024-10-21 by maximbaz