max-baz/wluma #123: [Feature request]: allow to change webcam capture interval

wolf-yuan-6115, 2024-10-24:
### Please describe your feature request

Is it possible to change webcam capture interval in the config? From https://github.com/maximbaz/wluma/blob/8e81a8f0f3bd0c33d5b368bf909e18caea60b2a1/src/als/webcam.rs#L16 it seems like the 2000ms delay is hard coded into config parser.

maximbaz, 2024-10-24:
Hey, could you please share some motivation behind your request? There are also some downsides to this, see also #91 for some context.

wolf-yuan-6115, 2024-10-24:
> Hey, could you please share some motivation behind your request? There are also some downsides to this, see also #91 for some context.

Oh I didn't notice there was already an existing issue talking about this, sorry for the duplicate.

My problem is similar, the blinking LED is kinda annoying, and I think 2000ms delay is too short when on battery.

I think I'll modify the source code to change that.

closed 2024-10-24 by wolf-yuan-6115

maximbaz, 2024-10-25:
No worries, and I'd really appreciate if you could experiment with different values and share your experience - I am not against making it configurable if necessary, but I also want to prevent people from getting a bad experience. 

To  give you an example, say you used wluma for a while and collected enough training data for a perfect experience both when you sit with lights on and off. Say you set this interval to 30s, you have lights off and you turn them on. For the next 30s wluma will not see that, so you will sit with too-dim brightness value, maybe being annoyed that wluma doesn't react when it should. And if you decide to intervene and manually increase the brightness, you just inadvertently destroyed all your existing training data, by introducing a conflicting data point.