max-baz/wluma #124: [Bug]: Got Perceived lightness is 0, this could be a bug I'm trying to catch.

wolf-yuan-6115, 2024-10-25:
### Steps for reproducing the issue

- Run `wluma`, got:
  ```
  [2024-10-25T14:14:24Z WARN  wluma::frame::vulkan] Perceived lightness is 0, this could be a bug I'm trying to catch. If you see this value while your screen is not pitch-black, please open a GitHub issue (or use an already opened one) and share your logs, ideally with RUST_LOG=trace and this message.
  ```

### What is the buggy behavior?

`wluma` shows warnings in log.

### What is the expected behavior?

`wluma` works correctly.

### Logs

```shell
[2024-10-25T14:14:23Z DEBUG wluma] Using Config {
        als: Webcam {
            video: 0,
            thresholds: {
                60: "bright",
                30: "dim",
                0: "night",
                75: "outdoors",
                15: "dark",
                45: "normal",
            },
        },
        output: [
            Backlight(
                BacklightOutput {
                    name: "eDP-1",
                    path: "/sys/class/backlight/amdgpu_bl0",
                    capturer: Wayland(
                        WlrScreencopyUnstableV1,
                    ),
                    min_brightness: 1,
                },
            ),
        ],
    }
[2024-10-25T14:14:23Z DEBUG wluma::brightness::backlight] Using direct write on /sys/class/backlight/amdgpu_bl0 to change brightness value
[2024-10-25T14:14:23Z INFO  wluma] Continue adjusting brightness and wluma will learn your preference over time.
[2024-10-25T14:14:23Z TRACE wluma::als::webcam] ALS (webcam): outdoors (100)
[2024-10-25T14:14:23Z DEBUG wluma::frame::capturer::wayland] Detected support for wlr-screencopy-unstable-v1 protocol
[2024-10-25T14:14:23Z DEBUG wluma::frame::capturer::wayland] Using output 'AU Optronics 0xD1ED  (eDP-1)' for config 'eDP-1'
[2024-10-25T14:14:23Z DEBUG wluma::frame::capturer::wayland] Using wlr-screencopy-unstable-v1 protocol to request frames
[2024-10-25T14:14:23Z WARN  wluma::frame::vulkan] Perceived lightness is 0, this could be a bug I'm trying to catch. If you see this value while your screen is not pitch-black, please open a GitHub issue (or use an already opened one) and share your logs, ideally with RUST_LOG=trace and this message.
[2024-10-25T14:14:24Z TRACE wluma::als::webcam] ALS (webcam): outdoors (100)
[2024-10-25T14:14:24Z WARN  wluma::frame::vulkan] Perceived lightness is 0, this could be a bug I'm trying to catch. If you see this value while your screen is not pitch-black, please open a GitHub issue (or use an already opened one) and share your logs, ideally with RUST_LOG=trace and this message.
[2024-10-25T14:14:24Z TRACE wluma::als::webcam] ALS (webcam): outdoors (100)
[2024-10-25T14:14:24Z WARN  wluma::frame::vulkan] Perceived lightness is 0, this could be a bug I'm trying to catch. If you see this value while your screen is not pitch-black, please open a GitHub issue (or use an already opened one) and share your logs, ideally with RUST_LOG=trace and this message.
[2024-10-25T14:14:24Z TRACE wluma::als::webcam] ALS (webcam): outdoors (100)
[2024-10-25T14:14:24Z TRACE wluma::als::webcam] ALS (webcam): outdoors (100)
[2024-10-25T14:14:24Z WARN  wluma::frame::vulkan] Perceived lightness is 0, this could be a bug I'm trying to catch. If you see this value while your screen is not pitch-black, please open a GitHub issue (or use an already opened one) and share your logs, ideally with RUST_LOG=trace and this message.
[2024-10-25T14:14:24Z TRACE wluma::als::webcam] ALS (webcam): outdoors (100)
[2024-10-25T14:14:24Z WARN  wluma::frame::vulkan] Perceived lightness is 0, this could be a bug I'm trying to catch. If you see this value while your screen is not pitch-black, please open a GitHub issue (or use an already opened one) and share your logs, ideally with RUST_LOG=trace and this message.
[2024-10-25T14:14:24Z TRACE wluma::als::webcam] ALS (webcam): outdoors (100)
[2024-10-25T14:14:24Z WARN  wluma::frame::vulkan] Perceived lightness is 0, this could be a bug I'm trying to catch. If you see this value while your screen is not pitch-black, please open a GitHub issue (or use an already opened one) and share your logs, ideally with RUST_LOG=trace and this message.
[2024-10-25T14:14:24Z TRACE wluma::als::webcam] ALS (webcam): outdoors (100)
[2024-10-25T14:14:24Z WARN  wluma::frame::vulkan] Perceived lightness is 0, this could be a bug I'm trying to catch. If you see this value while your screen is not pitch-black, please open a GitHub issue (or use an already opened one) and share your logs, ideally with RUST_LOG=trace and this message.
[2024-10-25T14:14:24Z TRACE wluma::als::webcam] ALS (webcam): outdoors (100)
[2024-10-25T14:14:24Z WARN  wluma::frame::vulkan] Perceived lightness is 0, this could be a bug I'm trying to catch. If you see this value while your screen is not pitch-black, please open a GitHub issue (or use an already opened one) and share your logs, ideally with RUST_LOG=trace and this message.
[2024-10-25T14:14:24Z TRACE wluma::als::webcam] ALS (webcam): normal (51)
[2024-10-25T14:14:24Z WARN  wluma::frame::vulkan] Perceived lightness is 0, this could be a bug I'm trying to catch. If you see this value while your screen is not pitch-black, please open a GitHub issue (or use an already opened one) and share your logs, ideally with RUST_LOG=trace and this message.
[2024-10-25T14:14:24Z TRACE wluma::als::webcam] ALS (webcam): normal (51)
[2024-10-25T14:14:24Z WARN  wluma::frame::vulkan] Perceived lightness is 0, this could be a bug I'm trying to catch. If you see this value while your screen is not pitch-black, please open a GitHub issue (or use an already opened one) and share your logs, ideally with RUST_LOG=trace and this message.
[2024-10-25T14:14:25Z TRACE wluma::als::webcam] ALS (webcam): normal (51)
[2024-10-25T14:14:25Z WARN  wluma::frame::vulkan] Perceived lightness is 0, this could be a bug I'm trying to catch. If you see this value while your screen is not pitch-black, please open a GitHub issue (or use an already opened one) and share your logs, ideally with RUST_LOG=trace and this message.
[2024-10-25T14:14:25Z TRACE wluma::als::webcam] ALS (webcam): normal (51)
```

### Version

https://github.com/maximbaz/wluma/commit/db8bf319c5f7e63edea2b9ed249d6f296e215676

### Environment

```shell
LSB Version:	n/a
Distributor ID:	Arch
Description:	Arch Linux
Release:	rolling
Codename:	n/a
Linux snowflake 6.11.5-zen1-1-zen #1 ZEN SMP PREEMPT_DYNAMIC Tue, 22 Oct 2024 18:31:33 +0000 x86_64 GNU/Linux
2253 swayosd-libinpu
2669 swayosd-server
clang 18.1.8-4
lib32-vulkan-icd-loader 1.3.295-1
lib32-vulkan-radeon 1:24.2.5-1
rust 1:1.82.0-2
swayosd-git 0.1.0.r0.gdbb8b72-1
vulkan-headers 1:1.3.295-1
vulkan-icd-loader 1.3.295-1
vulkan-radeon 1:24.2.5-1
vulkan-tools 1.3.269-1
vulkan-validation-layers 1.3.290-2
wlroots0.17 0.17.4-3
WLR_DRM_NO_MODIFIERS=
```

maximbaz, 2024-10-25:
Thanks for sharing! Are you on Hyprland or on what desktop environment? Has wluma ever worked for you? It looks like you aren't getting a single frame captured at all.

wolf-yuan-6115, 2024-10-25:
> Thanks for sharing! Are you on Hyprland or on what desktop environment? Has wluma ever worked for you? It looks like you aren't getting a single frame captured at all.

I'm using the latest Hyprland, I remember I just ran `paru -Syu` an hour ago. It does work when using this commit with only screen capture: https://github.com/maximbaz/wluma/commit/633aeafc1d6987ee717a9ede2986b15e6b78e9f9

But if I use `als.webcam` + `capture` it only shows current webcam luma, for example:

```
...
TRACE wluma::als::webcam] ALS (webcam): outdoors (100)
TRACE wluma::als::webcam] ALS (webcam): outdoors (100)
TRACE wluma::als::webcam] ALS (webcam): outdoors (100)
...
```

The commit https://github.com/maximbaz/wluma/commit/db8bf319c5f7e63edea2b9ed249d6f296e215676 seems has same issue too.

I just saw a new commit made several minutes ago, I'll try it out when I'm free.

maximbaz, 2024-10-25:
Yeah, this is very suspicious too. The only commit I can think of, that could cause issue, is this commit:

https://github.com/maximbaz/wluma/commit/a599f262ad8b5e26f64d70f964211b32af864f77

When you get some time, could you please test with and without `als.webcam` that commit and a commit before, to confirm what exactly went wrong there?

maximbaz, 2024-12-25:
Closing as maybe not actionable anymore, as I rewrote that part of code, but if screen contents is still not being captured, please open a new issue ( we also have e.g. this open hyprland issue https://github.com/maximbaz/wluma/issues/125 ).

closed 2024-12-25 by maximbaz