max-baz/wluma #128: [Feature request]: DB with already made config files for different computers

Peiprjs, 2024-12-16:
### Please describe your feature request

I think it would be useful to have a database of config files already made for different models of laptop or computer usually used.

Peiprjs, 2024-12-16:
`[als.iio]
path = "/sys/bus/iio/devices"
thresholds = { 0 = "night", 20 = "dark", 80 = "dim", 250 = "normal", 500 = "bright", 800 = "outdoors" }

# [als.webcam]
# video = 0
# thresholds = { 0 = "night", 15 = "dark", 30 = "dim", 45 = "normal", 60 = "bright", 75 = "outdoors" }

# [als.time]
# thresholds = { 0 = "night", 7 = "dark", 9 = "dim", 11 = "normal", 13 = "bright", 16 = "normal", 18 = "dark", 20 = "night" }

# [als.none]

[[output.backlight]]
name = "AMD GPU"
path = "/sys/class/backlight/amdgpu_bl1/"
capturer = "none"

# [[output.ddcutil]]
# name = "Dell Inc. DELL P2415Q"
# capturer = "none"

[[keyboard]]
name = "keyboard-dell"
path = "/sys/bus/platform/devices/cros-keyboard-leds.8.auto/leds/chromeos::kbd_backlight/"

# [[keyboard]]
# name = "keyboard-thinkpad"
# path = "/sys/bus/platform/devices/thinkpad_acpi/leds/tpacpi::kbd_backlight"`

Peiprjs, 2024-12-16:
```
[als.iio]
path = "/sys/bus/iio/devices"
thresholds = { 0 = "night", 20 = "dark", 80 = "dim", 250 = "normal", 500 = "bright", 800 = "outdoors" }

# [als.webcam]
# video = 0
# thresholds = { 0 = "night", 15 = "dark", 30 = "dim", 45 = "normal", 60 = "bright", 75 = "outdoors" }

# [als.time]
# thresholds = { 0 = "night", 7 = "dark", 9 = "dim", 11 = "normal", 13 = "bright", 16 = "normal", 18 = "dark", 20 = "night" }

# [als.none]

[[output.backlight]]
name = "AMD GPU"
path = "/sys/class/backlight/amdgpu_bl1/"
capturer = "none"

# [[output.ddcutil]]
# name = "Dell Inc. DELL P2415Q"
# capturer = "none"

[[keyboard]]
name = "keyboard-dell"
path = "/sys/bus/platform/devices/cros-keyboard-leds.8.auto/leds/chromeos::kbd_backlight/"

# [[keyboard]]
# name = "keyboard-thinkpad"
# path = "/sys/bus/platform/devices/thinkpad_acpi/leds/tpacpi::kbd_backlight"
```

Peiprjs, 2024-12-16:
I have a FW13 so I tried making this but I'm not sure if it's correct.

maximbaz, 2024-12-16:
Hello! I've just made the wiki editable by anyone, I believe this would be a great place to contribute configs and other tips, without me being the bottleneck. Would you like to take the initiative to place your config there? Feel free to create the structure as you see fit.

Peiprjs, 2024-12-17:
Sure! I would like to verify if it works first though, so I'll give it a couple of days

maximbaz, 2024-12-17:
Awesome! Will close the issue, feel free to go wild on the wiki :grin: And if you have any other questions, dont hesitate either to ping me here or in a new issue.

closed 2024-12-17 by maximbaz