max-baz/wluma #129: [Bug]: Internally calculated luma value goes only as high as 93 even on a full white contents

salman-farooq-sh, 2024-12-18:
### Steps for reproducing the issue

Log luma values in wluma by opening a fullscreen white color [here](https://deadpixeltest.org/) without any screen dimming software etc. running that might interfere with wluma's luma reading.

### What is the buggy behavior?

It only goes as high as 93.

### What is the expected behavior?

It should go as high as 100.

### Version

master

### Environment

```shell
> lsb_release -a; uname -a; pgrep -l sway; pacman -Q | egrep "(wlroots|vulkan|sway|hyprland|clang|rust)"; dpkg -l | egrep "(wlroots|vulkan|sway|hyprland|clang|rust)"; echo
 "WLR_DRM_NO_MODIFIERS=$WLR_DRM_NO_MODIFIERS"
fish: Unknown command: lsb_release
Linux archlinux 6.12.4-zen1-1-zen #1 ZEN SMP PREEMPT_DYNAMIC Mon, 09 Dec 2024 14:30:31 +0000 x86_64 GNU/Linux
1539 swayosd-server
1591 swaybg
egrep: warning: egrep is obsolescent; using grep -E
clang 18.1.8-4
hyprland 0.45.2-3
lib32-vulkan-icd-loader 1.4.303-1
lib32-vulkan-intel 1:24.3.1-3
lib32-vulkan-mesa-layers 1:24.3.1-3
lib32-vulkan-radeon 1:24.3.1-3
lib32-vulkan-validation-layers 1.3.296.0-1
python-pywlroots 0.17.0-2
rust-analyzer 20241216-1
rustup 1.27.1-1
sway 1:1.10-1
swaybg 1.2.1-1
swayidle 1.8.0-2
swaylock-fprintd-git r314.ffd639a-2
swayosd-git 0.1.0.r0.gdbb8b72-1
vulkan-extra-tools 1.3.298-1
vulkan-headers 1:1.4.303-1
vulkan-icd-loader 1.4.303-1
vulkan-intel 1:24.3.1-3
vulkan-mesa-layers 1:24.3.1-3
vulkan-radeon 1:24.3.1-3
vulkan-tools 1.4.303-2
vulkan-utility-libraries 1.4.303-1
vulkan-validation-layers 1.3.296.0-1
wlroots 0.18.2-1
wlroots0.17 0.17.4-3
fish: Unknown command: dpkg
fish: 
lsb_release -a; uname -a; pgrep -l sway; pacman -Q | egrep "(wlroots|vulkan|sway|hyprland|clang|rust)"; dpkg -l | egrep "(wlroots|vulkan|sway|hyprland|clang|rust)"; echo "WLR_DRM_NO_MODIFIERS=$WLR_DRM_NO_MODIFIERS"
                                                                                                        ^~~^
WLR_DRM_NO_MODIFIERS=
```

salman-farooq-sh, 2024-12-18:
@maximbaz Can you tell what it shows on your setup and what your setup is?

maximbaz, 2024-12-18:
Ah sorry I should have said more explicitly 😊 run that branch using `RUST_LOG=trace cargo run` and open fully white screen, long list of numbers should be printed, if not adjust the condition in the latest commit of that branch - my list consists only of 255, which is the expected result on fully white screen.

salman-farooq-sh, 2024-12-18:
With a small change on top the main branch:

```
-const BUFFER_PIXELS: u64 = 500 * 4;
+const BUFFER_PIXELS: u64 = 3840 * 2400 * 4; // 3840x2400 is my display resolution, the only display attached
```

I can see the luma value shown in new learning entries with [als.none] be a full 100. I think the bug is solved?

maximbaz, 2024-12-18:
Impressive bug :joy: In fact `563 * 4` would be enough for your resolution. Very nice of me to test on my 4k screen `16:9` back in the days and think "ah who will ever buy a larger screen than me", let alone `16:10` :joy: 

Thanks for catching this!

maximbaz, 2024-12-18:
Should be fixed by https://github.com/maximbaz/wluma/commit/f6f0312941b32a92622cbdc94a3c34e3763b33e8 :pray:

closed 2024-12-18 by maximbaz