max-baz/wluma #132: [Feature request]: chromebook support

cypa, 2024-12-19:
### Please describe your feature request

how to add chromebook support? my device is Google Pixel 1 (platform name LINK)
```
$ cat /sys/bus/iio/devices/iio:device0/name
isl29018
⁅RUST_BACKTRACE=full wluma                
[2024-12-19T10:39:21Z WARN  wluma] Skipping 'keyboard-dell' as it might be disconnected: No such file or directory (os error 2)
[2024-12-19T10:39:21Z INFO  wluma] Continue adjusting brightness and wluma will learn your preference over time.
thread 'als' panicked at src/main.rs:112:26:
Unable to initialize ALS IIO sensor: "No iio device found"
stack backtrace:
   0:     0x591c1026dee2 - <unknown>
   1:     0x591c10290d63 - <unknown>
   2:     0x591c1026b32f - <unknown>
   3:     0x591c1026dd92 - <unknown>
   4:     0x591c1026f06e - <unknown>
   5:     0x591c1026eeb0 - <unknown>
   6:     0x591c1003b4fe - <unknown>
   7:     0x591c1026f7b9 - <unknown>
   8:     0x591c1026f50a - <unknown>
   9:     0x591c1026e3a9 - <unknown>
  10:     0x591c1026f16c - <unknown>
  11:     0x591c1002bd50 - <unknown>
  12:     0x591c1002c196 - <unknown>
  13:     0x591c10082fbf - <unknown>
  14:     0x591c10033920 - <unknown>
  15:     0x591c10271e4b - <unknown>
  16:     0x7fbc6308439d - <unknown>
  17:     0x7fbc6310949c - <unknown>
  18:                0x0 - <unknown>
```

cypa, 2024-12-19:
in the daylight I have these 
```
⁅cat /sys/bus/iio/devices/iio:device0/in_illuminance0_input
0
⁅cat /sys/bus/iio/devices/iio:device0/in_illuminance0_calibscale 
1.000000
⁅cat /sys/bus/iio/devices/iio:device0/in_illuminance0_scale 
0.015258
⁅cat /sys/bus/iio/devices/iio:device0/in_illuminance0_integration_time 
0.090000
```

cypa, 2024-12-19:
https://lightrun.com/answers/kootenpv-brightml-new-als-isl29018-on-acer-c720p

maximbaz, 2024-12-19:
Thanks for the report! Could you try latest `main` with https://github.com/maximbaz/wluma/commit/db77d27eadb7fe015fbf8e9c91e149a6525aa674 ?

I can find surprisingly little information about how to properly work with all those different files. As far as I understand, when you have `_input` file, that's already "processed" value, ready for us to use. It's only the `_raw` files that we have to apply `_offset` and `_scale` on top.

I also found in chromium's source code how they work with ALS, that seems wrong to me and contradicts what little pieces I could find elsewhere - they seem to take whatever file is present, `raw` or not, ignore offset and always multiply it `in_intensity_scale` (not even by the matching `_scale`). It _feels_ wrong, but time will tell if it was me who was wrong all along :joy: 

https://chromium.googlesource.com/chromium/src/+/refs/heads/main/services/device/generic_sensor/linux/sensor_data_linux.cc#21

maximbaz, 2024-12-25:
Fixed in 4.6.0 (probably :grin:) - please do leave a comment here if it's not working.

closed 2024-12-25 by maximbaz