max-baz/wluma #133: [Bug]: Brightness change makes the brightness change to what value you attempted to set it to, then instantly reverts back to where you started, then finally settles to what you attempted to set it to

salman-farooq-sh, 2025-01-01:
### Steps for reproducing the issue

Title basically.

More details: https://github.com/maximbaz/wluma/pull/127#issuecomment-2561017868

### What is the buggy behavior?

-

### What is the expected behavior?

-

### Version

`main`

### Environment

```shell
-
```

salman-farooq-sh, 2025-01-01:
Sorry for not fleshing this out properly.

maximbaz, 2025-01-01:
Could you try latest `main` with https://github.com/maximbaz/wluma/commit/8648d9b93d839b83048e08f789b16dc5de6a66fa ?

salman-farooq-sh, 2025-01-01:
It seems to be working great now. This issue was also problematic with screen dimming using hypridle with `brightnessctl --save/--restore` and that also seems to be fixed. So thanks a bunch for this!

I will need to observe it for a few days to be sure and then I will close the issue.

maximbaz, 2025-01-07:
I've just published 4.6.1 with this fix included :muscle: If you anyway stumble upon this or similar issue in the future, lets reopen and debug some more!

closed 2025-01-07 by maximbaz

salman-farooq-sh, 2025-01-07:
Thanks. I forgot to close this myself.