max-baz/wluma #134: Did not receive initial ALS value in time

pasikonik, 2025-01-03:
### Steps for reproducing the issue

I have a long approach to the setup after buying a new laptop and unfortunately, I have a problem that I am unable to solve.

I have seen that this error repeats a few times and requires deep debugging. Unfortunately, I don't know Rust, so I'm not sure if I can help with this, but I will do everything in my power.

### What is the buggy behavior?

app crashes with `Did not receive initial ALS value in time`

### What is the expected behavior?

run smoothly without errors

### Logs

```shell
[2025-01-03T16:30:51Z DEBUG wluma] Using Config {
        als: Time {
            thresholds: {
                16: "normal",
                20: "night",
                0: "night",
                7: "dark",
                13: "bright",
                9: "dim",
                11: "normal",
                18: "dark",
            },
        },
        output: [
            Backlight(
                BacklightOutput {
                    name: "eDP-1",
                    path: "/sys/class/backlight/amdgpu_bl1",
                    capturer: None,
                    min_brightness: 1,
                },
            ),
            DdcUtil(
                DdcUtilOutput {
                    name: "DELL U2417H",
                    capturer: None,
                    min_brightness: 1,
                },
            ),
        ],
    }
[2025-01-03T16:30:51Z DEBUG wluma::brightness::backlight] Using direct write on /sys/class/backlight/amdgpu_bl1 to change brightness value
thread 'predictor-eDP-1' panicked at src/predictor/controller.rs:62:25:
Did not receive initial ALS value in time
```

### Version

wluma 4.5.1

### Environment

```shell
LSB Version:	n/a
Distributor ID:	ManjaroLinux
Description:	Manjaro Linux
Release:	24.2.1
Codename:	Yonada
Linux elrakieta 6.11.11-1-MANJARO #1 SMP PREEMPT_DYNAMIC Thu, 05 Dec 2024 16:26:44 +0000 x86_64 GNU/Linux
lib32-vulkan-icd-loader 1.4.303-1
lib32-vulkan-intel 1:24.2.8-1
lib32-vulkan-nouveau 1:24.2.8-1
lib32-vulkan-radeon 1:24.2.8-1
vulkan-headers 1:1.4.303-1
vulkan-icd-loader 1.4.303-1
vulkan-intel 1:24.2.8-1
vulkan-nouveau 1:24.2.8-1
vulkan-radeon 1:24.2.8-1
vulkan-validation-layers 1.3.296.0-1
zsh: command not found: dpkg
WLR_DRM_NO_MODIFIERS=
```

maximbaz, 2025-01-10:
I'm not really sure it's related to external screens specifically (does it work if you comment out the internal display instead?). If it still works, it sounds more like an issue with having multiple screens, whereas if it doesn't work then it sounds more like an issue specifically with external screen.

pasikonik, 2025-01-10:
It turns out to be an issue with multiple screens. After commenting out the internal monitor, it works correctly.

maximbaz, 2025-01-10:
Okay, good to know! Hmm I can't imagine how it's possible that nobody else has reported this so far, I thought there are plenty of users using it with multiple screens :thinking: 

Unfortunately I don't have an easy way to get a second screen to try this, how interested would you be to try to debug this? 

The first simple thing could be to just run the latest code from `main` with `RUST_LOG=trace cargo run`, I just pushed some additional debugging code that might help us figure out what's wrong.

Harmiess-01, 2025-01-20:
I'm receiving the same error on my Surface Laptop Studio but am only using the built in monitor. I followed both the permissions and systemd methods in the README. However, using the webcam does work with the same monitor output.

**Logs**
Ambient light sensor:
```
[2025-01-20T21:46:52Z DEBUG wluma] Using Config {
        als: Iio {
            path: "/sys/bus/iio/devices",
            thresholds: {
                20: "dark",
                250: "normal",
                80: "dim",
                500: "bright",
                800: "outdoors",
                0: "night",
            },
        },
        output: [
            Backlight(
                BacklightOutput {
                    name: "eDP-1",
                    path: "/sys/class/backlight/intel_backlight",
                    capturer: None,
                    min_brightness: 1,
                    predictor: Adaptive,
                },
            ),
        ],
    }
[2025-01-20T21:46:52Z DEBUG wluma::brightness::backlight] Using direct write on /sys/class/backlight/intel_backlight to change brightness value
[2025-01-20T21:46:52Z INFO  wluma] Continue adjusting brightness and wluma will learn your preference over time.
thread 'predictor-eDP-1' panicked at src/predictor/controller/adaptive.rs:31:25:
Did not receive initial ALS value in time
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

Webcam:
```
[2025-01-20T21:55:46Z DEBUG wluma] Using Config {
        als: Webcam {
            video: 0,
            thresholds: {
                30: "dim",
                60: "bright",
                0: "night",
                45: "normal",
                75: "outdoors",
                15: "dark",
            },
        },
        output: [
            Backlight(
                BacklightOutput {
                    name: "eDP-1",
                    path: "/sys/class/backlight/intel_backlight",
                    capturer: None,
                    min_brightness: 1,
                    predictor: Adaptive,
                },
            ),
        ],
    }
[2025-01-20T21:55:46Z DEBUG wluma::brightness::backlight] Using direct write on /sys/class/backlight/intel_backlight to change brightness value
[2025-01-20T21:55:46Z INFO  wluma] Continue adjusting brightness and wluma will learn your preference over time.
```

**Version**
4.6.1-1

**Environment**
Package: `wluma` AUR
DE: KDE Plasma 6.2.5
WM: KWin (Wayland)

```
LSB Version:    n/a
Distributor ID: EndeavourOS
Description:    EndeavourOS Linux
Release:        rolling
Codename:       rolling
Linux eos-surface 6.12.7-arch1-1-surface #1 SMP PREEMPT_DYNAMIC Sat, 28 Dec 2024 04:04:04 +0000 x86_64 GNU/Linux
egrep: warning: egrep is obsolescent; using grep -E
lib32-vulkan-icd-loader 1.4.303-1
vulkan-headers 1:1.4.303-1
vulkan-icd-loader 1.4.303-1
vulkan-tools 1.4.303-2
bash: dpkg: command not found
egrep: warning: egrep is obsolescent; using grep -E
WLR_DRM_NO_MODIFIERS=
```

maximbaz, 2025-01-20:
Thanks for sharing your details! Does it crash right away or does it really wait 5 seconds and then crash?

4.6.1 unfortunately doesn't include the additional debugging I pushed for this issue, so we don't have much info to work against. Would you be interested to try debugging this locally, just adding some logs here and there to see where it gets stuck? Key locations are `src/als/controller.rs` and `src/als/iio.rs`.

pasikonik, 2025-03-24:
Sorry for being away for a bit, I forgot about this thread. I'll try to provide more logs soon.

pasikonik, 2025-03-24:
Result of `RUST_LOG=trace wluma` so far.

```
[2025-03-24T13:21:25Z DEBUG wluma] == wluma v4.9.0 ==
[2025-03-24T13:21:25Z DEBUG wluma] Using Config {
        als: Time {
            thresholds: {
                0: "night",
                7: "dark",
                18: "dark",
                16: "normal",
                20: "night",
                11: "normal",
                9: "dim",
                13: "bright",
            },
        },
        output: [
            Backlight(
                BacklightOutput {
                    name: "card1-eDP-1",
                    path: "/sys/class/backlight/amdgpu_bl1",
                    capturer: None,
                    min_brightness: 1,
                    predictor: Adaptive,
                },
            ),
            DdcUtil(
                DdcUtilOutput {
                    name: "DELL U2417H",
                    capturer: None,
                    min_brightness: 1,
                    predictor: Adaptive,
                },
            ),
        ],
    }
[2025-03-24T13:21:25Z DEBUG wluma::brightness::backlight] Using direct write on /sys/class/backlight/amdgpu_bl1 to change brightness value
[2025-03-24T13:21:25Z TRACE ddc_hi] DisplayInfo::from_edid(I2cDevice, 22786)
[2025-03-24T13:21:30Z TRACE ddc_hi] DisplayInfo::from_capabilities(I2cDevice, 22786)
[2025-03-24T13:21:30Z DEBUG wluma::brightness::ddcutil] Discovered displays (check_caps=true): ["DELL U2417H 5K9YD87J128L DEL"]
[2025-03-24T13:21:30Z DEBUG wluma::brightness::ddcutil] Using display 'DELL U2417H 5K9YD87J128L DEL' for config 'DELL U2417H' (check_caps=true)

thread 'predictor-card1-eDP-1' panicked at src/predictor/controller/adaptive.rs:31:25:
Did not receive initial ALS value in time: Timeout
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

I'll try do some debugging locally.

pasikonik, 2025-03-24:
Unfortunately. I tries adding logs everywhere (`controller.rs` and `iio.rs`) where it might be helpful, and I didn't get any output except for the ones I have in the message above.

maximbaz, 2025-03-25:
Please try using branch `issue143`, I added some extra logs around the initialization, that happens just before your timeout occurs, lets see what happens now, I'm especially curious to see the time of the log messages.

pasikonik, 2025-03-26:
```
[2025-03-26T15:18:01Z DEBUG wluma::brightness::backlight] Using direct write on /sys/class/backlight/amdgpu_bl1 to change brightness value
[2025-03-26T15:18:01Z DEBUG wluma::predictor::controller::adaptive] Inside adaptive controller, waiting for initial ALS value
[2025-03-26T15:18:01Z TRACE ddc_hi] DisplayInfo::from_edid(I2cDevice, 22786)
[2025-03-26T15:18:01Z TRACE ddc_hi] DisplayInfo::from_edid(I2cDevice, 22796)
[2025-03-26T15:18:06Z TRACE ddc_hi] DisplayInfo::from_capabilities(I2cDevice, 22786)
[2025-03-26T15:18:06Z DEBUG wluma::brightness::ddcutil] Discovered displays (check_caps=true): ["DELL U2417H 5K9YD87J128L DEL"]
[2025-03-26T15:18:06Z DEBUG wluma::brightness::ddcutil] Using display 'DELL U2417H 5K9YD87J128L DEL' for config 'DELL U2417H' (check_caps=true)
[2025-03-26T15:18:06Z DEBUG wluma::predictor::controller::adaptive] Inside adaptive controller, timeout waiting for initial ALS value
```

maximbaz, 2025-03-26:
Very curious, does the code execution not reach this line at all? Could you try to debug where it gets stuck inside main.rs?

https://github.com/maximbaz/wluma/blob/ccb77db2d7ad6708d97d40c17ee8601ce2b0b584/src/main.rs#L131

pasikonik, 2025-03-26:
```

maximbaz, 2025-06-12:
The errors unfortunately most likely mean that the monitor doesn't properly speak DDC protocol, and the monitor matching of course depends on the fact that you put the correct value in the config, that would match the proper monitor, as described in https://github.com/maximbaz/wluma#displays 🙂

maximbaz, 2025-06-12:
I don't own an external monitor, so I don't think I would be able to do anything on this issue. Let me know if you discover something interesting in the code!