max-baz/wluma #136: [Bug]: wluma keep setting the brightness to minimum value

Tim-C1, 2025-01-23:
### Steps for reproducing the issue

compile the source code, `cargo run`

### What is the buggy behavior?

wluma set the brightness to minimum value

### What is the expected behavior?

should adjust the brightness according to ambient sensors

### Logs

```shell
RUST_LOG=debug wluma

[2025-01-23T03:06:12Z DEBUG wluma] == wluma v4.6.1-3-g45c61c2 ==
[2025-01-23T03:06:12Z DEBUG wluma] Using Config {
        als: Iio {
            path: "/sys/bus/iio/devices",
            thresholds: {
                8: "normal",
                15: "outdoors",
                0: "night",
                2: "dark",
                5: "dim",
                13: "bright",
            },
        },
        output: [
            Backlight(
                BacklightOutput {
                    name: "eDP-1",
                    path: "/sys/class/backlight/intel_backlight",
                    capturer: Wayland(
                        Any,
                    ),
                    min_brightness: 1,
                    predictor: Adaptive,
                },
            ),
        ],
    }
[2025-01-23T03:06:12Z DEBUG wluma::brightness::backlight] Using DBUS for /sys/class/backlight/intel_backlight to change brightness value
[2025-01-23T03:06:12Z DEBUG wluma::brightness::backlight] max_brightness: 400
[2025-01-23T03:06:12Z DEBUG wluma::brightness::backlight] min_brightness: 1
[2025-01-23T03:06:12Z INFO  wluma] Continue adjusting brightness and wluma will learn your preference over time.
[2025-01-23T03:06:12Z DEBUG wluma::frame::capturer::wayland] Detected support for wlr-screencopy-unstable-v1 protocol
[2025-01-23T03:06:12Z DEBUG wluma::frame::capturer::wayland] Detected support for linux-dmabuf-v1 protocol
[2025-01-23T03:06:12Z DEBUG wluma::frame::capturer::wayland] Using output 'Samsung Display Corp. 0x416D  (eDP-1)' for config 'eDP-1'
[2025-01-23T03:06:12Z DEBUG wluma::frame::capturer::wayland] Using wlr-screencopy-unstable-v1 protocol to request frames
[2025-01-23T03:06:13Z TRACE wluma::als::iio] ALS (iio): outdoors (44)
[2025-01-23T03:06:13Z TRACE wluma::predictor::controller::adaptive] Prediction: 0 (lux: outdoors, luma: 0)
[2025-01-23T03:06:13Z TRACE wluma::predictor::controller::adaptive] Prediction: 0 (lux: outdoors, luma: 21)
[2025-01-23T03:06:13Z TRACE wluma::als::iio] ALS (iio): outdoors (44)
[2025-01-23T03:06:13Z TRACE wluma::predictor::controller::adaptive] Prediction: 0 (lux: outdoors, luma: 21)
[2025-01-23T03:06:13Z TRACE wluma::predictor::controller::adaptive] Prediction: 0 (lux: outdoors, luma: 21)
[2025-01-23T03:06:13Z TRACE wluma::als::iio] ALS (iio): outdoors (44)
[2025-01-23T03:06:13Z TRACE wluma::predictor::controller::adaptive] Prediction: 0 (lux: outdoors, luma: 21)
[2025-01-23T03:06:14Z TRACE wluma::predictor::controller::adaptive] Prediction: 0 (lux: outdoors, luma: 22)
[2025-01-23T03:06:14Z TRACE wluma::als::iio] ALS (iio): outdoors (44)
[2025-01-23T03:06:14Z TRACE wluma::predictor::controller::adaptive] Prediction: 0 (lux: outdoors, luma: 22)
[2025-01-23T03:06:14Z TRACE wluma::als::iio] ALS (iio): outdoors (44)
[2025-01-23T03:06:14Z TRACE wluma::predictor::controller::adaptive] Prediction: 0 (lux: outdoors, luma: 21)
[2025-01-23T03:06:14Z TRACE wluma::predictor::controller::adaptive] Prediction: 0 (lux: outdoors, luma: 22)
[2025-01-23T03:06:14Z TRACE wluma::als::iio] ALS (iio): outdoors (44)
[2025-01-23T03:06:14Z TRACE wluma::predictor::controller::adaptive] Prediction: 0 (lux: outdoors, luma: 22)
[2025-01-23T03:06:14Z TRACE wluma::predictor::controller::adaptive] Prediction: 0 (lux: outdoors, luma: 22)
[2025-01-23T03:06:14Z TRACE wluma::als::iio] ALS (iio): outdoors (44)
[... 77 lines cut ...]

The latest source code on github: `45c61c2054fc6e09d2e590d6853a2afb137d575c`

### Environment

```shell
lsb_release -a; uname -a; pgrep -l sway; pacman -Q | egrep "(wlroots|vulkan|sway|hyprland|clang|rust)"; dpkg -l | egrep "(wlroots|vulkan|sway|hyprland|clang|rust)"; echo "WLR_DRM_NO_MODIFIERS=$WLR_DRM_NO_MODIFIERS"

Linux evael 6.6.72-1-lts #1 SMP PREEMPT_DYNAMIC Fri, 17 Jan 2025 14:04:26 +0000 x86_64 GNU/Linux
clang 19.1.7-1
hyprland 0.46.2-7
hyprland-protocols-git 0.4.0.r1.a7c1838-1
hyprland-qt-support 0.1.0-2
hyprland-qtutils 0.1.3-2
rust 1:1.84.0-1
vulkan-headers 1:1.4.303-1
vulkan-icd-loader 1.4.303-1
vulkan-intel 1:24.3.3-2
vulkan-validation-layers 1.3.296.0-1
wlroots 0.18.2-1
wlroots0.15 0.15.1-6
wlroots0.15-debug 0.15.1-6
wlroots0.17 0.17.4-3
xdg-desktop-portal-hyprland 1.3.9-3
```

maximbaz, 2025-01-23:
Hello, after you install and run wluma you are expected to continue adjusting brightness manually - wluma will be observing what values you set brightness to, in which environment conditions, and then start doing it for you, and every time it makes incorrect prediction you are expected to teach it by again setting the desired value. Give this a go 😊

Tim-C1, 2025-01-23:
> Hello, after you install and run wluma you are expected to continue adjusting brightness manually - wluma will be observing what values you set brightness to, in which environment conditions, and then start doing it for you, and every time it makes incorrect prediction you are expected to teach it by again setting the desired value yourself, manually. Give this a go 😊

Hey, thank you for your response! It looks like it does not allow me to manually adjust the brightness. After running wluma I tried to raise my brightness through the hotkeys, but wluma keeps setting the brightness to the lowest at the same time I adjust the brightness.

maximbaz, 2025-01-23:
Interesting, could you please provide the same `RUST_LOG=trace cargo run` output while you are trying that? It will perhaps tell us more what is going on there, why it is ignoring your inputs 🤔 

Do you have some file in `~/.local/share/wluma/` ? Could you also try to stop wluma, remove it and restart?

Tim-C1, 2025-01-23:
> Interesting, could you please provide the same `RUST_LOG=trace cargo run` output while you are trying that? It will perhaps tell us more what is going on there, why it is ignoring your inputs 🤔
>
> Do [...]

Oh that's it. There are some legacy wluma config files there, I remove them and it's working normally! Thank you for your timely response and creation for such a wonderful project!

closed 2025-01-23 by Tim-C1