max-baz/wluma #139: [Bug]: Keyboard backlight not being adjusted

regenman, 2025-02-18:
### Steps for reproducing the issue

Set up keyboard backlight as follows:
```
[[keyboard]]
name = "keyboard-thinkpad"
path = "/sys/bus/platform/devices/thinkpad_acpi/leds/tpacpi::kbd_backlight"
```
Run wluma, adjust keyboard backlight according to preference. Move to a different ambient light where wluma should change keyboard backlight. Observe that the backlight isn't being adjusted. 

### What is the buggy behavior?

On my system (Thinkpad laptop), wluma does not adjust the keyboard backlight automatically, even when the trace log suggests that it would (judging by the "Prediction"). There are no errors or other signs of things going wrong.

Using `dbus-monitor`, I tried to catch any wluma attempts at adjusting the backlight, but all I could find were requests related to display brightness (intel_backlight), through which the display is correctly being auto-adjusted. My keyboard backlight (`/sys/bus/platform/devices/thinkpad_acpi/leds/tpacpi::kbd_backlight`) has three levels: 0=off, 1=medium, 2=bright. 

### What is the expected behavior?

wluma should change the keyboard backlight according to "learned" preferences.

### Logs

```shell
$ RUST_LOG=trace wluma
[2025-02-18T16:23:08Z DEBUG wluma] Using Config {
        als: Iio {
            path: "/sys/bus/iio/devices",
            thresholds: {
                500: "outdoors",
                80: "bright",
                2: "dark",
                15: "normal",
                5: "dim",
                0: "night",
            },
        },
        output: [
            Backlight(
                BacklightOutput {
                    name: "eDP-1",
                    path: "/sys/class/backlight/intel_backlight",
                    capturer: Wayland(
                        Any,
                    ),
                    min_brightness: 1,
                    predictor: Adaptive,
                },
            ),
            Backlight(
                BacklightOutput {
                    name: "keyboard-thinkpad",
                    path: "/sys/bus/platform/devices/thinkpad_acpi/leds/tpacpi::kbd_backlight",
                    capturer: None,
                    min_brightness: 0,
                    predictor: Adaptive,
                },
            ),
        ],
    }
[2025-02-18T16:23:08Z DEBUG wluma::brightness::backlight] Using DBUS for /sys/class/backlight/intel_backlight to change brightness value
[2025-02-18T16:23:08Z DEBUG wluma::frame::capturer::wayland] Detected support for wlr-screencopy-unstable-v1 protocol
[2025-02-18T16:23:08Z DEBUG wluma::frame::capturer::wayland] Detected support for linux-dmabuf-v1 protocol
[2025-02-18T16:23:08Z DEBUG wluma::brightness::backlight] Using DBUS for /sys/bus/platform/devices/thinkpad_acpi/leds/tpacpi::kbd_backlight to change brightness value
[2025-02-18T16:23:08Z DEBUG wluma::frame::capturer::wayland] Using output 'LG Display - 0x05A7 - eDP-1' for config 'eDP-1'
[2025-02-18T16:23:08Z DEBUG wluma::frame::capturer::wayland] Using wlr-screencopy-unstable-v1 protocol to request frames
[2025-02-18T16:23:08Z INFO  wluma] Continue adjusting brightness and wluma will learn your preference over time.
[2025-02-18T16:23:09Z TRACE wluma::als::iio] ALS (iio): night (1)
[2025-02-18T16:23:09Z TRACE wluma::predictor::controller::adaptive] Prediction: 9 (lux: night, luma: 22)
[2025-02-18T16:23:09Z TRACE wluma::predictor::controller::adaptive] Prediction: 2 (lux: night, luma: 0)
[2025-02-18T16:23:09Z TRACE wluma::predictor::controller::adaptive] Prediction: 9 (lux: night, luma: 22)
[... 377 lines cut ...]
path = "/sys/class/backlight/intel_backlight"
capturer = "wayland"

[[keyboard]]
name = "keyboard-thinkpad"
path = "/sys/bus/platform/devices/thinkpad_acpi/leds/tpacpi::kbd_backlight"

$ ls -al /sys/bus/platform/devices/thinkpad_acpi/leds/tpacpi::kbd_backlight
total 0
drwxr-xr-x  3 root root     0 18 feb 15.46 .
drwxr-xr-x 10 root root     0 18 feb 15.46 ..
lrwxrwxrwx  1 root root     0 18 feb 15.46 device -> ../../../thinkpad_acpi
drwxr-xr-x  2 root root     0 18 feb 15.46 power
lrwxrwxrwx  1 root root     0 18 feb 15.46 subsystem -> ../../../../../class/leds
-rw-rw-r--  1 root video 4096 17 feb 14.34 brightness
-r--r--r--  1 root root  4096 18 feb 15.46 brightness_hw_changed
-r--r--r--  1 root root  4096 18 feb 15.46 max_brightness
-rw-r--r--  1 root root     0 18 feb 15.46 trigger
-rw-r--r--  1 root root  4096 18 feb 15.46 uevent

$ cat /sys/bus/platform/devices/thinkpad_acpi/leds/tpacpi::kbd_backlight/brightness
0
$ cat /sys/bus/platform/devices/thinkpad_acpi/leds/tpacpi::kbd_backlight/max_brightness
2
```

maximbaz, 2025-02-18:
Hello! Thanks for the detailed report.

It looks like it may be impossible to change keyboard brightness using dbus 🤔 

A quick workaround for you could be to [join `video` group and reboot](https://github.com/maximbaz/wluma?tab=readme-ov-file#permissions).

If we can find a dbus command that is capable of changing keyboard brightness, I'd be happy to incorporate it.

The `dbus-send` command similar to what `wluma` executes for your screen is probably something like this:

```
dbus-send --system --dest=org.freedesktop.login1 \
          --type=method_call \
          --print-reply \
          /org/freedesktop/login1/session/auto \
          org.freedesktop.login1.Session.SetBrightness \
          string:"backlight" string:"intel_backlight" uint32:90
```

For keyboard, `wluma` tries to execute the command below, which I suspect does not work:

```
dbus-send --system --dest=org.freedesktop.login1 \
          --type=method_call \
          --print-reply \
          /org/freedesktop/login1/session/auto \
          org.freedesktop.login1.Session.SetBrightness \
          string:"backlight" string:"tpacpi::kbd_backlight" uint32:2
```

Do you think you could help finding a command that would work for keyboard?

regenman, 2025-02-18:
> dbus-send --system --dest=org.freedesktop.login1 \
>           --type=method_call \
>           --print-reply \
>           /org/freedesktop/login1/session/auto \ [...]

Thank you! This was of great help. Indeed, the above command failed with "No such device". So I consulted [1](https://www.freedesktop.org/software/systemd/man/latest/org.freedesktop.login1.html), and understood that I should probably change "backlight" to "leds" (since my backlight is under `/sys/bus/platform/devices/thinkpad_acpi**/leds/**`. And indeed, the following command successfully alters my keyboard backlight:
```
dbus-send --system --dest=org.freedesktop.login1 \
          --type=method_call \
          --print-reply \
          /org/freedesktop/login1/session/auto \
          org.freedesktop.login1.Session.SetBrightness \
          string:"leds" string:"tpacpi::kbd_backlight" uint32:2
```

maximbaz, 2025-02-18:
Ooooh, amazing finding! Thank you!

So it sounds like we need to update this line to pass not a hardcoded "backlight", but the folder name preceeding the ID.

https://github.com/maximbaz/wluma/blob/45c61c2054fc6e09d2e590d6853a2afb137d575c/src/brightness/backlight.rs#L58

Would you like to give a go at the PR? No worries if not, I can also take it from here.

regenman, 2025-02-18:
> Would you like to give a go at the PR? No worries if not, I can also take it from here.

I could give it a go yeah, it's an interesting learning opportunity given that I have zero experience with Rust.

maximbaz, 2025-02-18:
Awesome, I'll all about making it interesting 😁 Enjoy, dont hesistate to ping me if you get stuck or unsure about anything!

regenman, 2025-02-18:
> Awesome, I'm all about making it interesting 😁 Enjoy, dont hesistate to ping me if you get stuck or unsure about anything!

Thank you :)
One question that comes to mind, do you think we should or check strictly that the parent directory is either "leds" or "backlight" (the only two subsystems that exist in the specification), or use a fallback in case of some unusual path? What if the user specifies a symlink residing in a different location, for example..

maximbaz, 2025-02-18:
I think validating that it is one of the two documented values is a good idea, let's just throw an error in that case - this project has a friendly community and I'd like to hear from people about their use-cases, why their path is special. I don't know anyway what a good fallback would be in any case, if we fallback to "backlight" but it was a keyboard, then wluma would just pretend that it works but nothing would actually be happening (like in your original case), I don't like that :)

PR #140 (2025-02-18): fix(backlight): Extract subsystem from device path
... instead of using hardcoded value in the dbus call. Fixes setting backlight on keyboards that appear under the "leds" subsystem.

Closes #139
closed 2025-02-18 by maximbaz