max-baz/wluma #141: [Feature request]: Support 10 bit monitor

savaobay, 2025-02-21:
### Please describe your feature request

Trying to use wluma on hyprland with 10 bit monitor active

[image]

maximbaz, 2025-02-21:
Very interesting! I would like to support that, but this will touch a lot of code and require extensive testing to ensure that it _actually works_, and not just gives random results. This won't be possible without doing a lot of tests on 10 bit monitor, which I don't have.

How interested are you in helping with this, e.g. building code from different branches, collecting and sharing logs, etc? Do you have time and energy to help? 🙂

maximbaz, 2025-02-26:
Thanks! Let's start with this:

I pushed some code in a branch called [`10bit`](https://github.com/maximbaz/wluma/tree/10bit), let's use it for this ticket.

Here's the first experiment I'd like to ask you to run:

Run `RUST_LOG=trace cargo run` from that branch, it will be very slow and produce a lot of output (all RGBA values of your screen, all the time).

While it's running, go to https://deadpixeltest.org, press start, such that this runs in fullscreen, i.e. entire screen is filled with one color, there should be nothing else visible on the screen.

Switch color to red, keep red on the screen for a few seconds, then quickly go back to the running wluma and copy somewhere the logs for the past few seconds. What you see there is a list of all pixels on your screen, as wluma sees them - an array with a lot of numbers, where each 4 numbers represent a single pixel. On my 8-bit screen I'm getting `255, 0, 0, 255, ...` on repeat, meaning that wluma sees 255 for red, 0 for green, 0 for blue, 255 for alpha.

Right now this code on the branch will maybe (?) convert your 10 bit screen to 8 bit, but I'm not sure. I'm very curious what you will see.

Collect the same for green and blue and white and black too.

Let me know how it goes!

savaobay, 2025-02-26:
This is my log

[log.zip](https://github.com/user-attachments/files/18985915/log.zip)
It seems like the program is still running but the screen brightness doesn't change

maximbaz, 2025-02-26:
Whoa, to be honest I disabled the brightness change just because I didn't believe this first experiment would lead to such results 😅 

Fine, I just pushed the code to the same branch that should change brightness for you (once you train wluma) - you can verify by changing brightness manually a few times in different environments.

Please also run this second experiment:

The same setup, you open deadpixel.org in full screen, and have `RUST_LOG=trace cargo run` run in background. Open a specific color, leave it for a second or two, and then look at the logs that wluma rights, in particular the "luma: XXX" part:

```
Prediction: 260 (lux: none, ---->  luma: 13 <---- )
Prediction: 260 (lux: none, luma: 13)
Prediction: 260 (lux: none, luma: 13)
Prediction: 260 (lux: none, luma: 13)
Prediction: 237 (lux: none, luma: 17)
Prediction: 215 (lux: none, luma: 19)
Prediction: 215 (lux: none, luma: 19)
Prediction: 215 (lux: none, luma: 19)
```

Please confirm that when on deadpixel.org you have the following color in full screen, luma values are the following:

```
black: 0
white: 100
red: 49
green: 83
blue: 26
```

Thanks!

savaobay, 2025-02-26:
That works! The brightness has changed. 

> black: 0
white: 100
red: 49
green: 83
blue: 26

I confirm these values ​​are correct

maximbaz, 2025-02-26:
Thanks for confirming! it's honestly wild that it just works, I expected a lot more changes would be needed. But I do not complain 😅 We might be losing some precision by "downgrading" colors produced by your hardware to 8 bit before determining perceived lightness of your screen content (luma value in the log), but I suspect it's more than precise for our purposes here.

Thanks for your help, I'll cut a release shortly.

maximbaz, 2025-02-26:
Released in 4.8.0 🚀

closed 2025-02-26 by maximbaz