max-baz/wluma #144: [Feature request]: Predictor algorithm with window properties input

GordianDziwis, 2025-06-06:
### Please describe your feature request

With window properties, the algorithm could be trained to go brighter when a game or Firefox with a video or a video player/media player is full-screen.

maximbaz, 2025-06-06:
That's an interesting idea, do you know if it's actually possible to detect whether Firefox is full-screen, and not only that, but a Firefox that has a playing video is full-screen?

If you are just curious whether that would actually be useful in daily life, a much more manual, but theoretically possible approach to test this idea, is to play around with symlinks inside `~/.local/share/wluma` folder - for example, if you currently have `eDP-1.yaml` in there, rename it to `eDP-1_normal.yaml` and create another one `eDP-1_gaming.yaml`. Then point `eDP-1.yaml` as a symlink to the corresponding file when you want to switch "profile". Wrap it in a script (that does this + restarts wluma), put it on a hotkey, and you have a manual but relatively easy way to toggle between "normal" and "gaming" modes.

I'm definitely not saying that this shouldn't be built-in, but it could be a quick way to get started, and if you do end up heavily using it, it only makes sense to try to automate this selection of profile based on some rules or window properties.

GordianDziwis, 2025-06-07:
You are right, something like this would be a more elegant solution.

With sway this works:

```bash
#!/bin/bash

swaymsg -t subscribe -m '[ "window" ]' | while read -r event; do
    if echo "$event" | jq -e '.change == "fullscreen_mode"' > /dev/null; then
        if echo "$event" | jq -e '.container.fullscreen_mode == 1' > /dev/null; then
            notify-send "Fullscreen activated"
            systemctl --user stop wluma
            brightnessctl -c backlight set +100%
        else
            notify-send "Fullscreen deactivated"
            systemctl --user start wluma
        fi
    fi
done
```

closed 2025-06-07 by GordianDziwis

maximbaz, 2025-06-07:
Nice, very clean, and you have flexibility to extend this with very custom rules to match your preferences 👍